terjeio/grblHAL #72: Unexpected Homing command response

HuubBuis, 2020-07-15:
The homing command ($HX) returns a status message before returning OK.  This differs from grbl and Grbl_ESP32 and could be an issue for grbl gui's. 
A think a solution would be removing the status message or sending OK first

Tested driver: 
ESP32

Command send:
$HX

Response:
< Home | WPos:9.330,9.999,3.960 | Bf:35,1023 | FS:0,0 >
 ok

Response expected:
ok

terjeio, 2020-07-15:
I'll remove it when [COMPATIBILITY_LEVEL](https://github.com/terjeio/grblHAL/wiki/Changes-from-grbl-1.1#workaround) is set to > 0. Sequencing of ok messages and status reports are undefined as status report requests bypasses the normal command queue, all senders should handle that - if not they need to be fixed.

The HOME status is defined in the grbl specification but never sent by any other implementation I know of, defined and then forgotten about? Many of my drivers are for MCUs/cards that does not reset the processor on connect ulike the Arduino variants. This is a good thing since resetting the processor is likely to lead to loss of communication if it is over native USB or networking. I have made some changes related to that such as a possibility to query the homing status. An updated and common grbl comms specification would be nice, this to keep ports and senders in sync...

HuubBuis, 2020-07-15:
Ok,

I have adapted my sender already and will continue testing.

closed 2020-07-15 by HuubBuis