terjeio/grblHAL #74: Problem with BluePill - STM32F103C8 - Error codes and alarms.

AxelGUILLAUMET, 2020-07-20:
Dear friends. I'm very enthusiastic to use grblHAL on BluePill - STM32F103C8 because it looks the best supported project on the web at the date. Well done and very good work there!

As described into the information of the STM32F103C8 driver, it is loosely based on https://github.com/robomechs/6-AXIS-USBCNC-GRBL so I can assume the pinout is the same:

[image]

When I connect to GRBL Panel https://github.com/gerritv/Grbl-Panel/releases, I get partial results of a ALARM code instead of the machine coordinates that make dying the software:

[image]

When I send a laser cutting in LightBurn it works but when I stop the engraving, I get ALARM 3 and ERROR 9... One minte after, I can resume the engraving.

[image]

I put PB5 to ground (control reset)
I put PB8 to +3.3V (Safety door)

I tried to compile with compatibility levels 0,1 ,2 and 10 and I follow to get the same errors...

Where I can look to try to fix that?

Regards.

terjeio, 2020-07-20:
Alarm 3 is raised when the reset pin (PB 5) is asserted or a CTLR-X is received when a gcode program is running. So this could be due to noise triggering the reset input?

What is the output from a status request (`?` character sent) when this happens?  The `|Pn:` part holds the status of the input signals and with luck you may see if the reset pin asserted. However if the problem is due to noise the pulse triggering the reset pin could be too short to be detected by the report. From the screenshot it looks like that LaserBurn can show the reports when "Show all" is enabled in the console.

You could also [try with my sender](https://github.com/terjeio/Grbl-GCode-Sender), it is a bit similar to GrblPanel with additional functionality - among them display of status of the input signals. Releases can be found [here](https://github.com/terjeio/Grbl-GCode-Sender/releases).

AxelGUILLAUMET, 2020-07-20:
Dear friend, thanks for your quick answer! 

Your GRBL sender looks very nice! Good work keep it up!

For the reset problem with PB5, I connected PB5 directly to the ground to avoid those problems as you can see and I doubt this is due to a noise on the pin. When it is a noise, a problem appears randomly, in this case the problem is constant.

[image]

[image]

This is what I get in IDLE with your sender.

[image]

And here is what I get with LightBurn with the details:

```
Puerto abierto, esperando respuesta
ok
Comenzando transmisión
G00 G17 G40 G21 G54
G90
M4
M8
G0X153.18Y303.6
G91
G1X-150F12000S1000
G1Y-0.1S0
G1X150S1000
G1Y-0.1S0
G1X-150S1000
error:9
En o cerca de la línea 0:
error:9
En o cerca de la línea 0:
Transmisión completada en 0:00
error:9
error:9
error:9
error:9
error:9
error:9
error:9
error:9
error:9
```

My objective is getting to get it working on lightburn with GRBL standard in M4 mode.

terjeio, 2020-07-20:
This is strange, are you sure the shorting cable you are using has good contact? I have some that I am starting to throw away as they are making sloppy contact. Perhaps try with another one as a first step if you haven't already done so?

I have just started to work through all drivers to adjust them to changes in the grbl core, I guess the STMF103 driver will be updated tomorrow - I will run some tests on it then to see if I can reproduce this issue.

HuubBuis, 2020-07-20:
In the grbl real time report the limit pins are active, PN:PXYZHS

If you don't use limit & probe pins, it is better to connect them to GND because grblHAL is NC switches by default.

> I tried to compile with compatibility levels 0,1 ,2 and 10 and I follow to get the same errors...

If you change the compatibility level, some settings are changed and not all settings are reported by $$ or can be set by $xx=
reset the configuration using $rst=* and then configure grbl again.

terjeio, 2020-07-20:
Running test with a RedPill and STM32F103 driver now, no problems seen yet. Will run a longer test tomorrow.

> If you don't use limit & probe pins, it is better to connect them to GND because grblHAL is NC switches by default.

$5 and $14 can be used to invert the inputs as well, but preferably use NC switches as they provide better noise immunity and is inherently safer (a broken connection will trigger the input). Note that $14 is not available at all compatibility levels.

> If you change the compatibility level, some settings are changed and not all settings are reported by $$ or can be set by $xx=
reset the configuration using $rst=* and then configure grbl again.

This will be changed in the next build (20200716):

Most $-settings can be set at all compatibility levels, the exceptions are some new flags in settings extended in grblHAL such as the report options ($10). I will change the documentation when this is committed to the repository.

`$+` can be used to report all settings regardless of compatibility level.

AxelGUILLAUMET, 2020-07-21:
OK. 

I grounded the pins PB12,13,14,15 which are the limit-switches inputs.

I recompiled and uploaded the firmware with #define COMPATIBILITY_LEVEL 0

It doesn't fix and it's unstable.

Now I made a test with the classic "Universal G-Code sender" https://winder.github.io/ugs_website/

It connects but I can't jog, and when I start the G-Code streaming, this error is directly what I get.

[image]

Test it with UGS.

All the tests I made with GRBL Panel, Universal G-Code sender, Lightburn I made them with the original Arduino UNO and the official GRBL firmware and all there three applications are working perfectly without any problem.

terjeio, 2020-07-21:
> It connects but I can't jog, and when I start the G-Code streaming, this error is directly what I get.

Try with COMPATIBILITY_LEVEL set > 0 - it UGS does not work correctly with it set to 0, I can run gcode with it set to 1.
My UGS version:

[image]

$I response after sending

[image]

The reset input pin is read in this function:

https://github.com/terjeio/grblHAL/blob/0c0d6afcb85f9b5d22646046e9628274a6dc8463/drivers/STM32F1xx/Src/driver.c#L316-L341

As a test can you set the reset signal permanently to 0? This will show if reading the input is the source of this issue or something else is. Do it by adding this line at line 339:

`signals.reset = 0;`

AxelGUILLAUMET, 2020-07-21:
[image]

I tried and the error follows the same in lightburn...

Compatibility level 1 and I made a $RST=*

[image]

terjeio, 2020-07-21:
> And exactly the same problems with UGS... I you want I can make a video to show you how the behaviour of the softwares...

No need for that - I believe you. ALARM 3 is only set in the function `mc_reset()` in motion_control.c. We have to figure out from where this is called from when it should not. A couple of options remains which we can disable as well:

Real time command from the sender, handled here:

https://github.com/terjeio/grblHAL/blob/0c0d6afcb85f9b5d22646046e9628274a6dc8463/grbl/protocol.c#L674-L684

Comment out line 675, 676, 681 and 682 to disable.

Hard limits triggered, a bug if this happens - as hard limits are disabled according to your $21-setting:

https://github.com/terjeio/grblHAL/blob/0c0d6afcb85f9b5d22646046e9628274a6dc8463/grbl/limits.c#L55-L66

Comment out line 58, 59, 62 and 63 to disable.

This reminds me that pin mappings in the board map file @shaise made could be a reason for this, there is a potential problem with pin assingments in this file related to the way GPIO interrupts are assigned in the STM32. On  my todo list to investigate. This mapping file is used when `//#define BOARD_CNC3040` is uncommented in driver.h, it is not per default.

The third place is in the homing function, which is not likely to be called randomly. The calls to `mc_reset()` are further down in the limits.c file. There are also some calls made to `mc_reset()` in system.c apart from the control interrupt handler which were disabled by setting the reset signal to 0. These are triggered by $ commands.

Pleas try the two first options and report back.

HuubBuis, 2020-07-21:
I have hooked up a STM32F103F8 blue pill, flashed using #define COMPATIBILITY_LEVEL 1
Reset the configuration $rst=*
**I have only the USB port connected, nothing else**
my status: <Idle|WPos:10.935,-17.723,-9.211|Bf:35,512|FS:0,0|Pn:PXYZRHS|Ov:100,100,100>
I started UGS  Product Version: Universal Gcode Platform 20190423
I could jog and run a small gcode file multiple times without any problems.
Then I hit the Check Mode button and couldn't get it to work again, just got errors on safety door and more. Even a reset and disconnect didn't help!!!

inverted the control bits $14=255
And now it works again without any problems!
<Idle|WPos:6.273,-17.723,-10.341|Bf:35,512|FS:0,0|Pn:PXYZ>

inverted the limit pins $5=255
<Idle|WPos:6.273,-17.723,-10.341|Bf:35,512|FS:0,0|Pn:P>

inverted the probe pin $6=1
<Idle|WPos:6.273,-17.723,-10.341|Bf:35,512|FS:0,0|Pn:>

This is my default setup for bench testing, no signals connected. I do this on all controllers.

AxelGUILLAUMET, 2020-07-21:
Ok thanks for your answers, I'm reporting.

- First of all, I upgraded my UGS to the last 2019 stable build (I was using a 2015 version). I compiled COMPATIBILITY_LEVEL 1. This is working perfectly.

- GRBL Panel now is an abandoned project. So I will forget it and use https://github.com/terjeio/Grbl-GCode-Sender which look so much better!

Now the problems are with LightBurn (The more important, because this is the software I really need to use).

I connected the blue pill with drivers and steppers to see the full working of grblHAL firmware.

[image]

I configured GRBL with the laser cutter parameters to get a correct speed for a co2 laser cutter.

FIRST:

When I launch an engrave, it works perfectly. But when it ends, I don't have an home return. If I repeat the engrave, it will not engrave over the same engrave I made before.

Here is the generated GCode: [test laser.zip](https://github.com/terjeio/grblHAL/files/4955567/test.laser.zip)

Here is the summary of the gcode:

```
; LightBurn 0.9.14
; GRBL device profile, user origin
; Bounds: X3.18 Y3.74 to X18.88 Y21.54
G00 G17 G40 G21 G54
G90
G0 X0 Y0
G91
M4
; Scan @ 200 mm/sec, 100% power
M8
G0X15.7Y17.83
G1X-15.7F12000S1000
G1Y-0.1S0
G1X15.7S1000
.......
.......
......
G1Y-0.1S0
G1X-15.7S1000
G1Y-0.1S0
G1X15.7S1000
G1Y-0.1S0
G1X-15.7S1000
M9
G1S0
M5
G90
; return to user-defined finish pos
G0 X0 Y0
M2
```

When I load this GCode into Universal GCode Sender, it won't back home... Same problem...

SECOND (where is the initial problem):

When I want to abort an engrave I press the STOP button into LightBurn and exactly there is when the ALERT3 and the ERROR9 Appears. And when I want to restart the engrave, it repeats a lot of time ERROR9 (see the previous captures) and the system get stuck "busy state". I have to disconnect the USB and restart LightBurn.

I you want to make tests, you can download a free 30 days trial of lightburn there https://lightburnsoftware.com/pages/trial-version-try-before-you-buy or contact me if you want a full version.

I tried too to comment the lines you said, it is the same, the system stucks but no error and warning seen on the terminal.

I will try to connect an arduino uno with original GRBL and another serial monitor with a FT232 module on PINs 0 an 1 to see which commands are sending LightBrun to GRBL...

terjeio, 2020-07-21:
Hmm... This could be a LaserBurn issue? If LaserBurn sends CTRL-X (reset) when Stop is pressed and grbl is in the RUN state then ALARM 3 is raised.  If LaserBurn then continue to send commands while the alarm is still active this will result in error 9 beeing reported for those commands. grblHAL and grbl (I just checked the v1.1 code) both behaves like this.

Not that stopping a running program when grbl is in the RUN state then it is likely that position is lost. Here is the original grbl code that is called on a reset:

```
// Method to ready the system to reset by setting the realtime reset command and killing any
// active processes in the system. This also checks if a system reset is issued while Grbl
// is in a motion state. If so, kills the steppers and sets the system alarm to flag position
// lost, since there was an abrupt uncontrolled deceleration. Called at an interrupt level by
// realtime abort command and hard limits. So, keep to a minimum.
void mc_reset()
{
  // Only this function can set the system reset. Helps prevent multiple kill calls.
  if (bit_isfalse(sys_rt_exec_state, EXEC_RESET)) {
    system_set_exec_state_flag(EXEC_RESET);

    // Kill spindle and coolant.
    spindle_stop();
    coolant_stop();

    // Kill steppers only if in any motion state, i.e. cycle, actively holding, or homing.
    // NOTE: If steppers are kept enabled via the step idle delay setting, this also keeps
    // the steppers enabled by avoiding the go_idle call altogether, unless the motion state is
    // violated, by which, all bets are off.
    if ((sys.state & (STATE_CYCLE | STATE_HOMING | STATE_JOG)) ||
    		(sys.step_control & (STEP_CONTROL_EXECUTE_HOLD | STEP_CONTROL_EXECUTE_SYS_MOTION))) {
      if (sys.state == STATE_HOMING) { 
        if (!sys_rt_exec_alarm) {system_set_exec_alarm(EXEC_ALARM_HOMING_FAIL_RESET); }
      } else { system_set_exec_alarm(EXEC_ALARM_ABORT_CYCLE); }
      st_go_idle(); // Force kill steppers. Position has likely been lost.
    }
  }
}
```

terjeio, 2020-07-21:
This:

`�M5`

The `�` is likely CTRL-X (reset) and it is followed by M5 which will trigger error 9. It is then followed by `$X` to clear the alarm.

It _might_ be possible that the behaviour is different on a fast ARM processor vs the slower 8-bit Arduino as queueing and processing of commands will surely be different timing-wise.

When ALARM 3 is raised the serial input buffer will be cleared, it could well be that the BluePill does this before the M5 has entered the buffer and then it does not get cleared out. Notably there are USB buffers involved in the BluePill as well as it uses native USB, not a USB to serial chip, for communication. Perhaps I should check out if it is possible to flush the USB buffer as well?

This got me wondering:

> But when it ends, I don't have an home return.

When you press Stop is this due to the LightBurn "hanging" at the command before the M5 and never finishing the gcode program? If so this could be due to something else.

AxelGUILLAUMET, 2020-07-21:
I found the homing error!

The $110 and $111 are configured really high (40000) and in G0 mode, the motor will mechanically can't react to that frequency! I lowered them at 10000 and it worked perfectly.

The only problem is when I push STOP but when I wait like 30 seconds, GRBL goes back to "normality"...

I will study more the reaction with this STOP button into LightBurn....

AxelGUILLAUMET, 2020-08-10:
Dear friend.

I deactivated the USB into driver.h and I tried with real serial with a FT232 USB/Serial converter at 115200.

The "STOP" problem persists...

When I push STOP I get these errors:

[image]

And when I want to restart the process:

[image]

The machine is stuck and loose his home... Only hardware disconnecting and-reconnecting can remedy this.

terjeio, 2020-08-10:
Is the machine moving (in the Run state) when you press Stop?

AxelGUILLAUMET, 2020-08-10:
When I press STOP, the motors stops to move. 

When a engrave is going to end, the machine home perfectly and it's ready for another engrave. The problem is clearly and only when pushing the STOP button.

terjeio, 2020-08-10:
> When I press STOP, the motors stops to move.

Then the behaviour is as expected if, as I suspect, Lightburn continue to send commands. It should not.

What is your reason for pressing STOP? Normally this should only be used when something went badly wrong. If you want to terminate a job early use feed hold then stop, then you will not loose position and homed status.

closed 2022-01-27 by terjeio