terjeio/grblHAL #75: Can I use this on a Nucleo 411 board?

jappiemike, 2020-07-27:
Hi, I have a couple of Nucleo 411 boards lying around and was wondering if it's possible to put your grbl onto one. I assume I'ld use the STM32F4xx driver. I've only ever used the STM web IDE so if it's possible I'll probably bother you about how to do it.
Cheers
Mike

terjeio, 2020-07-27:
It should be possible as I have done that using the STM32F4xx driver/STM32CubeIDE. I'll commit an updated driver later this week as a few changes are required.

Is the STM web IDE the one at [mbed.com](https://os.mbed.com/docs/mbed-os/v6.2/quick-start/build-with-the-online-compiler.html)?

jappiemike, 2020-07-27:
Hi, yes it is the one at mbed.com, I just received an email saying that STM was doing some updates to their STMCUBE [(http://microelec-m.neolane.net/nl/jsp/m.jsp?c=%40IuBPmltqhamxDo%2B%2FLeYrTvVQqwHrj%2Fc4rlERDzSKysk%3D)]. I'll hold off until you've done your updates.
Cheers
Mike

hdo, 2020-07-30:
I think this Nucleo board could be a good starting point for STM32F4 based chips. There are cheap blackpill boards with this chip available from asia.

jappiemike, 2020-07-30:
Hi, I like these boards because they have the Arduino Uno pinout built in so it fits my CNC shields.
Cheers
Mike

terjeio, 2020-07-30:
@jappiemike 

You wrote:

> they have the Arduino Uno pinout built in so it fits my CNC shields.

Which shields?

There is an issue with pin assignments and the [Protoneer board](https://blog.protoneer.co.nz/arduino-cnc-shield/) I am creating a pin mapping file for. The STM32 way of assigning GPIO IRQ handlers does not allow using the same pin number on different ports for interrupts. The X and Z limits share the same pin number (C7 and A7) so only one can be configured to be interrupt capable. This means hard limits will only work for X or Z unless one is mapped to another pin not bound to an IRQ.

Are you capable of checking out 3.3V compatibility with your shields and test the driver?

terjeio, 2020-07-30:
Updated driver [published at the test branch](https://github.com/terjeio/grblHAL/tree/test/drivers/STM32F4xx). See the [ReadMe](https://github.com/terjeio/grblHAL/blob/test/drivers/STM32F4xx/README.md) for how to compile for the F411 and the Nucleo board.

__NOTE:__ Hard limits for Z-axis is not working with the current pin mapping for the Protoneer board.

@shaise : I have updated the F103 driver as well as the control signals handling was broken for your board. Can you verify both with your board?

shaise, 2020-07-30:
I'll try at the weekend.

terjeio, 2020-07-30:
> I'll try at the weekend.

No hurry, it will be a little while before I commit changes to master.

jappiemike, 2020-07-30:
I am just using a header with screw terminals, so nothing fancy so I can change the wiring to suit another pinout, I'm not sure if others can be so flexible though. and it wouldn't be possible to do homing without all 3 axis working. I made an optocoupler board for the proximity sensors for the 3 axis and E/S and probe which appears to work with both 5V and 3.3V. I did load another grbl (https://github.com/Schildkroet/GRBL-Advanced) onto a nucleo board and it seemed to work with the standard wiring.
Cheers
Mike

terjeio, 2020-07-31:
> I am just using a header with screw terminals

Ok, then you can choose from one of the four pin mappings currenly available or create a new one to suit your needs.

> and it wouldn't be possible to do homing without all 3 axis working.

Homing polls the limit pins, it is only the hard limit functionality (if enabled) that is not possible with the standard mapping.
Note that the Schildkroet port does not support hard limits at all.

> If you make your code support more than 3 axis then it will break the standard Uno pinout anyway as there are no more pins available anyway.

The driver supports up to 5 axes, the [cnc3040_map.h](https://github.com/terjeio/grblHAL/blob/test/drivers/STM32F4xx/Inc/cnc3040_map.h) mapping is ready for that. If using the pins available on the morpho headers then this map could be used as a base for a new mapping file.

terjeio, 2020-07-31:
[Pin mapping file](https://github.com/terjeio/grblHAL/blob/test/drivers/STM32F4xx/Inc/uno_map.h) for generic Uno board just added, compatible with the Schildkroet port.

jappiemike, 2020-08-02:
Thanks for the uno pin map file, I was looking at the cnc3040_map.h file and noticed several of the pins were used twice, once in the keypad (pin B15) and several times in the SD card (A3,B3,B4 and B5). Is this intentional?
Cheers 
Mike

terjeio, 2020-08-03:
@jappiemike It is a copy/paste oversight and should be removed as the board does not support the related plugins. I'll add checks for the plugins so that the compiler will fail with an error if enabled.

I have updated the i2c code to allow use of D14/D15 on the Nucleo board for the [I2C EEPROM plugin](https://github.com/terjeio/grblHAL/tree/master/plugins/eeprom). I recommend adding an EEPROM (or FRAM) to reduce flash wear. I'll commit this update later.

shaise, 2020-08-03:
Hi @terjeio ,
Looks like stm32f103 works correctly. Any particular feature you want me to test?

terjeio, 2020-08-03:
> Any particular feature you want me to test?

Control signals interrupts: cycle start, feed hold, reset and possibly safety door. Feed hold and cycle start inputs were assigned to pins not covered by a IRQ handler in the previous version. Note that the safety door input is now disabled by default in the grbl core, it has to be enabled in config.h or in the project properties by defining ENABLE_SAFETY_DOOR_INPUT_PIN.

STM32 is a bit odd in that only one given pin number regardless of which port it belongs to may be assigned to a IRQ number. There are 6 IRQ numbers available, pin 0 - 4 is allocated separate numbers, pins 5-9 one and 10-15 one. The previous version assumed pins 5-9 were used for control signals and 10-15 for limit signals (and keypad strobe).

shaise, 2020-08-04:
Hi @terjeio ,

Out of all I use only the safety door, so I can't comment on the others. However it looks like the safety door does not work at all (no indication that it is being detected) as the previous firmware it worked. I will investigate tomorrow.

shai

jappiemike, 2020-08-24:
Thanks for that, I spent a rather fruitless Sunday trying to compile the code in the mbed online compiler. Hopefully this goes easier now the .project file is includes.
Cheers
Mike

terjeio, 2020-08-25:
@jappiemike What are the issues you have with the mbed compiler?

jappiemike, 2020-08-28:
Sorry for the lat reply, I was away fixing a liquid oxygen tank. I think I had to remove all the STM library files and include just the MBED library to get rid of the compiler errors. This one still stopped me "Error: Identifier "extern __IO uint32_t uwTick" is undefined in "grblHAL411/Src/driver.c", Line: 63, Col: 9. There were also a lot of deprecated declaration warnings. I might have some free time to use your update code with the project file late this weekend.
Cheers
Mike

jappiemike, 2020-08-30:
Hello Terjeio, there appears to be a problem with the code zip download, the .xxproject files are not in the unzipped files. Am I doing something wrong, I've deleted all my old copes on my computer and downloaded it several times, all with the same result.
Cheers
Mike

terjeio, 2020-08-30:
You are using Linux? If so they are hidden files, use `ls - a` to see them. If not the something must be wrong with the program you are using to unzip as the files are in the .zip I just downloaded. Tested with Win7 and Linux:

```
ls -a grblHAL-test/drivers/STM32F4xx
.          eeprom                        Middlewares  Src
..         grbl                          .mxproject   Startup
.cproject  GRBL Driver STM32F401.launch  .project     STM32F401CCUX_FLASH.ld
Drivers    Inc                           README.md
```

Compiling with mbed: you should compile as a bare-metal project? Removing the STM library files will require a port of the driver to use mbed libraries instead.

jappiemike, 2020-08-30:
Hello again Terjeio, no I'm using macOS but I will try with my windows machine running version 10. Thanks for all the help.
Cheers
Mike

closed 2021-08-21 by terjeio