terjeio/grblHAL #76: "Stepper Error Stop" Plug-in questions

phil-barrett, 2020-07-27:
I have several people interested in using grblHAL with servo motors (not the RC hobby kind).  These have an error signal that gets asserted when something bad happens - like losing steps. I think the digital inputs can be used (or I'll create another BoB with specific inputs) for this.  I think this is essentially an EStop but don't think you can just wire-or all the servo error outputs to the EStop input pin so a plug in is needed.  I believe this is a very simple plug-in - if the error input is asserted, stop the job. 

So, is this as simple as 
`system_set_exec_alarm(Alarm_EStop);`
in the plug-in? Is there context needed?

terjeio, 2020-07-29:
> I think this is essentially an EStop

I am not sure what the correct way to handle motor issues is, a new alarm code should be assigned as well as a new control signal? There are similar issues with the Trinamic plugin. Trinamic drivers can also be configured to output overtemperature warnings, perhaps a feed hold and a message is the appropriate response to warnings?

> So, is this as simple as
system_set_exec_alarm(Alarm_EStop);
in the plug-in?

Yes? I think `mc_reset()` should be called immediately to terminate the job. Maybe better to assign new control signal(s) and handle that/those in the control interrupt handler? Recoverable faults and fatal faults?

https://github.com/terjeio/grblHAL/blob/0c0d6afcb85f9b5d22646046e9628274a6dc8463/grbl/system.c#L24-L62

> Is there context needed?

No.

phil-barrett, 2020-07-29:
OK, that makes sense to me.  It looks like there needs to be a general warning signal as well as an error signal defined. I've got a bit more to dig into.  Thanks.

einencool, 2020-08-02:
On my closed Loop Steppers I also have an alarm Output. 
I haven't wired it up, but the Wires are waiting for it to be connected.
I wanted to wire them all in a row with the "emergency Button"
But if you have another solution, that would be great...

Yesterday it could have saved my machine, but the Motor doesn't get into Alarm state, maybe I was fast enough to push the butten, or the Force was not high enough to make the Motor loosing steps...

I just crashed into my Vise with a 13mm Face Mill. And it was in the Z direction (that was not funny... :-( )
Luckily I reduced the speed with the encoder, but it was to fast for me to react...
So I got a 1mm deep hole in my Vise and a chipped 13mm Endmill...

In the next days I must check if everything is fine from the Hardware...

phil-barrett, 2020-08-02:
For the BoB you have, it would use the digital inputs so you would lose the encoder input.  Depending on your stepper alarm output you might be able to tie them all into estop. If the outputs are pulled high or open collector then no problem but if they are directly driven from a micro or other logic then when one asserts an error it would create a short to ground in the ones not asserting the error signal.  Do you have link to the steppers? I'd like to take a look.

einencool, 2020-08-03:
Hi Phil,
Thank you. This is the website with some Infos and I think there should be also the data sheet. From the mobile phone I can't see it clearly...

https://www.jmc-motor.com/product/954.html
It's the JMC IHSS60 
I can switch the software from the motor from NO to NC state. But there is also a resistor in the diagram and I don't know if I would need it.

I put 36v into the power input from the driver. 
I hope you can bring some light into the darkness for me :-)

phil-barrett, 2020-08-03:
Thanks for the link.  So the ALM output (Alarm, I believe) needs to have a pullup resistor to +5V that you provide. That what they show in the diagram. You could tie all of the ALMs from your steppers together with a single pullup resistor (10K Ohm or greater) and connect that to the estop input.  This presumes you have your Estop set up to be active low (NO).  I believe you could make it work with an NC Estop switch by using a pulldown resistor. 

Note that the way they have drawn the schematic, both PED and ALM would be pulled low by either signal from the ISSH60. (I'm not sure what PED is but I would have separate pullups for PED and ALM.) 

I'm thinking about adding more inputs in the next version so that up to 5 Alarm signals could be supported.

einencool, 2020-08-04:
The PED Signal is that the Motor arrives his Position and could give a feedback to the Software.

I‘ve got the E-Stop as a Normally Closed Switch.
So if I understand it right, I have to do the following wiring:

5V -> 10k Resistor -> Motor 1 ALM+ -> Motor 1 ALM- -> Motor 2 ALM+ - > M2 ALM- ..... -> M4 ALM- -> E-Stop Switch Pin 1 -> E-Stop Switch 2 - > E-Stop Input Teensy BOB

Is this what it should be when the Motors are all „NC“?
And when everything is wired up „NC“ then I also could detect a broken wire and have less Problems with „noise“

terjeio, 2020-08-04:
> Is this what it should be when the Motors are all „NC“?

If optocoupler saturation voltage (VCEsat) is not too high then it should work to connect like you suggest. E.g. HCPL-817 has VCEsat of typically 0.2V @ 1mA current. A 10k resistor and 5V supply will result in approx 0.4mA when four motors are connected in series. I would check the voltage at Motor 1 ALM+ to be sure it is low enough to ensure the input is securely held low.

BTW I have started working on tool length offset/probing again - hopefully I'll have something ready for testing soon. Currently I am testing manual tool change functionality with optional automatic touch off added to grblHAL itself...

einencool, 2020-08-06:
I would like to try to wire it up on Saturday.
Does it matter, in which order I place the resistor? In my thinking, when I put 5V and then the resistor, and then the Motors, it could be possible, that the Optocoplers don‘t get enough Power to work, otherwise when I go with 5V directly into the Motors and after the fourth Motor I put the Resistor. Or doesn‘t that matter?

I would like to wire all 4 Motors and my E-Stop Switch in a row as a Normally Closed state. 
Phil, can you say something to this, because I don‘t want to blow the Board or the Teensy :-)

phil-barrett, 2020-08-06:
Just got home from a few days trip so will need a little time for unpacking and such.  But can you see the actual opto? It would be helpful to know. I think Terje's guess at Vsat is probably pretty close.  I've been meaning to SPICE a model of this kind of stuff so maybe now is a good time.

phil-barrett, 2020-08-06:
OK, I spent some time working it through. First, nothing is going to be damaged with what we are talking about.  I don't believe you will need a pull-up resistor. Look at the schematic diagram.  This presumes that the opto output on your hybrid motors is on when there is no error signal and off when there is an error.  Do you have a way of testing that?  The theory of operation is that all 4 ALM output phototransistors are conducting in saturation when there is no error signal and this will light the LED in the optocoupler on the breakout board which then causes grblHAL to see a low level on the input pin.  When one of the ALM outputs changes to alarm, the chain of the phototransistors stops conducting and the LED is no longer lit up. This causes grblHAL to see a high level on the input pin.  If you measure the voltage between sig and g on the HALT input terminal on the BoB, it should read a bit less than 1 V but could be as high 1.2 V when all ALM are outputs are conducting.  If you wanted to be extra careful, you could just hook up just 1 ALM output to the HALT input and measure the voltage across the terminal.  It should be something like .2 to .4 V.

When I designed the board, I didn't expect to be seeing a series of optocouplers like that hooked up to an input but it should work as long at the phototransistors in the ALM circuit does not have a very high Vce(sat).  I took a look at the popular optos and they all spec a Vce(sat) of .2-.4 V. The resistor on the BoB is 330 which results in around 11 mA through the input opto's LED if the terminal is shorted. If the 4 phototransistors add up to 1.2V (which appears to be the worst case) in saturation then the forward current across the LED in the BoB opto will be around 7.9V mA which is sufficient for the optocoupler to work. If the voltage across the input terminal is much greater than 1.2V, you may have to reduce the resistor (R11) but let us try it first.

You can also use an NC EStop switch by adding it in series with the ALM signals.  Don't forget to use shielded wire for all these signals and ground the all the shields at your controller.

The only disadvantage of this approach is that it doesn't tell you which motor had the alarm.  Does the motor's controller have an indicator LED for that?

[image]

einencool, 2020-08-07:
Hello Phil,
Thank you for your great support, hope your trip was fine :-)

The motors have an led to indicate which has an error. So it's not necessary for me to see which one has the error. 

The steppers are working really good and I changed the Microsteps to 2000/rev and now I can go easily with 6000mm/min but I set it in the firmware to 5000mm/min but when one of my Y motors has a problem, my whole gantry would be damaged, so that's why I want to have the alarm input. Maybe it would be enough only to work with the two motors...

But where in your drawing would be the 5v supply and the 10k resistor for the motor alarm? That's where I'm hanging a little bit. Because in the drawing from the data sheet, the ALM+ is not connected, or did I miss something?

Sorry for my bad knowledge of electronics :-(

phil-barrett, 2020-08-07:
No need to apologize.

+5V is on the breakout board. You don't need to do anything for that.  The 10K ohm resistor is not needed. I misunderstood what was needed. 

The schematic in the data sheet is a little misleading - it shows two lines coming off the collector. One goes to a pullup resistor (R1) and the other is the signal.  But there is only one terminal and they are both connected to it.  The schematic is assuming that you will use the signal as a direct input to digital logic and so it needs to be pulled high. In this case we are using the signal to drive an optocoupler's LED and so we use it differently.

einencool, 2020-08-07:
So if I understood, all I have to do is to wire it like the  schematic you showed above? 
Only from the E-Stop Connector through all 4 motors and the switch and then on ground?
Could it be that simple :-)

phil-barrett, 2020-08-07:
I think it is that simple. Be sure to pay attention to the plus and minus polarities. Do you have a way of testing?

einencool, 2020-08-07:
Well, not really :-)
I wanted to test it first with only one Motor and lower the Parameter for the holding torque. Then I should be able to rotate the Motor so that it will loose his position and get an error, and then the Motor should raise the alarm state.
That‘s the way I was planing to do it...

phil-barrett, 2020-08-07:
that is a good plan. let me know how it works out

einencool, 2020-08-08:
So, now I'm ready with the tests, and yes, it really works like it should.
I have wired all 4 Motors in a Row and at last the E-Switch.
I didn't use an aditional resistor. It works great.

For testing I set a driver to very low current so that I can move it by hand and after the Position was to far from the desired one, the machine and Software set an alarm. Although the Switch is working like before....

Thanks for your help :-)

phil-barrett, 2020-08-08:
Excellent! I will redraw the schematic with the EStop switch in it, write up a description and make it available.

By the way, would you mind measuring the voltage across the Sig and G pins of the Halt terminal?  That will tell us what the Vce(sat) is (well, 4 times Vcs(sat)). I like to know that sort of thing

einencool, 2020-08-08:
I measure 1.37V on the Signal Pin.
Could that match?

phil-barrett, 2020-08-08:
OK, that's a little higher than I had hoped but perfectly fine.  It says the average Vce(sat) is 340 mV which is close to the typical max of 400 mV of most optocouplers.

einencool, 2020-08-08:
Should I add a resistor or will it no problem with these values?

phil-barrett, 2020-08-08:
An extra resistor is not needed.  If Vce(sat) is too high it would not work as the input logic would not see the low voltage it needs for a zero to be read on the input pin. It would always see an alarm state. There is a resistor on the breakout board (R11) that is limiting the current already.

einencool, 2020-08-08:
Alright, thank you, I'll raise a new issue, because I can't compile the test branch, I always get this error, thought it would be because there was the "CNC-Boosterpack" activated...

phil-barrett, 2020-08-08:
Yes, I am following that thread.  Once I get my ethernet testing done, I will look into it.  It's probably something pretty simple but our good friend Mr Arduino does things in his own peculiar way...

terjeio, 2020-08-15:
@phil-barrett I have added `motor_fault` and `motor_warning` to the control_signals_t union in the test branch. Code support needs to be added to `control_interrupt_handler()` in system.h and to drivers. `motor_fault` should be handled like an e-stop, not sure how to handle `motor_warning`.

https://github.com/terjeio/grblHAL/blob/40c92d88ad1db8adea4101bea6bce2cb065eafd7/grbl/system.h#L131-L150

einencool, 2020-08-17:
Could this also be mapped to an Endstop Input? Maybe for the A or B-Axis?
It‘s only a thought, because I‘ve wired all the Motors with the E-Stop-Button.

terjeio, 2020-08-18:
> Could this also be mapped to an Endstop Input?

Not sure I understand. Endstop input = limit switch input? If so then these already has a per-axis input and is handled elsewhere (in limit.c).

In general mapping is done at the driver level, and what is connected is up to the user. The `motor_fault` signal should have the same functionality as the e-stop but issue a different message and/or alarm. At the driver level there could be separate inputs for each axis triggering the same signal/handling in the core, if so more detailed messages/info could be provided. Same goes for the `motor_warning` signal.

Perhaps the `motor_warning`  should result in a feed hold and possibly disabling motors allowing a restart of any ongoing job after the issue is corrected. E.g. the Trinamic stepper drivers has an overtemperature pre warning output that could be handled like this.

It could be argued that the `motor_fault` and `motor_warning` signals should be split into per axis signals - but IMO that is not needed by the core as I cannot see that the core should act differently (and in a standardized way) according to which axis the signal originated from.

einencool, 2020-08-18:
Yes, I meant the Limit switch (Sorry, I didn‘t know that Endstop was not the right wording for it *g*)

Well, I have wired it up all to the E-Stop Signal, but maybe some other People want to connect them as Normally Open Inputs, and then it would be better to use another Input Signal than the E-Stop...

terjeio, 2020-08-18:
> Yes, I meant the Limit switch (Sorry, I didn‘t know that Endstop was not the right wording for it g)

Limit switch, endstop and even home switch has all the same meaning - at least in the grbl context.

> Well, I have wired it up all to the E-Stop Signal, ...

Now I do not understand again, you have wired the endstops/limit switches to the e-stop signal input in addition to the motor fault signals?

einencool, 2020-08-18:
Oh no :-)

I‘ll give it a new try *g*

In my machine I have wired the „Motor Alarm signals“ and the „E-Stop Button“ as Normally Closed Signals to the „E-Stop Input“.
I‘ve done this, because I didn‘t have another free Input Signal Connector only for the Motors.

——

That‘s why I asked, if it would be possible, to wire the „Motor Alarm Signals“ to the maybe „Limit Switch Input Axis „B““. So that I would have the Motor Alarm Signal on the Limit Switch Input and the „E-Stop Button“ on the „E-Stop  Input“

I hope I‘ve explained it better now *g*

terjeio, 2020-08-18:
Ok, I understand now. Yes, this would be possible. This as the input routing is done in the driver.

But not yet, @phil-barrett said above he would "dig into it" - I do not know if he has done so already. I have just done the very first step by assigning bits in the data structure.

S2000Stefan, 2021-01-24:
I would also like to convert my milling machine to closed loop steppers. And I would like to connect the alarm output of the drivers to the A limit pin (PB15) like @einencool. 
My question now is @terjeio, since you added "Added handling of motor fault signal to core" on 2 january, is there already a way to connect the motor driver fault signal? 

It would be important if an error occurs that the machine stops immediately. Since the motor driver switches the motor currentless when there is a fault and the other motors simply continue to run.  ;-)

Since my driver (STM32F303) also does not have an e-stop to which I could connect the signal. I do use the reset pin as an e-stop by connecting it to ground, but that is not a real e-stop in the sense of the word.

terjeio, 2021-01-24:
> is there already a way to connect the motor driver fault signal?

Not yet, driver code has to be changed for that and that would be the easy part. Reassigning pins and reflect that back into senders settings UI is going to be the complicated bit. As a start I could add, for now, a compile time option that reassigns A limit input to motor fault. But signal inversion and pullup disable would still be for the A limit input in the UI until that can somehow be sorted out.

einencool, 2021-01-24:
Good morning @S2000Stefan 
There is a little mistake :-)
I've wired my alarms into the e-stop circuit and when there is a issue or I press the e-stop button, then all motors are stopped immediately.
I have also a extra switch for the motors because they make a terrible high frequency noise and when I don't use the machine for a while but still working in that area, I switch off the motors and so the alarm will directly raise.

Wiring it to the A-limit pin was a goal I would appreciate, because then I can also start the machine/software without turning on the motors.

@phil-barrett mentioned this a while ago, but sadly I didn't hear about it any more. :-)

S2000Stefan, 2021-01-24:
> As a start I could add, for now, a compile time option that reassigns A limit input to motor fault. But signal inversion and pullup disable would still be for the A limit input in the UI until that can somehow be sorted out.

Would be a quick and easy solution, not very advantageous but the person who uses it knows what he has connected to the A-limit pin. 
Whereas, I've been sitting here all morning trying to connect to my motor driver with my tinker cables and can't manage it. I'll have to get an original cable. :-)
And I want to connect the alarm output to my STM32? ;-)

> There is a little mistake :-)
I've wired my alarms into the e-stop circuit and when there is a issue or I press the e-stop button, then all motors are stopped immediately.

@einencool, sorry, I must have misunderstood something. ;-)

einencool, 2021-01-24:
> Whereas, I've been sitting here all morning trying to connect to my motor driver with my tinker cables and can't manage it. I'll have to get an original cable. :-)
> And I want to connect the alarm output to my STM32? ;-)
> 

What type of motors do you have, and what do you want to connect?

I have JMC IHSS CL Steppers with an HISU. These I have connected with a modified network cable.

S2000Stefan, 2021-01-24:
I was being ironic. I am not even able to connect the motor driver output stage to my computer and then I want to connect the alarm line to my controller. 

> I have JMC IHSS CL Steppers with an HISU
What type of motors do you have, and what do you want to connect?

You've got some fine stepper motors there. I treated myself to Nema24 closed loop steppers with CL57T motor drivers. The driver only needs to be connected to the computer via RS232, but I don't have the right adapter for RS232 to USB.
But it has already been ordered. :-)

closed 2022-01-27 by terjeio