terjeio/grblHAL #81: Support for plasma Thc

Yantec, 2020-08-09:
Hello 
Can you please add support for plasma Thc ?

terjeio, 2020-08-09:
That is quite a generic request. Which THC do you have?

I have to admit I have already bought some components for a DIY design, but that was a couple of years ago - too many ideas too little time...

Yantec, 2020-08-10:
I am building a plasma cutting table and am now looking for a controller. Do not have a Thc but hope for a grbl based Thc
I have a router that I run with grbl and am happy with it
I think one solution is to have a DIY voltage divider. Which feeds an analog input on the grbl card and all logic is in the plugin module. Something like LinuxCnc

terjeio, 2020-08-10:
> I think one solution is to have a DIY voltage divider.

If it only was that simple... The arc voltage is around 100V and there are spikes all over the place. So in practice the divider needs to be followed by a filter and then some kind of galvanic isolation before the signal is fed to the AD. And there needs to be signals for starting the arc and arc good? My memory is a bit dim regarding this. Anyhow, it is not something added in a whim.

A good starting point would thus be to use a proven design, commercial or existing DIY.

If you want to make a new DIY design then IL300 is the optocoupler I bought for the one that I have in mind...

terjeio, 2020-08-13:
Some info here: http://linuxcnc.org/docs/devel/html/plasma/plasma-cnc-primer.html

closed 2022-01-27 by terjeio