terjeio/grblHAL #82: Two steppers on one axis

raumneun, 2020-08-16:
Hi there.
First of all, thanks for the great work you did with grblHAL. I've been a huge grbl fan but was always annoyed by the limitations of an Arduino and dreamed of an hardware abstraction layer. Thanks for making this dream come true :)

For my next mill, I am going to have two steppers on one of the axes. Because of the breakout board I'm using (Phil's nice design for the Teensy 4.1), I can't really hook up two motors on the same output pins because my drivers draw quite some current for their internal optocouplers.

So I was wondering, if I could simply mirror axes by using the output pin for, let's say the B axis, by a simple change in the driver:

```
inline static void set_step_outputs (axes_signals_t step_outbits)
{
    step_outbits.value ^= settings.steppers.step_invert.mask;

    DIGITAL_OUT(stepX, step_outbits.x);
    DIGITAL_OUT(stepY, step_outbits.y);
    DIGITAL_OUT(stepZ, step_outbits.z);
    DIGITAL_OUT(stepB, step_outbits.x);
#ifdef A_AXIS
    DIGITAL_OUT(stepA, step_outbits.a);
#endif
#ifdef B_AXIS
    DIGITAL_OUT(stepB, step_outbits.b);
#endif
}
```
And of course the same for direction and enable.
Of course I'd have to make sure that the correct pins are defined prior to this function, it's more a question if this idea goes in the right direction.
In principle, one could implement this as compile time configuration options, like Mirror X to B or so.

terjeio, 2020-08-16:
> So I was wondering, if I could simply mirror axes by using the output pin for, let's say the B axis, by a simple change in driver.c 

Yes, it is simple as that.

> In principle, one could implement this as compile time configuration options, like Mirror X to B or so.

True, but a bit more complicated if auto squaring is to be enabled. The SAM3X8E driver has an attempt at conding for that, for now untested in a real machine. The iMRXT1062 is a candidate for adding auto squaring so if this is something you need and can test that would be nice.

raumneun, 2020-08-16:
Thanks for your quick answer! I'll try that and see where it goes.
I heard of autosquaring before, but I have to say I'm not a fan ;)
For starters, if the machine is very rigid, very high forces will act for offsets of fractions of millimeters. So in case something goes wrong with the limit switches or with the software, things are bound do go sideways. Probably the ball screws or the nuts will break.
Second, squaring the machine and keeping it square should imho be implemented mechanically. If things are not aligned properly, forcing it to be so inevitably leads to a pre-load of the system, making the rigidity different for different directions of travel. Surface finishes will be very different for different sides of the workpiece.

The only reason how a machine could go "unsquared" is by crahing or losing steps. The latter can be avoided by coupling the ball screws mechanically, e.g. with a not so tight timing belt.

This is of course only my opinion and other's might think differently about that, but from my perspective as an mechanical engineer, mechanical problems should not be fixed in software :)
But that also really much depends on the construction of the machine.

raumneun, 2020-08-17:
What I forgot to say: probably the community is not yet very large and hardware diversity probably makes it harder to find testing candidates. If there is such a feature, I will see what I can do to test things to at least contribute a little.
I would also code a few lines, but my capabilities are somewhat limited to barely acceptable dirty hacks :)

terjeio, 2020-08-17:
> If there is such a feature, I will see what I can do to test things to at least contribute a little.

Thanks for that - it is impossible for me to test every driver/every configuration in a real machine.

> I would also code a few lines, but my capabilities are somewhat limited to barely acceptable dirty hacks :)

I will add motor ganging and auto squaring to the iMRXT1062 driver later, great if you can hack in ganging yourself until then.
Note that A can be ganged/linked to Y by solder links on the underside of Phil's board as well.

> I heard of autosquaring before, but I have to say I'm not a fan ;)

Well, there is many different types of CNC machines out there - some rather flimsy, some extremely rigid. So autosquaring has use-cases for sure - and it would be a good option to have available. Compile-time configurable...

phil-barrett, 2020-08-19:
Yes, autosquaring is something that is important.  I will be getting a mill that can use it so will be happy test it.  Probably not very soon though.

By the way, you can link the A axis to the Y axis via a solder jumper on the bottom of the board.  This connects before the drivers so it can drive two steppers from Y.  The procedure is described in the manual on page 12 (PCB Level Configuration/Link A Axis to Y).

raumneun, 2020-08-20:
Thanks guys for your answers and your support, I appreciate the time you put into all this.
For me it'll take some time, too, the parts for the machine are not even close to finished, so every test will be either on my old mill (with only one stepper per axis) or verified by scopes or so. First tests will be possible soon, though, electronics is mostly finished.

I actually saw your option to do that in hardware, which I personally think is the most reliable way to do that, however: the way my mill will be in the room and how I stand with respect to the machine, the axis moving the gantry (with two steppers) has to be X. Everything else would confuse me forever :)
Also, I'll probably add a rotational axis A later, so I don't want to block this port (which does not matter terribly, I could still rename it in grblHAL, but if I change it in code anyway, there is no real reason not to do that in the first place I guess).

phil-barrett, 2020-08-20:
Makes sense.  Please feel free to reach out to me if you have questions and I'd love to see photos when you get your machine going.

Phil

terjeio, 2020-08-21:
I have added options for ganging and auto-squaring in the [test branch](https://github.com/terjeio/grblHAL/tree/test):

https://github.com/terjeio/grblHAL/blob/a5b4abe7eda0c6520599e9da0d5652b2ea6b7e95/drivers/IMXRT1062/main/T41U5XBB_map.h#L26-L36

Not tested in a real machine! Please report back test results.

@phil-barrett I have added a initial board mapping file for your board, it should be expanded and complemented with a .c file with the same name for the extra GPIO options available? I think that an optional `board_init()` function call should be added to driver.c and implemented in the board .c file in order to keep driver.c generic.

terjeio, 2020-09-07:
I have now improved (fixed?) autosquaring and tested it with my new CNC machine simulator and the iMRXT1062 driver. Nice if someone can confirm it working in a real machine.

raumneun, 2020-09-08:
Thanks, I'll definitely do some testing when the machine is finished. My old one has just one ball screw per axis, so it may take some time to get the new machine ready:

[image]

Note the two drilled holes through the sides of the gantry where the two ball screws for the x axis will be fed through. The gantry itself serves as the ball screw nut housing.
With this machine, I could do some real world testing. Although I'll have to check that there is no possible damage due to the concerns I posted above. The aluminium parts have a thickness of 40mm 💪 :)

terjeio, 2020-09-08:
@raumneun That is a heavy duty machine, agree it is not suited for simple auto-squaring.

What is lacking from most (all?) implementations is perhaps firmware calibration. High precision auto-squaring need high precision limit sensors that is either mechanically calibrated or calibrated in firmware (by adding an offset to one of the motors on homing). Also, the amount of travel allowed for auto-squaring should be limited and if exceeded the homing sequence should be aborted.

raumneun, 2020-10-20:
I just wanted to report success with the ganging feature. I simply enabled ganging for my board (Phils T41U5XBB) using ```#define X_GANGED 1``` and verified using a two-channel scope that the signals are indeed a copy of one another.
At least for the teensy, I can report that there is no significant delay (i.e. the delay between the channels is within the steepness of the rising edge) between them. This is not really unexpected, but I guess it is nice to know. I could do some more testing of the ganging feature if you are interested (i.e. checking the other axes). Otherwise I guess this can be closed? Thanks for the great work! So far, I am really pleased how well everything works and how easy and straightforward it is to use it.

phil-barrett, 2020-10-20:
Glad to hear of your success!

I would be proud to add your machine to [the showcase](https://www.grbl.org/projects).  It's definitely unique. Let me know.

terjeio, 2020-10-20:
Great!

> At least for the teensy, I can report that there is no significant delay (i.e. the delay between the channels is within the steepness of the rising edge) between them. 

FYI: The skew between pulses should be in the sub 100ns range for drivers that supports mixed ports for the pulse outputs. Some drivers even none if all pins are written to at once. The Teensy has a 1.66ns clock cycle - the skew is a few clock cycles + the delay incurred writing to the slower(?) peripheral bus. Less than 10ns? So yes, insignificant.

>  I could do some more testing of the ganging feature if you are interested (i.e. checking the other axes).

I am pretty sure the ganging is working for all axes as that is simple to implement, the auto squaring feature is another story - a lot more complicated to handle during homing. So auto-squaring is more important to test - but your heavy-duty machine is not well suited for that as discussed above. Or are you willing to try auto squaring as well? If so then the next build has some changes to how auto squaring is handled so better wait until that is available.

> Otherwise I guess this can be closed?

If you want to test some more you can keep this open and report here, if not you may close it.

raumneun, 2020-10-20:
As mentioned above, I did not expect it to be any different, but I had the scope connected anyway to do some tests, so I figured I might as well look at that to see if there is additional lag.

Regarding auto-squaring: I'd love to test it, but I'm afraid it will break something... Fixing the gantry at one end, I can barely move it by 0.05-0.1 mm at the other end with my arms. I.e. for displacements in the order of 0.2 mm (which could easily happen while testing auto-squaring) I'd probably break my spindles. So I am not to keen to do it ;) I'll close it here. Thanks for your help!

closed 2020-10-20 by raumneun

einencool, 2020-10-26:
> I am pretty sure the ganging is working for all axes as that is simple to implement, the auto squaring feature is another story - a lot more complicated to handle during homing. So auto-squaring is [...]

Hello @terjeio ,
about the Auto Squaring Option. Is it working with your new version? Is it somehow configurable, so that the Homing Process will be stopped, if the second Endstop is not checked within 0,x mm?
I have around 0,07mm that I would like to sqaure with homing, I'm having some good Endstops in the Mail, but I want to be sure, that the Software will stop, if the Endstop doesn't hit within a few steps...

terjeio, 2020-10-26:
@einencool - there are compile time settings for the following parameters. They should be moved to $-settings too and I am contemplating adding a setting per axis for switch offset (useful when the switches are hard to calibrate?).

```
// To prevent the homing cycle from racking the dual axis, when one limit triggers before the
// other due to switch failure or noise, the homing cycle will automatically abort if the second
// motor's limit switch does not trigger within the three distance parameters defined below.
// Axis length percent will automatically compute a fail distance as a percentage of the max
// travel of the other non-dual axis, i.e. if dual axis select is X_AXIS at 5.0%, then the fail
// distance will be computed as 5.0% of y-axis max travel. Fail distance max and min are the
// limits of how far or little a valid fail distance is.
#define DUAL_AXIS_HOMING_FAIL_AXIS_LENGTH_PERCENT  5.0f  // Float (percent)
#define DUAL_AXIS_HOMING_FAIL_DISTANCE_MAX  25.0f  // Float (mm)
#define DUAL_AXIS_HOMING_FAIL_DISTANCE_MIN  2.5f // Float (mm)
```

einencool, 2020-10-27:
Ah Ok,
So I can adjust the parameters.
Let's say my axis has 700mm of travel, then I can say for your 3 settings :
0.05% = 0.35mm
Max 0.35mm
Min 0.00mm

Would that be right?

terjeio, 2020-10-27:
> Would that be right?

I think so, I will run test it with my simulator (I do not have a machine with dual motors).

einencool, 2020-10-27:
That sounds good,
Then I‘ll break my brain, how to fix the Endstops to my machine, that they are rigid and absolut repeatable.
I think it will take some weeks till I find the time for the big step...

Thank you :-)

einencool, 2020-10-31:
Good Morning @terjeio ,
Were you able to make some tests for the dual Motors with dual Endstops in your Simulator?
Yesterday the Endstops have arrived and in the next days I‘m going to build some Mounting adapters for them.

Maybe you have some time for testing it in the next 2 weeks, so then I‘ll do some other stuff instead :-)
So take your time and test it with your simulator.

When I build it into the machine I‘ll disconnect the belts from the gantry so that the leadscrew will not turn.
Maybe you have an idea how i can check it on the real machine, without having the gantry connected.

Would it be possible to open this issue again, so that the topic is easier to find?

terjeio, 2020-11-03:
Hello @einencool - this just to tell you that you are not forgotten. I am currently deep into spindle sync coding and want to get this out of the way before I come back to this.

> Would it be possible to open this issue again, so that the topic is easier to find?

I'll post my test results here, if any further discussion is then needed it will be better to open a new issue as this is resolved. Or even better, open a new one now.

einencool, 2020-11-03:
I reopened the old issue #45  I closed a few weeks ago, because I thought that this issue would be enough for the same problems :-)