terjeio/grblHAL #85: Stm32f103 target - debug works, release fails

darkfibre-nl, 2020-09-03:
Hi!

I've given compiling the driver for stm32f103 a try.
'Debug' compiles flawless:
(22:07:57 Build Finished. 0 errors, 0 warnings. (took 12s.304ms)

'Release' seems to have a bit of a problem:
- https://pastebin.com/gyYxx6wp

and after adding a few path(s) so 'grbl.h:' can be found, still 'Computers-says-no'.
- https://pastebin.com/VmWy6Y8Q

This is over my head, perhaps is there something wrong with the 'release' configuration?

terjeio, 2020-09-03:
> This is over my head, perhaps is there something wrong with the 'release' configuration?

It surely is, thanks for reporting. I will add the missing paths when I merge test with master, maybe tomorrow? Until then you may use the debug build - there is no extra code added to that.

darkfibre-nl, 2020-09-04:
Thanks for the info! I'll use debug for now!

terjeio, 2020-09-04:
Release build should now be ok, FYI test branch is now merged with master.