terjeio/grblHAL #87: pull requests against the test branch?

phil-barrett, 2020-09-07:
So, I want to add my ioports code but am not sure how to do a pull request against the library version in test since the installed library doesn't exactly match the repository.  There are basically 2 files and, I think, 2 changes to driver.c. What's the best way to go?

HuubBuis, 2020-09-11:
I am to struggling using git. I first merge the current branch to my version and than do a pull request. This way, only my changes need to be verified.

phil-barrett, 2020-09-11:
We're holding off on this pull because there are other changes coming that will affect it.

And, yes, I find git to be very confusing.  I cut my teeth on check-out, check-in, branch and merge request style code management systems.

HuubBuis, 2020-09-11:
I still use Microsoft Source Safe as version control for all my projects. It has often rescued me when I screwed up. However I think Git is the way to go because it is used on almost all IDE's I use, so it must be decent. I suspect my lack of git knowledge and refusing to use the git command line, that is the real problem.

closed 2020-10-04 by phil-barrett

terjeio, 2020-10-08:
I have added a [howto to the Wiki](https://github.com/terjeio/grblHAL/wiki/Using-Eclipse-for-Arduino-development) for using Eclipse for development that should simplify pull requests. I have not testet creating pull requests yet, but I believe it should work.

The Eclipse editor is infinitely better than the Arduino IDE editor, just an example:

[image]

HuubBuis, 2020-10-08:
This is really needed to get going. To uniform format the code (ctrl K D  i think) everyone should use the same settings (tabs, indentation). this way git shows just what is changed!

The symbolic link tool is a great Idea. It makes maintaining the git files a lot easier. I think I can use it for other projects as well.

I have tried setting up eclipse and teensy. Seems there is a lot more involved like creating a build target and setting up the toolchain.