terjeio/grblHAL #88: Maintenance counter

raumneun, 2020-09-08:
It would be great, for maintenance purposes, to track the total distance travelled and the total machine runtime on the controller.
Do you think this is possible? If so, could you point me in the right direction where to start? I would like to give it a go.
I guess the easiest way is to integrate the number of steps times the current steps/mm value and store this in eeprom. Obviously not at every step so it won't give up after some seconds ;)
Any suggestions?
I'd love to have - just like in a car - 2 counter per axis, where one could be manually reset after maintenance. The sender may or may not use this information to notify about upcoming maintenance intervals.

terjeio, 2020-09-08:
Add counters (64-bit) in driver.c and increment them in `stepperPulseStart()`. Hook a function into the `hal.execute_realtime` chain and perform timing stuff there and write out the data when the machine is in idle state. I would go for FRAM, not EEPROM - this as FRAM can sustain a lot more write-cycles and needs no delay as part of a write.

In fact this could be written as a plugin, by also chaining the `hal.stepper_pulse_start` entry point. Only a init call to the plugin is then needed to set it up. `hal.stepper_pulse_start` needs to be made chainable in driver.c for that.

"Chaining" function calls on HAL entry points is in principle simple, save the current entry point (function pointer) redirect it to to the one you want called and call the original either before or after your own code. It is a bit more complicated when the driver changes a entry point on the fly, this is true for `hal.stepper_pulse_start`. This would require the chained function to be inserted first in the chain and the `stepperPulseStart()` variants to call it.

terjeio, 2020-09-08:
Some quick and dirty code suggenstions for a plugin:

```
#ifndef _ODOMETER_H_
#define _ODOMETER_H_

typedef struct {
    uint64_t distance;
    uint64_t time; //??
} odometer_h;

typedef struct {
    uint64_t time;
    odometer_h odometer[N_AXIS];
} odometers_t;

void odometer_init();

#endif
```
and 

```

bool odometer_changed = false;
uint32_t odometers_address;

static void stepperPulseStart (stepper_t *stepper)
{
    odometer_changed = true;

    if(stepper->step_outbits.x)
        driver_settings.odometers.odometer[X_AXIS]++;

    if(stepper->step_outbits.y)
        driver_settings.odometers.odometer[Y_AXIS]++;

    if(stepper->step_outbits.z)
        driver_settings.odometers.odometer[Z_AXIS]++;
}

void odometer_execute_realtime (uint_fast16_t state)
{
    uint32_t ms;

    if(state == STATE_CYCLE)
        ms = millis(); // millis() should called be over a new HAL entry point?

    if(state == STATE_IDLE)
    {
        if(odometer_changed) {
            odometer_changed = false;
            ms = millis() - ms;
            driver_settings.odometers.time += ms;
            hal.eeprom.memcpy_to_with_checksum(odometers_address, (uint8_t *)&driver_settings.odometers, sizeof(odometers_t));
            ms = 0;
        }
    }
}

void odometer_init()
{
    odometers_address = hal.eeprom.driver_area.address + (&driver_settings.odometers - &driver_settings);
    // add some $ commands for displaying/resetting
    // hook odometer_execute_realtime into hal.execute_realtime
    hal.stepper_pulse_start = stepperPulseStart; // this needs to be first in the chain (called last) - drivec.c must be mofied for that
}
```

MARIOBASZ, 2020-09-08:
Interesting idea.
Perhaps it is useful to time hours of work of the tool.
I have never programmed in C, maybe I can learn and contribute.

HuubBuis, 2020-09-08:
I have checked the data sheet of an SAMD21. The number of flash write cycles is about 25k (table 36-40). For an Arduino Mega or Uno it is about 10k.
Because the processor power can be disconnected without a warning, the safest way is to increase a counter in flash memory every xxxxx mm travel knowing that partial counts will be lost when the processor is powered off. To increase the amount of writes, more than one (8 bit) counter in flash can be used (increase the counter with the lowest value, max value = 255).  The sum of all counters, is the total travel.

terjeio, 2020-09-08:
> I have checked the data sheet of an SAMD21. The number of flash write cycles is about 25k (table 36-40). For an Arduino Mega or Uno it is about 10k.

This is why is better with external storage, preferably FRAM. The eeprom plugin can be used for that.

> Because the processor power can be disconnected without a warning, the safest way is to increase a counter in flash memory every xxxxx mm travel knowing that partial counts will be lost when the [...]

With flash or EEPROM this is of no use as before the write is performed the sector(s) written to are erased (set to 0xFF). If FRAM is used then this is a valid point, thanks for that.

terjeio, 2020-09-10:
[Plugin code](https://github.com/terjeio/grblHAL/tree/master/plugins/odometer) added for review.

Not sure about everthing... I am using 64 bit integers for logging. Could have used floats but this makes it impossible to write only changed bytes to storage in a meaningful way(?). Currently all data is written out with a checksum, outside of the grblHAL EEPROM buffering scheme which is enabled by default.

Data is written out on state changes from motion to no motion. Should it be written out periodically as well (during motion)? If so IMO FRAM is required as writing to EEPROM incurs a 5ms delay for each page/sector written.

Note that the plugin is not usable with flash storage and will refuse to install itself if detected.

Another issue is where to place the data in storage, currently they are added at the back of the grbl area (the first 1K) or at the back of the driver settings are if this is defined. If the latter is true any increase in length of this will overwrite the odometer counters... Perhaps better to add them to the end of the EEPROM/FRAM?

Two new $-commands:

`$ODOMETERS=RST` - resets all counters to 0.

`$ODOMETERS` - report current values, e.g:

```
[MSG: ODOMETERHRS 0:18]
[MSG: ODOMETERX 4.2]
[MSG: ODOMETERY 3.4]
[MSG: ODOMETERZ 0.0]
```
Time in h:mm and distances in meters.

terjeio, 2020-09-10:
@MARIOBASZ : IMO timing tool use will fast become tricky. If using more than one tool then a tool table with separate counters for each tool is required and a strict tool change regime must be followed. And logging should only take place for cutting moves. Feel free to give it a try, the basic stuff is already in the plugin just commited.

raumneun, 2020-09-10:
Wow you are incredibly fast. I'll have a look at the code and maybe do some dry testing (ie. without the machine attached) sometime during the weekend. Between family and being in my last months if my phd, my timescales are like two orders of magnitude longer than yours :D

terjeio, 2020-09-10:
> maybe do some dry testing (ie. without the machine attached) sometime during the weekend.

Not that fast, I have not committed any drivers with support for this plugin yet. I have testet a bit with a Teensy 4.1 with FRAM for storage, will update this driver later. It will take some time to add driver support for others, please add hints for which here.

terjeio, 2020-10-24:
[Odometer plugin](https://github.com/terjeio/grblHAL/tree/master/plugins/odometer) is now ready for testing for STM32, iMXRT1062 and MSP432 drivers. I'll add support for other drivers on request or when I get around to update them.

Logged data is now stored as float, not uint64_t, and at the end of the NVS area - adding/removing other plugins will not overwrite the log.

closed 2022-01-27 by terjeio