terjeio/grblHAL #89: STM32F103/411 EEPROM support

darkfibre-nl, 2020-09-08:
Hello!

Am I correct in assuming that if I want to use an I2C EEPROM to store, I would have to use PB10/11 as SCL/SDA pins (I2C2) for the stm32f103? For the F411 I found PB6/7 (I2C1). These pins appears to be used, according to their respective 'i2c.c' source files.

Can you give some guidance on how to configure the source so the eeprom is used?

Thx in advance for the info.

terjeio, 2020-09-09:
Copy the eeprom plugin code to the driver eeprom folder, then uncomment `EEPROM_ENABLE` in Inc/my_machine.h. Change the #define value to 2 if EEPROM is > 2K bytes. Pin numbers are as you have found out already. Which pin numbers are used for F4xx depends on what board you are compiling for:

```
   #ifdef NUCLEO_F411
    #define I2C_PORT 1
  #else
    #define I2C_PORT 2
  #endif
```

Don't forget pullup resistors if there are none on the board.

darkfibre-nl, 2020-09-10:
Thanks for your quick reply Terje! 

I followed the instructions, but unfortunately no success so far.
On both Linux and Windows I fail to get a successful compilation with EEPROM support.
https://pastebin.com/kKvEstB7

These are the step I followed (on windows, linux only differed in using 'git clone url 'to obtain the source):
a. Downloaded & unzipped the most recent version of grblHAL 'https://codeload.github.com/terjeio/grblHAL/zip/master'.
b. Copied the contents of 'grblHAL\grbl\' folder to 'grblHAL\drivers\STM32F1xx\grbl\'.
c. Copied the contents of 'grblHAL\plugins\eeprom' to 'grblHAL\drivers\STM32F1xx\eeprom'.
d. Uncommented '#define EEPROM_ENABLE 1'  in 'my_machine.h'.
(no other pin mapping file enabled, so 'generic_map.h' is used ).
e. Attempted to compile 'debug' & 'release', compilation log of 'release' can be found through the pastebin link above.

Note: Everything compiles without a hitch with EEPROM support not enabled.

terjeio, 2020-09-10:
Ouch - Src/eeprom.c is no longer used, delete it. It is left from a version pre common plugin. I'll commit a fix later.

darkfibre-nl, 2020-09-11:
That helped! It compiles! Settings appear to be retained as well, though I don't know where (flash/EEPROM).

Is there any way to query the firmware to see which plug-in(s) are enabled?

terjeio, 2020-09-11:
> Settings appear to be retained as well, though I don't know where (flash/EEPROM).

For now for checking you can alter a setting or set $I to some string and then flash without the plugin and check if it is lost. Flash again with the plugin enabled and it should reappear.

> Is there any way to query the firmware to see which plug-in(s) are enabled?

Not in a standardized way, some are obvious as they manifests themselves e.g. by providing additional settings, some not.

For nonvolatile storage I will add EEPROM or FLASH to the $I response (in NEWOPT) depending on what is detected:

[image]

darkfibre-nl, 2020-09-17:
Success! The settings are indeed stored in the EEPROM.

Thanks for your help! 
(and grblHAL in general as a project. Sonny appears to be caught up in life, work & family, so your reworking and extending/improving grbl is very much appreciated! )

terjeio, 2020-09-30:
Next build will list installed plugin versions as well. Thanks for reminding me.

[image]

closed 2020-09-30 by terjeio

S2000Stefan, 2020-10-05:
My eeprom board arrived today and I need your help.
This is it:
[image]
[image]
I have now connected Vcc Gnd and SCL SDA to the STM32F103. Vcc with 3.3v, Gnd with Gnd, SCL with PB10 and SDA with PB11. Copy the plugin files into the directory as described above and compile with EEPROM support with success.
If I now change a setting it works and after reload everything stays as it is. Only if i disconnect the controller from the usb and plug it in again, everything is back to default.
Am I doing something wrong, or is the board not suitable as eeprom?
In my_maschine.h i have comments and sets the following:
#define EEPROM_ENABLE      2 // I2C EEPROM support. Set to 1 for 24LC16(2K), 2 for larger sizes. Requires eeprom plugin.
#define EEPROM_IS_FRAM     1 // Uncomment when EEPROM is enabled and chip is FRAM, this to remove write delay.

phil-barrett, 2020-10-05:
Do you have pull ups on the SDA and SCL line?  My board has 2.2K resistors that pull SDA and SCL to +3.3V.

S2000Stefan, 2020-10-05:
Yes, you are right it seems to be 10k resistors, at least that is how I measured it.
The 10k resistors are in the VCC line, connected to SCL and SDA. Is this correct or do you need to add resistors in the line from the controller (PB10/11) to the SCL/SDA pin?
Unfortunately I am not an electronics engineer. :)

phil-barrett, 2020-10-05:
Here is the portion of the schematic from my Teensy 4.1 board.
[image]

The best way is to remove the 10K surface mount resistors (by holding a hot soldering iron so it heats both ends) and replace with 2.2K surface mount chips - looks like an 805 package. Don't remove the other chip (it's a bypass capacitor). 

However, you could just add two 2700 ohm resistors in parallel to the 10K resistors.  The would result in 2126 ohms which is close enough to 2.2K.

terjeio, 2020-10-05:
10k is a bit on the high side, but should work as IIRC I run the bus at a conservative 400 KHz. I have a 32K EEPROM breakout, give me a bit of time and I'll check if that works with a F103.

I have checked the datasheet, the default I2C address is the same as for the one I have used so that should not be a problem unless you have connected any of the A0 - A2 adress lines to 3.3V.

> However, you could just add two 2700 ohm resistors in parallel to the 10K resistors. The would result in 2126 ohms which is close enough to 2.2K.

Easier to do like this, the value is not critical. A value between 2K7 - 5K6 will be ok. I have used pullups in the range 2K2 - 6K8 for my projects. But I am not so sure this is really needed. Wait until I have tested the _eeprom_24AAxxx.c_ plugin with the F103?

phil-barrett, 2020-10-05:
Good point but 10K is very much on the high side -  10K is usually for 100KHz, 5V.  While you could well be right, it's easy enough to test without mangling the board.  And it won't damage anything if just adding a parallel 2.7K ohm resistor to each 10K.

terjeio, 2020-10-05:
Oops, seems that the 2 byte address protocol used by > 2K EEPROMS is not supported by my F103 I2C code. I'll have to update that...

terjeio, 2020-10-05:
@S2000Stefan 

Can you change `I2C_MEMADD_SIZE_8BIT` to `I2C_MEMADD_SIZE_16BIT` in the two lines below as a temporary workaround? This works with my 32K EEPROM.

> https://github.com/terjeio/grblHAL/blob/a08601881ebdb6c519fbfc228e11d5f65dd4f755/drivers/STM32F1xx/Src/i2c.c#L102-L105

I am not yet ready to commit the latest version as this has a lot of changes to the non-volatile storage handling that needs testing.

S2000Stefan, 2020-10-05:
Sorry for the late reply. We had a men's games evening at my place today.   :)   Darts night
I have changed i2c.c according to your specifications. I could only test it for a short time on my pc, but I can say it works. Thanks a lot.
Tomorrow I can tell you more if there are still problems.
Also many thanks to you phil-barrett for your quick help.
It is always nice how you get help here.