terjeio/grblHAL #92: Tool Change Spindle&Coolant activation issue

joharymanitra, 2020-09-18:
Hi, 

I tested the semi automatical tool change mode and it works correctly if send a manual command in the MDI.

However if I use a gcode file that includes tool change, spindle and coolant codes, the machine goes to the LINEAR_AXIS_HOME_OFFSET and starts spindle and coolant while waiting for the Cycle Start button Click. 
And it keeps the spindle and coolant ON during the probing phase.

terjeio, 2020-09-18:
What is your $341 setting?
Can you upload a program example that fails?

joharymanitra, 2020-09-18:
My $341 is set to 3. If the machine is at LINEAR_AXIS_HOME_OFFSET before tool change, it does not turn spindle&coolant ON. If not, it goes to this position, turns ONspindle&coolant then wait for CYCLE START
Below the program

(1001)
(T1  D=12 CR=0 - ZMIN=-0.5 - bullnose end mill)
G90 G94
G17
G21
(Face2)
T1 M6
S8000 M3
G54
M7
G0 X54.8 Y9.018
Z4
Z2
G1 Z0.7 F300
G18 G3 X53.6 Z-0.5 I-1.2 K0
G1 X47
X5 F600
G17 G2 X5 Y18.936 I0 J4.959
G1 X47
G3 X47 Y28.855 I0 J4.959
G1 X5
G2 X5 Y38.773 I0 J4.959
G1 X47
G3 X47 Y48.691 I0 J4.959
G1 X5
G2 X5 Y58.609 I0 J4.959
G1 X47
G3 X47 Y68.527 I0 J4.959
G1 X5
G2 X5 Y78.445 I0 J4.959
G1 X47
G3 X47 Y88.364 I0 J4.959
G1 X5
G2 X5 Y98.282 I0 J4.959
G1 X47
G3 X47 Y108.2 I0 J4.959
G1 X5
G18 G3 X3.8 Z0.7 I0 K1.2 F300
G0 Z4
G17
M9
M5
M30

terjeio, 2020-09-18:
Thanks. It seems to work correctly with the latest version \(just committed\), can you update to that and try again?

joharymanitra, 2020-09-19:
Hi,

This worked. Thanks for your help. My issue was due to out of date driver\serial.c

closed 2020-09-19 by joharymanitra