terjeio/grblHAL #93: Multiple processor support

phil-barrett, 2020-09-20:
I know nothing will happen for a long(ish) time but wanted to put this idea in play.

There are a number of ARM machines (here already or coming out soon) that will have 2 separate cores.  The iMXRT 1170 will have M7 (1 GHz!) and M4 cores. PJRC is publicly talking about a Teensy 5 series based on it. Probably less than $30 USD. The Portentia H7 is a non-starter due to high cost but it has the STM32H747 which is a dual core M4/M7 processor. Maybe there is a Rainbow Pill coming? We've come a LONG way from the Mega328P

I am thinking grbl core would run on the M7 and UI and Comm on the M4.  As we move to things like attached screens with DRO and UI capability, it makes sense not to burden core machine control with doing this. Though, it may require code refactoring.  I can envision a machine that pulls the GCode file from the network and runs it without a sender.  Or perhaps I should say the sender runs on the M4.

Anyway, I don't expect anything soon from this...

terjeio, 2020-09-20:
I have tinkered a bit with adding a display to grblHAL but decided against it for the simple reason that I do not want grblHAL to be distracted from the main task at hand. Only lightweight tasks or task needed for the operation of the controller should be added.

With a second core available display handling can be delegated to that, and I see no reason for why this cannot be just another plugin for sending the data relevant data over to that core. I do not think any major refactoring is needed for that.

I have put a PSoC6 development board on my todo order list - this is also dual core and a cheap entry point into this world.

Another idea of mine would be to delegate step generation to a FPGA for truly jitter free step pulses and higher order acceleration profiles... There are combined FPGAs and ARM cores out there as well.

> Anyway, I don't expect anything soon from this...
 
Not from me for sure, I need to spend some time in the workshop - it has been too much coding lately.

arferdz, 2020-09-21:
i had done giving 32bit grbl a ui using a 3.5" rpi tft display. 

i prefer using this fork since i believe this is really for 32bit processors. hopefully i can do the same thing for this .

regards.

terjeio, 2020-09-21:
> hopefully i can do the same thing for this .

Technically it should be easier as you can write a plugin and "hook" the plugin into grblHAL. Only an init function call needs to be added to the selected driver, functions in the plugin that needs to be called from grblHAL are added by modifying pointers in the HAL structure during that call. Feel free to start a new issue if you want some guidance on how to do this.

Unlike other ports there is no need to change many files in the original source making it hard to update to newer versions. If you keep the plugin private it is just a matter of readding the init function call after a update. If you make it public I may accept a pull request for it.

What might be a problem with your code is that you are using Arduino libraries in your current implementation. Many grblHAL drivers do not use the Arduino framework so a Arduino based plugin will either be limited to those that do or require additional code in the plugin.

A side note: I have made an external [MPG controller](https://github.com/terjeio/GRBL_MPG_DRO_BoosterPack) with a TFT display. The [display driver code used](https://github.com/terjeio/Display-libraries) for that is not Arduino based and is easily ported to new processors, at least for the display variants currently supported.

closed 2022-01-27 by terjeio