terjeio/grblHAL #94: Pin Mapping for SKR boards, trinamic stepper drivers, and such

raininja, 2020-09-27:
Hello @terjeio , thanks for your work here.  I'm attempting to add support for the BigTreeTech SKR 1.4 board, and I'm a bit stuck on the pin assignments for spindle and coolant.   

1.  I'm thinking to use the Heated Bed output as it's a PWM controlled spindle.  Is this appropriate? 
2.  I've noticed some code relating to coolant, I'm wondering if this can be made conditional?
3.  The codebase at present only supports the TMC2180 stepper driver. I would like to expand this to the TMC2208.

Any guidance from any parties is welcomed.

terjeio, 2020-09-27:
> I'm thinking to use the Heated Bed output as it's a PWM controlled spindle. Is this appropriate?

I think so, if it is an open drain output a pullup resistor or a buffer may be required.

> I've noticed some code relating to coolant, I'm wondering if this can be made conditional?

You can add conditions in driver.c code or even disable the output by just ignoring it, an example:

https://github.com/terjeio/grblHAL/blob/a08601881ebdb6c519fbfc228e11d5f65dd4f755/drivers/ESP32/driver.c#L958-L973

> The codebase at present only supports the TMC2180 stepper driver. I would like to expand this to the TMC2208.

Should be doable, you can use the TMC2130 plugin implementation as a start. You can see how it is added to a driver in the MSP432 [driver.c](https://github.com/terjeio/grblHAL/blob/master/drivers/MSP432/driver.c).

I will soon commit an update where memory for settings are dynamically allocated by the plugin making it easier to integrate.

IIRC the board uses a LPC1768 or 69 processor. If using similar pin assignments as ReARM then be aware that the limit switch inputs are not interrupt capable. Ok for homing but hard limits will not work.

raininja, 2020-10-01:
so I have convinced platformio to start building the code, and I'm running into an issue with the init system, it seems you've used both? I'm not using eclipse  and I see mention of CMSIS/LPCOpen in the .cproject file

I suppose I need to let platformio know that we are using CMSIS (as I need it for PWM in the spindle?)

Some guidance, please and thanks @terjeio 

### "pio run" output
```
> Executing task: C:\Users\%USER%\.platformio\penv\Scripts\platformio.exe run <

Processing LPC1769 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.3.zip; framework: arduino; board: nxp_lpc1768)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1768.html
PLATFORM: NXP Arduino LPC176x (0.1.3) > NXP LPC1768
HARDWARE: LPC1768 100MHz, 31.97KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES:
 - framework-arduino-lpc176x 0.2.5
 - toolchain-gccarmnoneeabi 1.90301.200702 (9.3.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ deep+, Compatibility ~ soft
Found 11 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <FatFs> 2.0.3
|-- <grbl>
|-- <grbl-lpc>
|-- <lpc17xx>
|-- <sdcard>
|   |-- <grbl>
|   |-- <FatFs> 2.0.3
Building in release mode
Compiling build\build\LPC1769\src\cdc_desc.c.o
Compiling build\build\LPC1769\src\cdc_vcom.c.o
Compiling build\build\LPC1769\src\cr_startup_lpc175x_6x.c.o
Compiling build\build\LPC1769\src\crp.c.o
Compiling build\build\LPC1769\src\diskio.c.o
In file included from src\cdc_desc.c:32:
src\app_usbd_cfg.h:32:10: fatal error: error.h: No such file or directory

***************************************************************
* Looking for error.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:error.h"
* Web  > Compiling build\build\LPC1769\src\driver.c.o
https://platformio.org/lib/search?query=header:error.h
*
***************************************************************

   32 | #include "error.h"
Compiling build\build\LPC1769\src\eeprom.c.o
      |          ^~~~Compiling build\build\LPC1769\src\flash.c.o
~~~Compiling build\build\LPC1769\src\grbl-lpc\pwm_driver.c.o
~~
Compiling build\build\LPC1769\src\grbl\coolant_control.c.o
compilation terminated.
Compiling build\build\LPC1769\src\grbl\corexy.c.o
In file included from src\cdc_vcom.c:32:
Compiling build\build\LPC1769\src\grbl\gcode.c.o
src\app_usbd_cfg.h:32:10: fatal error: error.h: No such file or directory

***************************************************************
* Looking for Compiling build\build\LPC1769\src\grbl\grbllib.c.o
error.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:error.h"
* Web  > Compiling build\build\LPC1769\src\grbl\limits.c.o
https://platformio.org/lib/search?query=header:error.hCompiling build\build\LPC1769\src\grbl\maslow.c.o

*
[... 99 lines cut ...]
Terminal will be reused by tasks, press any key to close it. 
```
### from file grblHAL\drivers\LPC1769\src\cr_startup_lpc175x_6x.c
```
#if defined (__USE_CMSIS) || defined (__USE_LPCOPEN)
// Declaration of external SystemInit function
extern void SystemInit(void);
#endif

```

### from .cproject in drivers\LPC1769
```
								<option id="gnu.both.asm.option.flags.crt.243121290" name="Assembler flags" superClass="gnu.both.asm.option.flags.crt" useByScannerDiscovery="false" value="-c -x assembler-with-cpp -DDEBUG -D__CODE_RED -DCORE_M3 -D__USE_CMSIS=CMSIS_CORE_LPC17xx -D__LPC17XX__ -D__NEWLIB__" valueType="string"/>

```

### from .project in drivers\LPC1769
```

	<projects>
		<project>CMSIS_CORE_LPC17xx</project>
	</projects>

```

raininja, 2020-10-01:
I've answered my own question haven't I??

I need to tell platformio which init system to use and probably define USE_CMSIS??

terjeio, 2020-10-01:
Here are some paths and symbols defined in MCUExpresso that may be of use:

[image]

[image]

N_AXIS is a grblHAL symbol for number of axes.

Options on the compiler command line:

`-DDEBUG -D__CODE_RED -DCORE_M3 -D__LPC17XX__ -D__NEWLIB__ -D__USE_LPCOPEN -DNO_BOARD_LIB -DN_AXIS=3 -I"D:\Projects\NXP\GRBL Driver LPC176x\src" -I"D:\Projects\NXP\GRBL Driver LPC176x\src\FatFs" -I"D:\Projects\NXP\lpc_chip_175x_6x\inc" -I"D:\Projects\NXP`

raininja, 2020-10-08:
hrm platformio is doing something weird as I seem to be missing some clock signals during linking 

### platformio.ini
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[platformio]
; workspace_dir = build
src_dir = src
lib_dir = src

[env:LPC1769]
; platform = nxplpc
; board = lpc1768
; framework = mbed
; monitor_speed = 115200
; lib_ldf_mode = deep+

; platform          = https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.3.zip
; platform_packages = framework-arduino-lpc176x@^0.2.5
; framework = arduino
; board             = nxp_lpc1768
; lib_ldf_mode      = deep+
; lib_compat_mode   = strict

; platform          = https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.3.zip
; platform_packages = framework-arduino-lpc176x@^0.2.5
; framework = arduino
; board             = nxp_lpc1768
; lib_ldf_mode      = deep+
; lib_compat_mode   = strict

platform          = nxplpc
; framework         = mbed
upload_protocol   = mbed
board             = lpc1768
lib_ldf_mode      = deep+
lib_archive       = yes

lib_deps =
    FatFs
    lpc17xx
    externals/lpc_chip_175x_6x/inc
    externals/lpc_chip_175x_6x/inc/usbd
    sdcard
    grbl
    grbl-lpc
    ; externals
    ; externals/CMSIS_5
    ; externals/NXP_LPC
    ; externals/lpc_chip_175x_6x/libs

; build_type = debug deprecated?
build_flags =
    -D CHIP_LPC175X_6X
    -D CORE_M3
    -D __LPC17XX__
    -D __USE_CMSIS=CMSIS_CORE_LPC17xx
    -D NO_BOARD_LIB
    -Wl,-Map,output.map
    -v

[... 67 lines cut ...]
c:/users/denkijin/.platformio/packages/toolchain-gccarmnoneeabi@1.80201.190214/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: eeprom.c:(.text.eepromInit+0x18): undefined reference to `Chip_IOCON_PinMuxSet'
c:/users/denkijin/.platformio/packages/toolchain-gccarmnoneeabi@1.80201.190214/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: eeprom.c:(.text.eepromInit+0x2e): undefined reference to `Chip_I2C_Init'
c:/users/denkijin/.platformio/packages/toolchain-gccarmnoneeabi@1.80201.190214/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: eeprom.c:(.text.eepromInit+0x36): undefined reference to `Chip_I2C_SetClockRate'
c:/users/denkijin/.platformio/packages/toolchain-gccarmnoneeabi@1.80201.190214/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: eeprom.c:(.text.eepromInit+0x50): undefined reference to `Chip_I2C_EventHandlerPolling'
c:/users/denkijin/.platformio/packages/toolchain-gccarmnoneeabi@1.80201.190214/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: eeprom.c:(.text.eepromInit+0x42): undefined reference to `Chip_I2C_SetMasterEventHandler'
c:/users/denkijin/.platformio/packages/toolchain-gccarmnoneeabi@1.80201.190214/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio\build\LPC1769\src\usb_serial.o: in function `usbInit':
usb_serial.c:(.text.usbInit+0x4): undefined reference to `Chip_USB_Init'
c:/users/denkijin/.platformio/packages/toolchain-gccarmnoneeabi@1.80201.190214/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: usb_serial.c:(.text.usbInit+0x10): undefined reference to `Chip_IOCON_PinMuxSet'
c:/users/denkijin/.platformio/packages/toolchain-gccarmnoneeabi@1.80201.190214/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: usb_serial.c:(.text.usbInit+0x1c): undefined reference to `Chip_IOCON_PinMuxSet'
c:/users/denkijin/.platformio/packages/toolchain-gccarmnoneeabi@1.80201.190214/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: usb_serial.c:(.text.usbInit+0x28): undefined reference to `Chip_IOCON_PinMuxSet'
c:/users/denkijin/.platformio/packages/toolchain-gccarmnoneeabi@1.80201.190214/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio\build\LPC1769\src\usb_serial.o:(.rodata.g_usbApi+0x0): undefined reference to `hw_api'
c:/users/denkijin/.platformio/packages/toolchain-gccarmnoneeabi@1.80201.190214/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio\build\LPC1769\src\usb_serial.o:(.rodata.g_usbApi+0x4): undefined reference to `core_api'
c:/users/denkijin/.platformio/packages/toolchain-gccarmnoneeabi@1.80201.190214/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio\build\LPC1769\src\usb_serial.o:(.rodata.g_usbApi+0x14): undefined reference to `cdc_api'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\LPC1769\firmware.elf] Error 1
```
### armcc command for driver.c

```

COLLECT_GCC_OPTIONS='-o' '.pio\build\LPC1769\src\driver.o' '-c' '-Os' '-ffunction-sections' '-fdata-sections' '-Wall' '-mthumb' '-nostdlib' '-mcpu=cortex-m3' '-v' '-D' 'F_CPU=96000000L' '-D' 'PLATFORMIO=50002' '-D' 'CHIP_LPC175X_6X' '-D' 'CORE_M3' '-D' '__LPC17XX__' '-D' '__USE_CMSIS=CMSIS_CORE_LPC17xx' '-D' 'NO_BOARD_LIB' '-I' 'include' '-I' 'src' '-I' 'src\sdcard' '-I' 'src\grbl-lpc' '-I' 'src\grbl' '-I' '.pio\libdeps\LPC1769\usbd' '-I' '.pio\libdeps\LPC1769\inc' '-I' 'src\FatFs' '-I' 'C:\Users\denkijin\.platformio\platforms\nxplpc\builder\grbl' '-I' 'externals\lpc_chip_175x_6x\inc' '-I' 'externals\lpc_chip_175x_6x\inc\usbd' '-mfloat-abi=soft' '-march=a [... cut ...] rs\denkijin\.platformio\packages\toolchain-gccarmnoneeabi@1.80201.190214\bin\../arm-none-eabi -D__USES_INITFINI__ -D F_CPU=96000000L -D PLATFORMIO=50002 -D CHIP_LPC175X_6X -D CORE_M3 -D __LPC17XX__ -D __USE_CMSIS=CMSIS_CORE_LPC17xx -D NO_BOARD_LIB src\driver.c -quiet -dumpbase driver.c -mthumb -mcpu=cortex-m3 -mfloat-abi=soft -march=armv7-m -auxbase-strip .pio\build\LPC1769\src\driver.o -Os -Wall -version -ffunction-sections -fdata-sections -o C:\Users\denkijin\AppData\Local\Temp\ccdXiagh.s

```

terjeio, 2020-10-09:
> hrm platformio is doing something weird as I seem to be missing some clock signals during linking

Not so, I see now that the project has external project references to CMSIS_CORE_LPC17xx and lpc_chip_175x_6x. Not sure how I should handle this when publishing to gitHub, so I have attached the projects as a zip for now.

[lpc_175x_6x.zip](https://github.com/terjeio/grblHAL/files/5355227/lpc_175x_6x.zip)

I will install PlatformIO and check this out later when I get a new development machine, the current one has no room...

raininja, 2020-10-14:
OK so the main issue here @terjeio is PEBKAC

I've been compiling things so often, for so long,  that I didn't really realize that my knowledge of the process was not as complete as I would have liked. I'm reassessing.

raininja, 2020-10-18:
ok I'm trying with LPCXpresso>>

EDIT: it doesn't like my machine.
```
C:\nxp\LPCXpresso_8.2.2_650\lpcxpresso\msys\bin\make.exe: *** Couldn't reserve space for cygwin's heap, Win32 error 0
```
MCUXpresso? Let's try that -_-

raininja, 2020-10-20:
Oh lawd, you can't make this up.  Same failure in MCUXpresso
for some reason, it's trying to link "" and "" 

```
Building target: GRBL Driver LPC176x.axf
Invoking: MCU Linker
arm-none-eabi-gcc -nostdlib -L"C:\Users\denkijin\Documents\MCUXpressoIDE_11.2.1_4149\workspace\lpc_chip_175x_6x\Debug" -L"C:\Users\denkijin\Documents\MCUXpressoIDE_11.2.1_4149\workspace\lpc_chip_175x_6x\libs" -Xlinker --defsym=__heap_size__=0x600 -Xlinker -Map="GRBL Driver LPC176x.map" -Xlinker --cref -Xlinker --gc-sections -Xlinker -print-memory-usage -mcpu=cortex-m3 -mthumb -T "GRBL_Driver_LPC176x_Debug.ld" -o "GRBL Driver LPC176x.axf"  ./src/sdcard/sdcard.o  ./src/lpc17xx/_exit.o  ./src/grbl-lpc/pwm_driver.o  ./src/grbl/coolant_control.o ./src/grbl/corexy.o ./src/grbl/gcode.o ./src/grbl/grbllib.o ./src/grbl/limits.o ./src/grbl/maslow.o ./src/grbl/motion_control.o ./src/grbl/my_plugin.o ./src/grbl/nuts_bolts.o ./src/grbl/nvs_buffer.o ./src/grbl/override.o ./src/grbl/pid.o ./src/grbl/planner.o ./src/grbl/protocol.o ./src/grbl/report.o ./src/grbl/settings.o ./src/grbl/sleep.o ./src/grbl/spindle_control.o ./src/grbl/state_machine.o ./src/grbl/stepper.o ./src/grbl/system.o ./src/grbl/tool_change.o ./src/grbl/wall_plotter.o  ./src/FatFs/ff.o ./src/FatFs/ffsystem.o ./src/FatFs/ffunicode.o  ./src/cdc_desc.o ./src/cdc_vcom.o ./src/cr_startup_lpc175x_6x.o ./src/crp.o ./src/diskio.o ./src/driver.o ./src/eeprom.o ./src/flash.o ./src/main.o ./src/serial.o ./src/sysinit.o ./src/usb_serial.o   -l"" -l""
c:/nxp/mcuxpressoide_11.2.1_4149/ide/plugins/com.nxp.mcuxpresso.tools.win32_11.2.0.202008311134/tools/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: cannot find -l-l
collect2.exe: error: ld returned 1 exit status
make: *** [makefile:37: GRBL Driver LPC176x.axf] Error 1
"make -r -j16 all" terminated with exit code 2. Build might be incomplete.
```

terjeio, 2020-10-21:
Have you imported the two projects in lpc_175x_6x.zip? There are external dependencies to those. The grbl project should be self contained? I'll have to look into how to achieve that...

[image]

terjeio, 2020-10-21:
I have just committed a update to grblHAL, I have cleaned up the project files for this driver and removed dependency to the CMSIS_CORE_LPC17xx project. Still not sure how to handle the other dependeny short of adding the project to MCUXpresso.

raininja, 2020-10-22:
Good news! I finally convinced the code to build, albeit with SDCARD_ENABLE = 0.

next, I plan to clean up my repo, merge your new changes, and then see if this code actually works! 

thanks, @terjeio  for your help and patience with my inexperience :D you're a good chap

zmrdko, 2020-11-17:
@raininja how did it go?
I am currently trying to compile grblHAL for SKR 1.4 TURBO (LPC1769). I was able to link libraries and dependencies, but when I flash the firmware to board, the USB connection does not work. Windows will not detect anything. I checked USB pins D+ is 0.29, D- is 0.30 and USB_CONNECT is 2.9, which is same as SMOOTHIEBOARD or RE-ARM.

@terjeio can you please point out, what am I missing?

terjeio, 2020-11-17:
> can you please point out, what am I missing?

The [USB driver](http://smoothieware.org/windows-drivers)?

Drivers can be found in the IDE installation directory too.

FYI I just compiled and flashed with the latest code from the test branch and it works for me.

[image]

terjeio, 2020-11-17:
Or it could be the CPU clock, I am testing with a Re-ARM board running at 100MHz - your board runs at 120MHz it seems.
If the derived USB clock is off it will not enumerate.

zmrdko, 2020-11-17:
Thanks for reply. The issue is, that Windows will not even detect, that I connected the board, so I cant even get to install the driver. How can I change the USB clock?

terjeio, 2020-11-17:
>  How can I change the USB clock?

Good question, currently I have no good answer. The clock tree is set up assuming there is a 12MHz crystal for the main clock. So first question to ask is what is the crystal frequency? Looks like it is 12Mhz from a picture I found - it is stamped into the top of the crystal. Can you check that?
And:
Did you program via a debug probe or via copying to a SD-card?
Do you have access to a FTDI <-> serial breakout? We could use that as a start if UART pins are available.

phil-barrett, 2020-11-17:
A suggestion:  provide a pre-built binary for those that want to try it but don't want to (or can't) set up a build system.

terjeio, 2020-11-17:
> A suggestion: provide a pre-built binary for those that want to try it but don't want to (or can't) set up a build system.

I have to buy a board first to ensure the driver is complete and is working. Hard to debug without one.

And look at the BTT github site, [many issues](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues) - even for caps near the crystal catching fire... Scary.

phil-barrett, 2020-11-17:
OK, I retract that!  But to the point, it would be great if someone who has a build that they know works would make it available for others. I am thinking how I could do that on grbl.org. will get back to you on that point.

Wow, BigTree has some real problems.  I would guess that is only a small percentage of the complaints.

zmrdko, 2020-11-18:
@terjeio Yes its 12MHz crystal right next to main chip. I used SD card flash option. If its required, I can get FTDI breakout [https://www.gme.sk/data/product/480_480/pctdetail.775-060.1.jpg](url)
I had connected 24V source via DCIN and USB and it did not burn. But there is +VIN- sign with 4 pins right next to USB which is connected to DCIN.

terjeio, 2020-11-18:
Ok, do you have a voltmeter or a oscilloscope available? Or a LED and a drop resistor for that? Or is there a LED on the board that I can blink?

My idea is to add a bit of code to toggle a pin to see if grblHAL is alive. If not then getting a FTDI breakout is not going to be helpful. BTW the board has a header for UART mapped to the pins that the driver supports.

zmrdko, 2020-11-18:
Yes, I have voltmeter and i am ready with LED and resistor. I don't think that there is any led on SKR which is directly controllable, but I am not an expert..

[https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.4/Hardware/BTT%20SKR%20V1.4-SCH.pdf](skr 1.4 schematic)

terjeio, 2020-11-18:
Here is a binary that will toggle P2.12 every 0.5 seconds (the EN pin for the E0 driver).
Strange thing is that when I added the few lines of code to do that (in the systick ISR) the MCU hard faulted. I had to increase the heap size significantly to avoid that. I have no idea yet why, never seen anything like that before.
So if the pin is toggling then maybe the USB works too?

[GRBL Driver LPC176x.zip](https://github.com/terjeio/grblHAL/files/5561816/GRBL.Driver.LPC176x.zip)

zmrdko, 2020-11-18:
firmware got updated (via SD card - there is a file firmware.cur now).
No LED blinking on E0 EN pin. I tried USB power as well as DCIN power. USB connection is dead.
Constant 3.28V on E0 pin (same as on others). Definitely no blinking. LED lighted up a little, but barely visible.

terjeio, 2020-11-18:
Well, then it is likely it goes into hardfault for some reason. Or the board is broken? - nah, then the bootloader would not change the filename. Marlin works?

There is IMO one more option to try - not to use flash for settings.

terjeio, 2020-11-18:
Here we go:

[GRBL Driver LPC176x NoFlash.zip](https://github.com/terjeio/grblHAL/files/5562046/GRBL.Driver.LPC176x.NoFlash.zip)

zmrdko, 2020-11-18:
If I flash Marlin, it works. Board is new and working.

zmrdko, 2020-11-18:
Same. No response.
But I did this test:
Flash Marlin, change setting: M92 X160 (from 80 to 160). Save settings M500. Settings are stored.
Flash grblHAL - no response.
Flash Marlin again, check settings and M92 X is still 160.
So maybe something with EEPROM?

terjeio, 2020-11-18:
Could it be the bootloader refusing to start the image? I have a OM13085 dev board that is LPC1769 based (Re-ARM is LPC1768) and that works for me as well. I program both via a debug probe.

Ok, I managed to reflash the bootloader and seemingly it loaded the bin as the name was changed. No USB and I have the following device with an exclamation mark in shown in the device manager.

[image]

Do you have that too?

> So maybe something with EEPROM?

Is there an EEPROM on your board, I cannot se any in the schematic? The MCU has none on-chip, by default grblHAL uses flash for settings storage.

zmrdko, 2020-11-18:
I only flashed using SD card. When I connect the board with USB, nothing shows in device manager, no new harware popup, nothing.
I have FTDI board, but I have to figure out, how to connect it..

zmrdko, 2020-11-18:
I tried to flash smoothieware and the board was detected by windows and I was able to connect to the board using CNCjs.
However if I tried to move I got alarm and board freezes. It might be caused by wrong pin mapping.

terjeio, 2020-11-18:
> I have FTDI board, but I have to figure out, how to connect it..

This will not help, I have to figure out how to make grblHAL run when bootloaded. From what I understand the image is  (has to be?) loaded at a different origin from what is has been compiled for and what is likely going pear shaped is the interrupt vectors - by default these are at fixed locations and moving the image means they are no longer where they are supposed to be. So first thing to do is to relocate the vector table to RAM and see if that helps. Then it is possible that I have to change the load address to somewhere above the bootloader.

So I have some more work to do before there is any point in you trying anything (short of getting a debug probe). Give me a bit of time...

terjeio, 2020-11-20:
@zmrdko Here are two images that boots, one for UART and one for USB. The USB image is unstable and may hang, it could be that is suffers from a known 64 byte message length problem.

> I have FTDI board, but I have to figure out, how to connect it..

Connect FTDI Tx to SKR Rx (0.3) and FTDI Rx to SKR Tx (0.2) and FTDI ground to GND. The SKR pins are in the TFT connector. You may have to ground the FTDI CTS pin too, or disable hardware handshake in the port configuration.
NB - ensure that the FTDI card use 3.3V for signals first! IIRC the LPC1769 _may_ tolerate 5V, to be looked up in the datasheet...

[Debug.zip](https://github.com/terjeio/grblHAL/files/5574384/Debug.zip)

zmrdko, 2020-11-20:
I tried USB version and it worked!!
Now the board is detected and I was able to connect to it using CNCjs. Upon trying to reset $X i got ERROR 18.

terjeio, 2020-11-20:
> Upon trying to reset $X i got ERROR 18.

grblHAL defaults to normally closed switches (NC), connect 0.18 to ground or invert the reset pin in settings.

zmrdko, 2020-11-20:
@terjeio i have bare board connected (no limit pins or anything) and my status message ? says:
<Alarm|MPos:0.000,0.000,0.000|Bf:35,1024|FS:0,0|Pn:PRHS|Ov:100,100,100>

I guess the important is Pn: PRHS, but I don't know how to get rid of it.

terjeio, 2020-11-20:
> I guess the important is Pn: PRHS, but I don't know how to get rid of it.

Here are the pin definitions for R, H and S respectively. Connect them them to ground or invert them in settings: `$14=7`

```
// Define user-control CONTROLs (cycle start, reset, feed hold) input pins.
#define RESET_PORT_PN       0
#define RESET_PORT          port(RESET_PORT_PN)
#define RESET_PIN           18
#define RESET_BIT           (1<<RESET_PIN)

#define FEED_HOLD_PN        0
#define FEED_HOLD_PORT      port(FEED_HOLD_PN)
#define FEED_HOLD_PIN       16
#define FEED_HOLD_BIT       (1<<FEED_HOLD_PIN)

#define CYCLE_START_PN      0
#define CYCLE_START_PORT    port(CYCLE_START_PN)
#define CYCLE_START_PIN     15
#define CYCLE_START_BIT     (1<<CYCLE_START_PIN)
```
P is for the probe input - does not matter if you are not using a probe.

zmrdko, 2020-11-20:
Thanks, it worked.  I tried to move stepper motors, but they dont move. it looks like they are locked when idle. But when I press X+ to move them, they will unlock. Is it possible that X_DISABLE_PIN logic is inverted?

phil-barrett, 2020-11-20:
I bet you need to invert the step pulse $2.

terjeio, 2020-11-20:
>Is it possible that X_DISABLE_PIN logic is inverted?

Yes, could well be. Try `$4=7` to invert the enable pins.

zmrdko, 2020-11-20:
It works now. Why 7?

phil-barrett, 2020-11-20:
Your driver are enabled by the signal going low, not high.

zmrdko, 2020-11-20:
Yes, but why 7 and not 1/0.

phil-barrett, 2020-11-20:
Ah,  sorry. Binary vs Decimal. Each axis enable is represented by a bit.  X is 0, Y is 1, Z is 2.  Setting a bit to 1 inverts its corresponding axis.  So all 3 set to 1 is a 7.

terjeio, 2020-11-20:
> Each axis enable is represented by a bit. X is 0, Y is 1, Z is 2.

bit 0 = 1, bit 1 = 2, bit 2 = 4. When all set --> 1 + 2 + 4 = 7

For some drivers/boards 0 or 1 may do the trick for all motors - this when there is one shared enable signal. If so I map it to the X enable (bit 0). The SKR board has separate enable signals for all.

zmrdko, 2020-11-20:
@terjeio  Thanks a million. The movement and endstops work perfect. And probably spindle, etc. as well.
I have one more question: is it possible to add text to GRBL settings in $$? Some items are missing the description.

terjeio, 2020-11-21:
> I have one more question: is it possible to add text to GRBL settings in $$?

No - not without breaking senders.

However you have got a version that I am about to release that has help for settings, from the changelog:

* Added `$HELP` command and `$$<n>` command for listing information about a specific setting.  
`$HELP` on its own prints arguments that can be used with `$HELP`.  
`$HELP Commands` - print `$` commands available with a short description.  
`$HELP Settings` - print information about all available settings.  
`$HELP <argument>` print information about settings from the setting group provided in `<argument>`. E.g. `$HELP Spindle` will print information about spindle settings.  
__NOTE:__ do _NOT_ issue these commands from a sender MDI as the output may crash it, output is in plaintext and thus intended for use from a terminal only.
---

`$$<n>` is `$$N<n>` in your version as this is brand new and not yet finalized, e.g.` $$N14` returns:

```
$14: Invert control pins as bitfield:
    0 - Reset (1)
    1 - Feed hold (2)
    2 - Cycle start (4)
    3 - Safety door (8)
    4 - Block delete (16)
    5 - Optional stop (32)
    6 - EStop (64)
    7 - Probe connected (128)
ok
```

terjeio, 2020-11-21:
> Some items are missing the description.

If in CNCjs then CNCjs has to be updated since grblHAL has many new settings compared to grbl - and CNCjs has this information hardcoded in the program:

https://github.com/cncjs/cncjs/blob/master/src/server/controllers/Grbl/constants.js

terjeio, 2020-11-22:
[Test branch ](https://github.com/terjeio/grblHAL/tree/test)now updated - [build configuration added](https://github.com/terjeio/grblHAL/blob/test/drivers/LPC1769/README.md) for bootloader programming.
USB problem I saw seems to be fixed too.

zmrdko, 2020-11-22:
Thanks a lot!!
I have one more question: I use BIGTREETECH TMC2209 v1.2 stepper drivers which are quite popular now and have an option for sensorless homing (stallguard feature).
The question is, if its reasonable to use it (if it would not false trigger under load during milling operation). The sensitivity would have to be quite high. 

But the main thing is, what is required to make it work.
From what I understand SKR 1.4 Turbo features both UART and SPI interfaces for stepper drivers.

terjeio, 2020-11-23:
> But the main thing is, what is required to make it work.

Support code and testing. There is [code available](https://github.com/terjeio/grblHAL/tree/master/plugins/trinamic) for TMC2130, that "only" needs driver support for SPI transfers. It is possible that the TMC2209 has a similar protocol as TMC2130 so the code for that could be used as a starting point.

zmrdko, 2020-11-23:
I just went through the tmc2209 datasheet, and it says it supports only uart mode. So that is a no go.

terjeio, 2020-11-23:
> So that is a no go.

Why? The board has wiring for UART mode so it is a matter of providing low level datagram transfer in the driver for that.

zmrdko, 2020-11-23:
Well, that is kind of above my skill level, so I am lost..

raininja, 2020-11-25:
> @raininja how did it go?
@zmrdko Mixed results.  After finally achieving a build in MCUXpresso by disabling the sdcard functionality, I became a bit discouraged .  . .  

my attempts are described in the list to follow:

- sdcard is broken, linking issues (probably PEBKAC) 
- platformio build, as I I would rather use mbedOS and flash over usb
- TMC2208 drivers
I'll give it some attention for the next few days  

> Why? The board has wiring for UART mode so it is a matter of providing low level datagram transfer in the driver for that.

so, there is a well-known library [here ](https://github.com/teemuatlut/TMCStepper)that is quite complete, seems like a valid starting point/reference?

> I am currently trying to compile grblHAL for SKR 1.4 TURBO (LPC1769). 

I'm working with the 1.4 right now but I just replaced the fan0 mosfet and fixed my turbo, so I can try this, assuming it builds properly in platformio

raininja, 2020-11-29:
ugh! a little insight here, hopefully, @terjeio ??

```
Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
Target: arm-none-eabi
Configured with: /Host/Work/arm-none-eabi-gcc-8.2.1-1.4/gcc/configure --prefix=/Host/Work/arm-none-eabi-gcc-8.2.1-1.4/install/win64/arm-none-eabi-gcc --infodir=/Host/Work/arm-none-eabi-gcc-8.2.1-1.4/install/win64/arm-none-eabi-gcc/share/doc/info --mandir=/Host/Work/arm-none-eabi-gcc-8.2.1-1.4/install/win64/arm-none-eabi-gcc/share/doc/man --htmldir=/Host/Work/arm-none-eabi-gcc-8.2.1-1.4/install/win64/arm-none-eabi-gcc/share/doc/html --pdfdir=/Host/Work/arm-none-eabi-gcc-8.2.1-1.4/install/win64/arm-none-eabi-gcc/share/doc/pdf --build=x86_64-unknown-linux-gnu --host=x86_64-w64-mingw32 --target=arm-none-eabi --with-pkgversion='GNU MCU Eclipse ARM Embedded GCC\x2C 64-bit' --enable-languages=c,c++ --enable-mingw-wildcard --enable-plugins --enable-lto --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --with-gnu-as --with-gnu-ld --with-newlib --with-headers=yes --with-python-dir=share/gcc-arm-none-eabi --with-sysroot=/Host/Work/arm-none-eabi-gcc-8.2.1-1.4/install/win64/arm-none-eabi-gcc/arm-none-eabi --with-multilib-list=rmprofile --disable-rpath --disable-build-format-warnings --with-system-zlib
Thread model: single
gcc version 8.2.1 20181213 (release) [gcc-8-branch revision 267074] (GNU MCU Eclipse ARM Embedded GCC, 64-bit)
COLLECT_GCC_OPTIONS='-o' '.pio\build\LPC1769\src\driver.o' '-c' '-Os' '-ffunction-sections' '-fdata-sections' '-Wall' '-mthumb' '-nostdlib' '-mcpu=cortex-m3' '-v' '-D' 'F_CPU=96000000L' '-D' 'PLATFORMIO=50003' '-D' 'CHIP_LPC175X_6X' '-D' 'CORE_M3' '-D' '__LPC17XX__' '-D' '__USE_CMSIS=CMSIS_CORE_LPC17xx' '-I' 'include' '-I' 'src' '-I' 'src\sdcard' '-I' 'src\grbl-lpc' '-I' 'src\grbl' '-I' '.pio\libdeps\LPC1769\usbd' '-I' '.pio\libdeps\LPC1769\inc' '-I' 'src\lpc17xx' '-I' 'src\FatFs' '-I' 'externals\lpc_chip_175x_6x\inc' '-I' 'externals\lpc_chip_175x_6x\inc\usbd' '-mfloat-abi=soft' '-march=armv7-m'
 c:/users/denkijin/.platformio/packages/toolchain-gccarmnoneeabi@1.80201.190214/bin/../libexec/gcc/arm-none-eabi/8.2.1/cc1.exe -quiet -v -I include -I src -I src\sdcard -I src\grbl-lpc 
-I src\grbl -I .pio\libdeps\LPC1769\usbd -I .pio\libdeps\LPC1769\inc -I src\lpc17xx -I src\FatFs -I externals\lpc_chip_175x_6x\inc -I externals\lpc_chip_175x_6x\inc\usbd -imultilib thumb/v7-m/nofp -iprefix c:\users\denkijin\.platformio\packages\toolchain-gccarmnoneeabi@1.80201.190214\bin\../lib/gcc/arm-none-eabi/8.2.1/ -isysroot c:\users\denkijin\.platformio\packages\toolchain-gccarmnoneeabi@1.80201.190214\bin\../arm-none-eabi -D__USES_INITFINI__ -D F_CPU=96000000L -D PLATFORMIO=50003 -D CHIP_LPC175X_6X -D CORE_M3 -D __LPC17XX__ -D __USE_CMSIS=CMSIS_CORE_LPC17xx src\driver.c -quiet -dumpbase driver.c -mthumb -mcpu=cortex-m3 -mfloat-abi=soft -march=armv7-m -auxbase-strip .pio\build\LPC1769\src\driver.o -Os -Wall -version -ffunction-sections -fdata-sections -o C:\Users\denkijin\AppData\Local\Temp\ccWR6Kdz.s
```

raininja, 2020-11-29:
so the environment knows where the symbol is as you can see in the frame center right

[image]

terjeio, 2020-12-01:
I have some code changes for the LPC176x driver nearly ready - among them moving the library code and the USB blob into the main project. No more external dependencies. That could help?

TMC2209 support via a plugin is on my todo list, will look into that when I get my Nucleo-64 breakout board. It should not be too hard to add driver level code for UART communication to the LPC176x when that is done.

zmrdko, 2020-12-02:
@terjeio Thanks for your effort!
Since you mentioned nucleo board, I have come across GRBL-Advanced version of GRBL: [GRBL Advanced](https://github.com/Schildkroet/GRBL-Advanced), which works on STM32 F411RE/F446RE boards.
And I am wondering, if you could point out in what aspects is your grblHAL better.
I understand the grblHAL is made for wide range of boards/chips.

Its awesome, that with SKR turbo board and grblHAL you can make high performance CNC controller for cheap.
I appreciate that a lot!

Thanks!!

[45 more replies]

closed 2022-01-27 by terjeio