terjeio/grblHAL #96: grblHAL chokes on long comment lines.

phil-barrett, 2020-10-02:
This came about because the new and "improved" Fusion 360 inserts a very long comment in generated gcode for non-commercial users:

> ( When using Fusion 360 for Personal Use, the feedrate of rapid moves is reduced to match the feedrate of cutting moves, which can increase machining time. Unrestricted rapid moves are available with a Fusion 360 Subscription.)

I will spare you all a diatribe about Autodesk...

Anyway, grblHAL just stops. It does not crash. Hitting the Stop button in Grbl-GCode-Sender works and grblHAL appears to function correctly after that. Attached is a simple test case to demonstrate the problem.

For what it is worth, grbl "Classic" also has this problem so you could call it a compatibility feature...

[long line test.zip](https://github.com/terjeio/grblHAL/files/5320648/long.line.test.zip)

terjeio, 2020-10-02:
> This came about because the new and "improved" Fusion 360 inserts a very long comment in generated gcode for non-commercial users:

Then it is no longer compliant gcode, from NIST RS274/NGC:

> 3.3 Format of a Line
A permissible line of input RS274/NGC code consists of the following, in order, with the
restriction that there is a maximum (currently 256) to the number of characters allowed on a line.

grblHAL supports 256, "Classic" only 80.

So their "improvement" is to be non-compliant. Done on purpose? Or has the limit been increased by NIST?

phil-barrett, 2020-10-03:
I don't know if the standard has changed but what is the correct response to non-compliant GCode?

terjeio, 2020-10-03:
The correct response is to raise an error.

BTW this is not a controller issue, it is hanging in the sender. Some senders does not send comment lines, mine does. 
I can add filtering for comment lines that does not contain messages, those who do should be sent as they can  provide real-time feedback when the program is running, e.g. which tool to select on a tool change.

"Classic" grbl strips all comments, grblHAL does not - it sends message comments `(MSG,...)` back in sync with execution. Since grblHAL can stream code from a SD card connected to the controller it is important that message comments are displayed and not thrown away. This as there is no program listing visible when streaming from a SD card.

#### grblHAL ####
if a line is longer than 256 characters is encountered error 11 is raised and execution is stopped. I am not going to change that unless it is possible to get hold of a newer version of the specification where the line length has been increased. Since the line buffer is statically allocated it is not possible to handle this differently as a long line containing comments _may_ have gcode embedded after the first 256 characters.

#### Sender ####

Comments should not be stripped on load as comments may contain important information. A warning or error should be issued on load if there are non-compliant lines in the file.

Comment lines should not be sent when streaming to the controller, lines with comments _and_ gcode should be sent as is. IIRC LinuxCNC parses information from comments in plasma mode, I want to keep that option open. So perhaps only comment lines longer than 256 characters should not be stripped?
 
So no simple fix for standard breaking files...

terjeio, 2020-10-04:
Comments are not read into the protocol input buffer so both "Classic" and grblHAL can handle infinite length comments. If a comment is a message then the 256 character limit applies to grblHAL only.

I have just uploaded a new edge version of the sender with a fix. This will strip the content from non-message comments in order not to upset the send algorithm causing it to stall. It happened when comments were larger than min(controller serial input buffer size * 0.9, 300). 

The serial buffer is 128 characters in "Classic" grbl and 1K in most grblHAL drivers.

phil-barrett, 2020-10-04:
Problem resolved.

closed 2020-10-04 by phil-barrett

terjeio, 2020-10-06:
Resolved but I am curious about how you discovered this. I only tested with the long line in the zip file, not the Fusion comment. The Fusion 360 comment is 228 characters long and does not hang the sender if grblHAL is the controller it talks to. It hangs if the controller is "Classic" grbl. The Fusion 360 comment is in fact compliant gcode. "Classic" grbl is the one to blame...

phil-barrett, 2020-10-06:
I saw the issue with F360 reported against Grbl Classic. So, I wanted to see what grblHAL + IoSender did with a long line.  To be honest, I didn't spend a lot of time digging into the nuances. A fair criticism is that I didn't try it against Grbl Classic.  One of my "ought to do tasks" is set up a Grbl Classic test system for cases like this.  I have all the pieces, just not the time.