terjeio/grblHAL #101: add more input and output

Symeth, 2020-10-20:
hi, 

is it possible to add more output and input for GRBL HAL ?
in my machine case i need to make 5 output and 5 input for my machine

Thanks ^^

phil-barrett, 2020-10-20:
Inputs and outputs for what specifically? grblHAL can support up to 6 axes along with limit switches for them.

What kind of machine do you have?

Symeth, 2020-10-20:
Digital input HIGH and LOW get signal from  sensor 5vdc
Digital Output HIGH and LOW to relay for control solenoid valve

it's cartesian robot 3 axis, the job is to pick and place part from feeder to tray due to it integrated with several sensor and solenoid valve, that's why i need more input and output

phil-barrett, 2020-10-20:
OK, that helps. Do you have a board you want to use?  There is support for multiple digital I/Os like that in grblHAL.  My Teensy 4.1 breakout board has 4 digital ins and 6 digital outputs (all with relay drivers) that could be used.  If you are willing to write some code you can support any number of digital I/Os.  The STM32nucleos would give you a lot of flexibility, I think.  Some of the TI dev boards may also be worth looking at - Terje would know.

However, I am a little curious about your software.  Is this a GCode application?

MARIOBASZ, 2020-10-20:
I am also interested in the subject: I will need more inputs and outputs, one or two inputs will be analog

Symeth, 2020-10-20:
hi phil,

i have looking for your board and interestin, i will purchase one
and now how about the code :(
any tutorial to build this code ?

phil-barrett, 2020-10-20:
Before you buy, I should explain the limits of my board.

4 Digital inputs.  One of those pins could be used for Analog though that code is not done.  Currently the inputs have a low pass filter - 15KHz. The caps could be removed if that is an issue.  There may be a plugin coming for a Plasma Cutter Torch Height Control that uses analog input.   

On the Digital Outputs. I have 7 relay outputs.  One is for the spindle relay.  5 are for Flood Coolant, Mist Coolant and 3 Aux Relays. The last one isn't controlled by the Teensy (ganged to the spindle). TheAll relays can be selected for 5V or 12V. Each one also has a TTL output that can drive smart relays or SSRs.

You could also use the I2C header for I/O expansion but that is mostly "uncharted" grblHAL teritorry. I have an I2C ADC design that works - single channel 12 bit. I've tested it on the board outside of grblHAL.  Works pretty well but no isolation. 

The wiki has a [developer's reference.](https://github.com/terjeio/grblHAL/wiki/Hardware-Abstraction-Layer-(HAL)----developer-reference) Might be a good place to start.

Symeth, 2020-10-20:
ouh i see, 
here's my plan after done create the script and test, then i will buy your board.
i also want to use teensy4 for dev board thats why i need your board, anyway good design for your expansion board

terjeio, 2020-10-20:
The grblHAL core supports [M62-M68](http://linuxcnc.org/docs/html/gcode/m-code.html#mcode:m62-m65) for analog and digital I/O for up to 256 ports for each category (for a total of 1024 ports). The actual I/O handling is declared and provided by the driver and is of course limited by the processor and the board.

[Plugin code](https://github.com/terjeio/grblHAL/tree/master/plugins) can access the HAL API directly and may "hide" some or all of the ports from the core. In the next build I am about to release I have started to utilize this, e.g. for laser tube coolant handling with calls such as these:

```
hal.port.digital_out(LASER_COOLANT_ON_PORT, coolant_on);

float coolant_temp = (float)hal.port.wait_on_input(false, LASER_COOLANT_TEMPERATURE_PORT, WaitMode_Immediate, 0.0f) / 10.0f;

hal.port.wait_on_input(true, LASER_COOLANT_OK_PORT, WaitMode_High, gc_block->values.p) 
```

Writing code this way means that a plugin is fairly easy to add to a new processor if the driver supports the needed I/O pins.
How these pins are handled at the driver level is likely of no interest to the plugin - it could be directly via GPIO pins, via I/O expanders, via ModBus or CAN and what not.

closed 2021-02-20 by terjeio