terjeio/grblHAL #102: Ethernet configuration // FRAM FM24CL64-B

raumneun, 2020-10-20:
During the last days, I finally managed to find the time to compile grblHAL for the first time. As mentioned in other issues, I use Phils board (which is great!), and I just wanted to report the following to things:

Ethernet issue:
When using a static IP, I had to set ```#define NETWORK_IPMODE 1```, i.e. to DHCP and set $301 to 0. For some reason, the define did not change the default setting and 0.0.0.0 was used instead of the IP I defined in my_machine.h. I don't know if this is a bug or the IP mode was moved from a compile time option to a $$ setting, but this is what I had to do to make it work.
Also, in the README.md of the networking plugin (https://github.com/terjeio/grblHAL/tree/master/plugins/networking) it might be helpful to point to the lwIP repository of ddrown for all Arduino/Teensyduino users to make compiling work. I had to do some digging in the issue tracker to find out what to use (which was merely inconvenient, but I guess it makes life easier for people less experienced with programming and such).

Larger size FRAM FM24CL64-B:
This is more a success report than an issue. By using ```#define EEPROM_ENABLE 2``` and ```#define EEPROM_IS_FRAM 1```, the FM24CL64-B works just nicely. I verified this using your suggestion (change a setting, compile without eeprom, setting is gone, recompile with eeprom, setting is back).

terjeio, 2020-10-20:
> For some reason, the define did not change the default setting and 0.0.0.0 was used instead of the IP I defined in my_machine.h. I don't know if this is a bug or the IP mode was moved from a compile time option to a $$ setting, but this is what I had to do to make it work.

Changing #defines will not update settings when changed after the first flashing as they are default values. You have to force a reset to default values with a $-command: `$RST=*` will reset all settings to default, `$RST=&` will reset driver/plugin specific settings only. There are other `$RST` variants too, `$RST=&` is grblHAL specific - the rest as inherited from grbl. I'll have to improve documentation regarding this.

> Also, in the README.md of the networking plugin (https://github.com/terjeio/grblHAL/tree/master/plugins/networking) it might be helpful to point to the lwIP repository of ddrown for all Arduino/Teensyduino users to make compiling work. I need to add a driver/plugin/board compatibility crossreference somewhere...

I'll add links to the driver ReadMe for the drivers that supports networking from the plugin ReadMe. The information you needed is already available in the [iMRXT1062 driver ReadMe](https://github.com/terjeio/grblHAL/tree/master/drivers/IMXRT1062).

> Larger size FRAM FM24CL64-B

The `$I` command will now show what kind of non-volatile storage is active. In the next build a `*` is prefixed to this when the storage area is buffered in RAM, e.g:

`[NEWOPT:*EEPROM,ES,TC,SS,PID]`

raumneun, 2020-10-21:
> Changing #defines will not update settings when changed after the first flashing as they are default values. You have to force a reset to default values with a $-command: $RST=* will reset all [...]

Thanks, I wasn't aware of that. I guess I may have read it at some place but didn't think about it when I compiled it for the first time. I honestly do not remember which version I compiled and flashed first, but I guess it was my mistake.

> I'll add links to the driver ReadMe for the drivers that supports networking from the plugin ReadMe. The information you needed is already available in the iMRXT1062 driver ReadMe.

Oi. A classic case of RTWM (W as in wrong). Sorry about that. It actually makes more sense as it is probably driver specific?

> The $I command will now show what kind of non-volatile storage is active. In the next build a * is prefixed to this when the storage area is buffered in RAM, e.g:

Great!

terjeio, 2020-10-21:
> It actually makes more sense as it is probably driver specific?

Yes, how to add networking is driver specific. There is a MCU vendor/board vendor layer (including lwIP), and a "middleware" layer written by me that glues everything together.

raumneun, 2020-11-30:
Hi Terje
I noticed something strange, and I don't know if it's a problem with the code or my board.
It compiles successfully and runs well, both via USB and Ethernet.
However, when I power cycle the board, I get a connection timeout in your sender, regardless of the connection type.
When I recompile it, I can connect again if the USB cable stays plugged in.
I don't use an external 5V supply at the moment.
I do not remember if that happened also before your update to the eeprom code for >2K storage, could it be related to that?
Thanks for your support!

raumneun, 2020-11-30:
Oh and one note, the board shows up alright in the windows device manager and as recompiling works without issues, I can exclude  any USB related trouble.

phil-barrett, 2020-11-30:
I restart ioSender each time I reload the firmware.  This probably should be an issue [there](https://github.com/terjeio/Grbl-GCode-Sender).

terjeio, 2020-12-01:
> However, when I power cycle the board, I get a connection timeout in your sender, regardless of the connection type.

To be expected as a hard reset drops the connection. I could try to add code to the sender for automatic reconnect on a connection loss but not sure if it is worth the effort. USB could be problematic, at least for me (Win7), a hard reset while a connection is active requires me to terminate the program and unplug/reinsert the cable before I can connect again. This behaviour is not the same between processors either, I suspect it has to do with USB OTG beeing supported or not.

Switching to UART comms and using a USB <> UART converter will work though. The sender can handle a hard reset if the connection stays up.

raumneun, 2020-12-01:
Hi Terje, Hi Phil.
thanks for your fast replies - as always 👍 
However, probably as I did not explain the situation adequately, there is a misunderstanding.
The problem is not that I get a timeout upon power cycling and having the software running (I agree, this is not a real issue as no one would itentionally do that), it is that I cannot (re)connect to the board if it lost power at some point, neither with Ethernet, nor with USB. I always have to recompile the code before the teensy let's me connect again. For clarification, here is what I do:

1. Connect the board via USB. No external 5V supply, so this is when the board turns on.
2. Start ioSender, try to connect by either USB or Ethernet. This fails.
3. Exit ioSender.
4. With the board still connected, compile using Arduino IDE, works without issues.
5. Start ioSender with USB connection: works. Exit again.
6. Start ioSender with Ethernet connection: works. Exit again.
7. Remove USB cable (i.e. removing power from the board)
8. Reconnect USB cable, restart ioSender, expect the sender to connect to the board.
9. Connection fails with both USB and Ethernet
10. Exit ioSender
11. With the USB cable plugged in, recompile the code
12. Start ioSender, now both connections work again (until next power down)

Currently, for testing and setting up the machine, the workaround is to recompile every time, but as you can imagine this is not really a feasible solution :)

Again, I am not sure if it is a problem with the board, the teensy or the code. I will try again later today with an older version or better yet with the hex-file @phil-barrett provided to see if the problem prevails.

Cheers
Max

phil-barrett, 2020-12-01:
Hmmm.  I power cycle the teensy frequently by unplug/replug (so frequently that I fear USB Micro connector failure...) and have never seen that kind of problem. 

I would be surprised if it is FRAM related. I am running with the smaller FRAM so my_machine.h has this:

> #define EEPROM_ENABLE      1 // I2C EEPROM support. Set to 1 for 24LC16(2K), 2 for larger sizes. Requires eeprom plugin.
> #define EEPROM_IS_FRAM     1 // Uncomment when EEPROM is enabled and chip is FRAM, this to remove write delay.
> 

You can test the FRAM hypothesis by commenting out those two line and running with emulated eeprom.

terjeio, 2020-12-01:
Same here, never seen this. Can you ping the controller and get a reply?
Seeing the USB port (or beeing able to ping) means that controller is running.

> Connection fails with both USB and Ethernet

With "Unable to open connection..." as the response?

If not have you tried to connect with a terminal console such as PuTTY? It could be that the controller is in a state where the sender cannot communicate.

phil-barrett, 2020-12-01:
Could you post a screen shot of ioSender when it is not responding?

raumneun, 2020-12-01:
Thanks for your replies again.

TL;DR: Disabling the EEPROM Plugin did the trick. However, I have no idea what's wrong.

Long version:
Here is a screenshot from ioSender, note the "controller not responding" message in the status bar (ioSender Beta 7):

[image]

This was using USB, same for Ethernet.
Interestingly, I can ping the controller: 

[image]

Even more interesting, Telnet connection with Putty works as well:

[image]

Apparantly, the controller is in idle state. This is without anything changed, ie. FRAM plugin still activated as you can see from the output after issuing ```$I```

And now it just get's weird:
[image]

It seems to work now?! I am quite confused. So again, trying to reproduce it.
1. Unplug USB cable, Plug back in after some 10 s.
2. Connection does not work, neither USB, nor Ethernet (Controller not responding message in ioSender)
3. Keep trying to restart ioSender for 5 Minutes to exclude that it's a matter of time until it responds
4. Ok, still nothing.
5. Ping the guy, recheck. Pinging works, still no connection.
6. Connect with Putty via Telnet. Telnet works. Closing without issuing a command.
7. Connect with ioSender via USB: Voila, it works!
8. Connect with ioSender via Ethernet: Voila, works too!
9. Repeat procedure: Power cycle, verify that connection does not work with USB/Ethernet, Connect putty, close putty, connect with Ethernet: works at first attempt.

Two things we can learn from that:
1. opening a telnet connection with putty and with the ioSender obviously has a different effect.
2. It seems to be network related

To further verify 2., I disabled the ethernet plugin by uncommenting the respective line in ```my_machine.h``` to check whether this makes connection via USB reliable.
Result: In fact, it does not. At first, without unplugging the USB cable, I verified that Ethernet ist really deactivated (from $I output and by unsuccessfully pinging it). I then unplugged and replugged the USB cable and tried to connect with ioSender to no avail. Trying to connect with Putty to COM6 at first did not seem to work, because I just got a blank console and no Grbl welcome message. Sending commands like ```$I``` does not work, too. However, realtime commands do seem to work:
[image]
As the status was hold, I issued ```~```` which returned the following:

```
<Hold:0|MPos:0.000,0.000,0.000|Bf:35,999|FS:0,0|Pn:XYZHS>
[MSG:]
[VER:1.1f(IMXRT1062).20201103:]
[OPT:VNMSL,35,1024,3,0]
[NEWOPT:*FRAM,ES,TC]
[DRIVER VERSION:201014]
[DRIVER OPTIONS:USB.2]
[BOARD:T41U5XBB]
[PLUGIN:ODOMETERS v0.02]
ok
[VER:1.1f(IMXRT1062).20201103:]
[OPT:VNMSL,35,1024,3,0]
[NEWOPT:*FRAM,ES,TC]
[DRIVER VERSION:201014]
[DRIVER OPTIONS:USB.2]
[BOARD:T41U5XBB]
[PLUGIN:ODOMETERS v0.02]
ok
[VER:1.1f(IMXRT1062).20201103:]
[OPT:VNMSL,35,1024,3,0]
[NEWOPT:*FRAM,ES,TC]
[DRIVER VERSION:201014]
[DRIVER OPTIONS:USB.2]
[BOARD:T41U5XBB]
[PLUGIN:ODOMETERS v0.02]
ok
[VER:1.1f(IMXRT1062).20201103:]
[OPT:VNMSL,35,1024,3,0]
[NEWOPT:*FRAM,ES,TC]
[DRIVER VERSION:201014]
[DRIVER OPTIONS:USB.2]
[BOARD:T41U5XBB]
[PLUGIN:ODOMETERS v0.02]
ok
[VER:1.1f(IMXRT1062).20201103:]
[OPT:VNMSL,35,1024,3,0]
[NEWOPT:*FRAM,ES,TC]
[DRIVER VERSION:201014]
[DRIVER OPTIONS:USB.2]
[BOARD:T41U5XBB]
[PLUGIN:ODOMETERS v0.02]
ok
[VER:1.1f(IMXRT1062).20201103:]
[OPT:VNMSL,35,1024,3,0]
[NEWOPT:*FRAM,ES,TC]
[DRIVER VERSION:201014]
[DRIVER OPTIONS:USB.2]
[BOARD:T41U5XBB]
[PLUGIN:ODOMETERS v0.02]
ok
[VER:1.1f(IMXRT1062).20201103:]
[OPT:VNMSL,35,1024,3,0]
[NEWOPT:*FRAM,ES,TC]
[DRIVER VERSION:201014]
[DRIVER OPTIONS:USB.2]
[BOARD:T41U5XBB]
[PLUGIN:ODOMETERS v0.02]
ok
[VER:1.1f(IMXRT1062).20201103:]
[OPT:VNMSL,35,1024,3,0]
[NEWOPT:*FRAM,ES,TC]
[DRIVER VERSION:201014]
[DRIVER OPTIONS:USB.2]
[BOARD:T41U5XBB]
[PLUGIN:ODOMETERS v0.02]
ok
<Idle|MPos:0.000,0.000,0.000|Bf:35,1023|FS:0,0|Pn:XYZHS>
```
The last line was a responsive to another ```?``` command to see if the machine returned to idle.
After that, the controller was responsive in ioSender - until the next power cycle.
Repeating the procedure made it turn up ok again, so this is reproducible.

To check if it is one of the other plugins, I did the following:
1. Disable odometer: Same behavior.
2. Additionally disabled EEPROM/FRAM: Now it works.

So either there is something wrong with my hardware, or the EEPROM plugin has an issue.
If you have ideas what I could check next, I am happy to try! At this point, as I am not a good enough programmer, I cannot really tell what to check.

Thanks for helping out!
Cheers
Max

phil-barrett, 2020-12-01:
One thing I see is that grblHAL is in the HOLD:0 state and H and S inputs are asserted.  Try changing $14 so H & S are not asserted - $14=6 if you use putty, Settings: Grbl check boxes if you use ioSender.

raumneun, 2020-12-01:
That did the trick. Connection works now reliably, also with all plugins enabled. However, I am quite confused that I am the first person to have this problem as it is the default to not invert the pins?

Some minutes later:
Ok now it gets weird. I just uninverted the pins again and now connection still works, regardless of power cycling the board 😕

raumneun, 2020-12-01:
Ok, figured it out. It is a bad combination of settings.
Inverting EStop and not inverting H&S breaks the connection.
Inverting all of them works, and so does not inverting all of them.
The thing with FRAM was a false positive then, probably because I had a working combination in flash.

phil-barrett, 2020-12-01:
for what it is worth, it IS confusing. Probably needs a wiki page on bringing the system up for the first time. Especially on a bare device like a Teensy, Nucleo  or Pill. I think I am signed up to do that...

terjeio, 2020-12-01:
> That did the trick. Connection works now reliably, also with all plugins enabled. However, I am quite confused that I am the first person to have this problem as it is the default to not invert the pins?

If the controller starts up in HOLD state then the likely explanation is noise on the hold input pin slipping through and generating an interrupt. Strange with optocoupled inputs though...

Only real-time commands are executed while in HOLD and DOOR states so that explains why the sender failed to start properly. I have to add a test/message on startup for that - I am not to keen to take it out of the states mentioned above as this could be dangerous. There are some alarms that blocks the same way but those are already handled.

> The thing with FRAM was a false positive then, probably because I had a working combination in flash.

I was about to suggest that...

phil-barrett, 2020-12-01:
> If the controller starts up in HOLD state then the likely explanation is noise on the hold input pin slipping through and generating an interrupt. Strange with optocoupled inputs though...

I see it sometimes on first startup where the $14 feed hold sense is opposite the way the pin is pulled. I think it mostly happens with a bare teensy. Are interrupts enabled before or after the pull-ups? I assumed it was after. Once the $14 settings are dialed in, I never see it start in the Hold:0 state.

terjeio, 2020-12-01:
> Are interrupts enabled before or after the pull-ups?

After, I disable interrupts when changing port parameters and clear them before enabling. However, even with pull-up/pull-down enabled a pin can be sensitive to noise, e.g. a touch on the pin could trigger interrupts.

closed 2021-08-22 by terjeio