terjeio/grblHAL #104: Spindle Sync: One stepper pulse routine for pulse, delayed pulse and spindle sync pulse

HuubBuis, 2020-10-26:
#50 
There is a separate hal entry for a delayed stepper pulses. This would "require" 2 versions of the stepper pulse for spindle sync.
The delayed stepper pulse is only used when the direction changes. I assume that all axis are stopped or this could be a requirement.

If the delay is executed by grbl (not the driver), than it would require only 1 stepper pulse implementation.

terjeio, 2020-10-27:
Not sure I understand yor reasoning...

Moving the delay to the core would require a new HAL entry point for microseconds delays. This would need to be called from the stepper ISR? Adding a blocking delay there is going a against normal ISR coding practices - IMO ISR code should be as fast as possible. And a delay could be entirely implemented in hardware - e.g. the my ESP32 driver does that. Another issue may be that having a blocking microsecond delay while interrupts are enabled is going to be imprecise.

> The delayed stepper pulse is only used when the direction changes. I assume that all axis are stopped or this could be a requirement.

A normal delay will be in the region of a few microseconds, this would introduce a only little bit of step jitter on direction changes. And I assume direction changes will normally happen at relatively low step rates so I consider this insignificant. I addition there is likely to be a bit of jitter anyway if there are any higher priority interrupts serviced during the delay.

> If the delay is executed by grbl (not the driver), than it would require only 1 stepper pulse implementation.

What is wrong with having more than one? I switch between "normal" step generation and "spindle sync" step output dynamically in the MSP432 driver as the latter is more resource intensive. I also assume that spindle synced motions will normally be at a fairly low feed rate so this overhead does not matter then.

And AFAIK spindle synced motions will never contain direction changes, at least for threading?

HuubBuis, 2020-10-27:
This is what i have in mind for spindle sync threading:

All drivers need to signal grbl the receive of a spindle pulse and sync pulse. How they do that is driver dependent but the rest of the processing is (could be) the same for all drivers and could be done by grbl. The driver needs code to execute the stepper pulse, to change the direction, and delay the stepper pulse. All drivers can do this already ,it is just a matter of copying code (index and synchronization pulse) and splitting code (stepperPulseStartDelayed) code. 

> Moving the delay to the core would require a new HAL entry point for microseconds delays. This would need to be called from the stepper ISR?

The stepper ISR would call:
- the change the direction code (only if needed)
- the stepperPulseStartDelayed code when a delayed step is needed. the stepperPulseStartDelayed should call the stepperPulseStart code.
- the stepperPulseStart code when a delayed step is not needed.

This would not block the stepper ISR more than it does now. and it  would remove the need for a separate stepperPulseStartSynchronized.

> What is wrong with having more than one?
It is easier to implement spindle sync when the decisions on what code to use is made by grbl. This will result in less bugs, less testing and faster implementation.

closed 2022-01-27 by terjeio