terjeio/grblHAL #106: STM32F411CEU6 compile in STM32cubeIDE

creisey, 2020-10-28:
hi there i tried to follow up the guide for compiling but i get an error:

In file included from ../Src/st_morpho.c:23:0:
../Src/st_morpho.c: In function 'digital_out':
../Src/st_morpho.c:30:26: error: 'AUXOUTPUT0_PORT' undeclared (first use in this function)

anyone can help?

terjeio, 2020-10-28:
Ouch - sorry about this, is a regression from my latest update where I added my upcoming CNC breakout. I'll fix it soon.

terjeio, 2020-10-28:
fix comitted.

creisey, 2020-10-28:
worked like a charm!

creisey, 2020-10-29:
compile works, but grbl not...

i tried to connect over USB & Serial but neither worked.

if i connect the stm32 to the PC no virtual serial port appears.

creisey, 2020-10-29:
STM32F411CEU6, to be precise

terjeio, 2020-10-29:
On a Blackpill? Do you have a link to any data about the board used?

It could be that the clock tree is not configured correctly, likely an incorrect master oscillator setting. See the [driver page](https://github.com/terjeio/grblHAL/tree/master/drivers/STM32F4xx): there is a configuration for the crystal frequency, `HSE_VALUE` that might need to be changed to match your board.

terjeio, 2020-10-29:
I've just tested a Blackpill with a 25MHz crystal. On my development machine it did not show up and I could not see any code changes explaining it. Since I have an USB hub that plays games with me I plugged it into a laptop - and it showed up there and worked. When I brought it back to my development machine it showed up there as well - and worked... Spooky. FYI the last couple of days I have worked extensively on the F4xx driver with a F411 Nucleo-board using UART comms, no problems related to that.

Do you have another machine you can try your board with?

creisey, 2020-10-30:
> I've just tested a Blackpill with a 25MHz crystal. On my development machine it did not show up and I could not see any code changes explaining it. Since I have an USB hub that plays games with me I [...]
[image]
this is a picture of my board.
and i bought it here: https://de.aliexpress.com/item/4000138305460.html?spm=a2g0s.9042311.0.0.27424c4dh6OnLS

i'll try it on another machine.
[image]
no serial appears

terjeio, 2020-10-30:
Have you changed the STM32F401xC symbol to STM32F411xE as descriped in the driver readme? In fact you have to remove STM32F401xC and add STM32F411xE

creisey, 2020-10-30:
i did
[image]

terjeio, 2020-10-30:
Thanks, your board is not a [Nucleo 64 board](https://www.st.com/en/evaluation-tools/nucleo-f411re.html)... Remove NUCLEO_F411 and try again  - hopefully then it will work.

creisey, 2020-10-31:
thx for your assistance! after removing nucleo board it worked

creisey, 2020-10-31:
the io-GCode Sender is one of the best i've used so far!

closed 2020-10-31 by creisey