terjeio/grblHAL #107: Spindle Sync: Handling acceleration and deceleration

HuubBuis, 2020-10-29:
> I want to understand how the feedback loop is tuned in your and how you handle acceleration and deceleration.

If I understood the MSP432 code well:

- at every stepper pulse, a new feed rate is calculated by a PID algorithm
- acceleration and deceleration limitations depends on the PID settings*
- the feed rate change is set immediately

* This worries me , but correct me if I am wrong.

I couldn't get the MSP432 implementation working in the SAMD21 to test it, so I changed it.
In my SAMD21 implementation:
- At every stepperPulseStartSpindleSync()  the actual Z-axis position is recalculated (float subtraction)
- At the receive of a synchronization pulse the new feed rate is calculated based on the current RPM and the travel needed to be in sync at the next synchronization pulse.
- a flag is set so that protocol_exec_rt_system () can handle the feed rate change.
- protocol_exec_rt_system ()  handles the feed rate change by by calling 
```
        plan_update_velocity_profile_parameters();	
        plan_cycle_reinitialize();		
```									
This feedback loop doesn't require tuning.

terjeio, 2020-10-29:
> at every stepper pulse, a new feed rate is calculated by a PID algorithm

Not correct, at the start of a new segment the timing of it is adjusted if needed. So about every 5ms.

> acceleration and deceleration limitations depends on the PID settings*

The PID loop is disabled during acceleration and deceleration - otherwise it will try to override the actual feed rate in order to catch up with the spindle.

> the feed rate change is set immediately

yes - but according to the PID parameters, this is where tuning comes into play.

> This feedback loop doesn't require tuning.

So there is no smoothing of fast/noisy RPM changes?

HuubBuis, 2020-10-29:
> Not correct, at the start of a new segment the timing of it is adjusted if needed. So about every 5ms.

That is also the way i think would be best.

> So there is no smoothing of fast/noisy RPM changes?

No, the implementation is copied from the grbl-l-mega.
I think the spindle RPM is leading and if there is no way to distinguish between noise or real RPM changes, the Z-axis should follow as close as possible.
On my small lathe, the signal is noisy and it requires debouncing. After debouncing, the (no load) RPM is steady.

closed 2022-01-27 by terjeio