terjeio/grblHAL #108: Pi4/Mac UGS "Grbl has not finished booting" Error

wbatie, 2020-11-07:
I am trying to use Universal G-code Sender with grblHAL on a MAC and Pi4.
I get a connection, but when every I run a command I get the error "Grbl has not finished booting. I am using version "grblHAL-3XU-compat.hex" on a Teensy 4.1-GrblHAL board.
bCNC works fine on the same setup.ie
Any suggestions?

Thank you,
Bill Bag

phil-barrett, 2020-11-07:
Hi Bill.  Could you post the result of $i from the console?

HuubBuis, 2020-11-07:
To get UGS working you need to set the compatibility level to 1 in the file config.h (line 50)
replace #define COMPATIBILITY_LEVEL 0
to         #define COMPATIBILITY_LEVEL 1

phil-barrett, 2020-11-07:
Actually, the prebuilt "compat" ones on my github repository are set to level 10 (should still work). I tested it with UGS on windows and did not see the problem so I wonder if he still has the non-compat build.

wbatie, 2020-11-08:
looked in all the UGS files on the Pi4 and found several Config folders, but no config.h file
$i
[VER:1.1f(IMXRT1062).20200907:]
[OPT:VNMSLW,35,1024,3,0]
[NEWOPT:ES,TC]
[DRIVER VERSION:200907]
[BOARD:T41U5XBB]
ok

HuubBuis, 2020-11-08:
> looked in all the UGS files on the Pi4 and found several Config folders, but no config.h file

This is a grblHAL file located in the grbl directory. You have to change the file and flash the grblHAL controller again.

phil-barrett, 2020-11-08:
Bill, I just built the attached .hex.  UGS seems to be fine with it (G0 works ok, for example).  You will need to unzip it and use the teensy loader to get it onto the board.
[grblHAL_Teensy4_Upload-1.ino.TEENSY41.zip](https://github.com/terjeio/grblHAL/files/5506988/grblHAL_Teensy4_Upload-1.ino.TEENSY41.zip)

terjeio, 2020-11-08:
FYI if `[NEWOPT:...]` and any following lines are only added to the `$I` report when `COMPATIBILITY_LEVEL` is 0.
Do we need a `$`-command for reporting the selected level when it is > 0? `$I+`?

phil-barrett, 2020-11-08:
Yes, I was just thinking of posting something like that. Or perhaps just add a [COMP LEVEL: x] to $I?

terjeio, 2020-11-08:
> Or perhaps just add a [COMP LEVEL: x] to $I?

That might be enough to break compatibility...

I'll add `$I+` and report the same details as when `COMPATIBILITY_LEVEL` = 0 _plus_ the configured level.
I could even make the level settable at run time? But that is not as simple as adding `$I+`...

phil-barrett, 2020-11-08:
> I could even make the level settable at run time? But that is not as simple as adding $I+...

That seems like a great solution though I wonder if it's really necessary?

wbatie, 2020-11-08:
Nothing changed! $i works with bCNC, does not with UGS which gives the same error

terjeio, 2020-11-08:
> That seems like a great solution though I wonder if it's really necessary?

If I ever do it it will have to wait - there are 37 matches for COMPATIBILITY_LEVEL in the source so not done on a whim.

A better idea will be to request grblHAL support from sender developers? Basic support is for them to just write the protocol parser in a defensive manner.

phil-barrett, 2020-11-08:
> grblHAL support from sender developers?

Yes, that is the heart of my "is it necessary" comment.

terjeio, 2020-11-08:
> Nothing changed! $i works with bCNC, does not with UGS which gives the same error

$I reports the same data as before?

wbatie, 2020-11-08:
No. [DRIVER OPTIONS:USB.2] has been added
[VER:1.1f(IMXRT1062).20200907:]
[OPT:VNMSLW,35,1024,3,0]
[NEWOPT:ES,TC]
[DRIVER VERSION:200907]
[DRIVER OPTIONS:USB.2]
[BOARD:T41U5XBB]

phil-barrett, 2020-11-08:
> Nothing changed! $i works with bCNC, does not with UGS which gives the same error

Bill, have you looked at this? [UGS Not Finished booting](https://winder.github.io/ugs_website/guide/troubleshooting/#:~:text=Common%20Problems-,%22Grbl%20has%20not%20finished%20booting.%22,very%20old%20versions%20of%20grbl.)

wbatie, 2020-11-08:
Yes I have. Works  fine with bCNC!

wbatie, 2020-11-08:
Is the grblHAL giving a "Grbl" response. Some say that can be the problem.

terjeio, 2020-11-08:
Since the $I report contains NEWOPT and additional lines after that then `COMPATIBILTY_LEVEL` is still 0 in the controller.

As I write this I am running a gcode program with UGS version 2.0 from Aug 14, 2019 with `COMPATIBILTY_LEVEL` set to 1.

phil-barrett, 2020-11-08:
Yes, I am testing it right now with UGS Win 10 to a T41 board. No problems at all.

Bill, I have a RasPi and will set it up to see if I can reproduce the problem. It's going to take some time, though. I suggest in the mean time you use bCNC.

wbatie, 2020-11-08:
I am. It's just so hard to use.

terjeio, 2020-11-08:
> Bill, I have a RasPi and will set it up to see if I can reproduce the problem. It's going to take some time, though. I suggest in the mean time you use bCNC.

Is the Pi needed to flash the controller with a grblHAL version that has `COMPATIBILTY_LEVEL` set > 0?

phil-barrett, 2020-11-08:
If Bill flashed the hex file I attached earlier in this thread, he is running with compat level set to 1.

wbatie, 2020-11-08:
I used Teensy Loader on my MAC Pro

phil-barrett, 2020-11-08:
By the way, Bill, have you tried completely clearing out UGS and reinstalling it?  That is one of the "Not finished booting" suggestions.

terjeio, 2020-11-08:
If Bill flashed the hex file I attached earlier in this thread, he is running with compat level set to 1.

Well, NEWOPT etc. is still in the $I report so not so.

wbatie, 2020-11-08:
Yes, and I tried it on a Pi3 with same problem.

wbatie, 2020-11-08:
Also tried it on MAC Pro which uses the same software.

breiler, 2020-11-09:
@wbatie there is now a feature request over at UGS to get grblHAL working natively on UGS: https://github.com/winder/Universal-G-Code-Sender/issues/1468

I do not have a functioning device to begin working on this at the moment though, but when I do this should be a quick fix.

phil-barrett, 2020-11-19:
I have set up a RasPi 3 and connected it to a Teensy 4.1 breakout board.  I had trouble connecting to a COMPATIBILIY_LEVEL 0 build of grblHAL.  It would give the now infamous "not finished booting" error.  However, building for COMPATIBILIY_LEVEL 1 allowed a connection and I was able to run a GCode program successfully.  Will continue testing but I think that COMPATIBILIY_LEVEL is sufficient to get it running.

FWIW, I downloaded the UGS Platform latest release (V2.0.6).  Followed [these instructions](https://github.com/winder/Universal-G-Code-Sender/wiki/Raspberry-Pi).

breiler, 2020-11-19:
@phil-barrett thanks! I've just received the Teensy 4.1, so whenever I get the time to look at this I will start fixing these problems in UGS so that it will work with `COMPATIBILIY_LEVEL 0`.

phil-barrett, 2020-11-19:
Excellent!  Hope you receive the T41 BOB soon.  US Postal Service tracking does not give a lot of info after it left the country on Nov 12. Hard to tell how much longer it will take.  

Let us know if you need any help.

terjeio, 2020-11-19:
@breiler FYI the next build of grblHAL will get some additional $ commands for fetching details about current settings, error codes and alarm codes. Perhaps build info too. Output from these can be used instead data from the csv files that has to be kept in sync with the controller - not easy for end users. 

Setting details will contain information about datatype, format, max and min value allowed etc. This can be used to build a settings UI similar to the one in ioSender. Settings grouping is on my todo list - when done it can be used to create a treeview of the settings.

breiler, 2020-11-19:
I have just merged a small fix to make grblHAL to work with UGS and it looks like its working decently. 

I noticed a couple of things that we need to keep working on:
* The [setup wizard](https://github.com/winder/Universal-G-Code-Sender/wiki/Configuration#setup-wizard) assumes that the setting for inverting limit switches ($5) can be either 0 or 1, but the new version uses a bit mask.
* Arduino UNO usually reboots when connecting to the machine but the Teensy does not, so there is no boot string that lets UGS know what it has connected to. The only way to fix this is to disconnect/connect the USB cable.

The new changes should be available in the nightly build:
https://github.com/winder/Universal-G-Code-Sender/#downloads

phil-barrett, 2020-11-19:
That's good to hear.  

Would it be possible for grblHAL to resend the boot string when it sees a connection? Does it know connection status?

Joacim,  are you considering supporting an Ethernet connection? Telnet might not be too hard.

terjeio, 2020-11-19:
> Arduino UNO usually reboots when connecting to the machine but the Teensy does not ...

This is a tricky part with 32-bit ports, with USB or a network connection a hard reset is not possible. A soft reset via the grbl reset pin may do the trick though. To complicate matters grblHAL has an option of connecting an advanced DRO/MPG that may take control too...

ioSender startup is done this way:

At connection listen to incoming traffic for a short time, 500mS or so. If real time reports are detected the sender starts up in wait-mode waiting for the MPG/DRO to release control.

If out of wait mode or silence I send a real-time report request. If a response is forthcoming further steps can be taken. If not in an Alarm state query the controller with a `$I` command to determine what it is.

If in an alarm state then what? Some alarm states are defined as critical and and no action except a reset is possible - depending on settings. A problem for me was that I could not find out the reason for the alarm, and resetting the controller for no good reason is something I do not like to do. To improve on this I have added an option to add the alarm code as a substate to the `Alarm` status text, in the same vein as `Hold` or `Door` statuses in vanilla grbl. The alarm code can then be used for deciding next steps. If not critical I proceed with a `$I` command to find out what I am talking too, load settings, parser status and request a full status report so I can configure the sender for action. A full status report is an extension is grblHAL, added since vanilla grbl sends some statuses only on changes. This is ok if the controller can be rebooted, but again not always possible with 32-bit controllers.

---

For safety reasons grblHAL has configurable options for more strict behaviour than grbl in some situations. E.g if reset or estop inputs are asserted then it will not be possible to continune before the switches are deasserted. Same goes for hard limits asserted and homing required on startup. grbl allows a ctrl-X followed by $X to unlock the controller, maybe ok for small hobby machines but not for larger and likely dangerous ones.  A sender should be able to handle this.

---

I should really start a Wiki page for sender developers and spell out the reasoning behind some of the extensions. Some, as mentioned above, is due to the different behaviour of 32-bit controllers, some are for safety reasons and some to add functionality that having of plenty of RAM and Flash available allows.

I have tried to follow the style/intentions in original protocol specification in order to make it simpler for sender developers to catch up. IMO "defensive" protocol parsing will go a long way to achieve that - and will ensure that any future extensions does not immediately render a sender unusable.

An issue for sender developers going forward is that ports may implement protocol extensions in different ways. IMO this should be avoided, possibly by having a separate project handling the protocol specification. A kind of standards body. And the protocol version should be well defined element in the $I output...

terjeio, 2020-11-19:
> Would it be possible for grblHAL to resend the boot string when it sees a connection? Does it know connection status?

Not generally possible for a UART connection (though some supports  a hard reset via RTS and some via DTR), sometimes possible for USB and always possible for a network connections. If a MPG/DRO is in control then sending one is not good as the boot string implies a hard reset when received out of the blue.

HuubBuis, 2020-11-19:
> Arduino UNO usually reboots when connecting to the machine but the Teensy does not, so there is no boot string that lets UGS know what it has connected to. The only way to fix this is to disconnect/connect the USB cable.

Currently I use the build options ($I) to identify the connected grbl implementation. 
Couldn't grbl and clones use a unique number for uniform identification. A 32 bit hex, 16 bit for the the main version and 16 bit for sub versions. A git repository could be used to release the number for the main version and the developer can assign a sub version when appropriate.

breiler, 2020-11-20:
> Joacim, are you considering supporting an Ethernet connection? Telnet might not be too hard.

@phil-barrett We currently have support for connecting to TCP enabled devices, but I have never tried it myself. Also the feature is a bit hidden in the settings so it would need some UX attention to be useful:
https://github.com/winder/Universal-G-Code-Sender/wiki/Connecting-the-Controller#smoothie--network

@terjeio thanks for the input of detecting the version of GRBL when connecting. I have some ideas on how to change this in UGS to allow for auto detecting different GRBL dialects and to load its implementations. But it will require some rework of the code to make this work.

HuubBuis, 2020-11-20:
> @phil-barrett We currently have support for connecting to TCP enabled devices, but I have never tried it myself. Also the feature is a bit hidden in the settings so it would need some UX attention to be useful:
https://github.com/winder/Universal-G-Code-Sender/wiki/Connecting-the-Controller#smoothie--network

I have used this feature on grbl & mega controllers works OK. I am currently testing on  grbl_ESP32.

HuubBuis, 2020-11-20:
UGS build Universal Gcode Platform 202011193 is working (telnet bench tested) on grbl_ESP32 build 
```
[VER:1.3a.20201004:]
[OPT:VNMPHBSW]
[MSG:Using machine:ESP32_V4]
```
I get an error at startup, not sure the cause of it. This error is also reported by an older UGS build 04 2020.

```
ok
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
[Error] Error while processing response <ok>: An unexpected command was completed by the controller
```

I don't have telnet working on grblHAL ESP32 yet but it is on my todo list.

I also did a short bench test on grblHAL SAMD21 at compatibility level 0, no problems found yet.

breiler, 2020-11-26:
I think that it would help to get a firmware variant from the $I-command, this would make things more simple to load a proper dialect implementation in the sender, eg: `[FIRMWARE: grblHAL]`

phil-barrett, 2020-11-26:
This is a good idea.  I would like to see compatibility level added to this. something like
`[FIRMWARE: grblHAL:1]`
If CLevel is 0, it could be left off so just
`[FIRMWARE: grblHAL]`
That would allow debugging of problems related to sender variants easier.

terjeio, 2020-11-26:
> I think that it would help to get a firmware variant from the $I-command

A problem is compatibility with senders, not all are going to be updated, how to handle that? Fail early or fail late? Always report as "grbl"?

I can of course add [FIRMWARE: grblHAL], that would make it easier for me too. But adding that regardless of compatibility level may break senders. Catch 22? Add to the new `$I+` command instead?

[COMPATIBILITY LEVEL:n] has already been added as a response to `$I+`. I am not sure senders needs this - IMO more relevant for support.
Adding partial sender support according to compatibility level should be avoided - it is for backwards compatibility with senders not forward.

Not easy this...

terjeio, 2020-12-02:
@breiler Am I loosing it? I have started writing a sender developers wiki page and while doing that I got back to the code and then I see that `$I` does not provide the neccessary information. I am sorry for that.

The way I detect grblHAL in my sender is to check for the presence off the `$14` setting, not a good approach I know.

Perhaps a better way is to request the full real-time report request by sending `0x87` and wait a little for a reply? Legacy grbl does not respond to that and the character is filtered out from the input stream so it is safe to do so . If a reply is forthcoming one can be sure that it is not a legacy grbl controller one is communicating with.

The easiest way is of course to ignore the protocol specification and add the neccesary information to the `$I` response and tell users of senders that does not like that to update. What do you think?

BTW I have added `[FIRMWARE: grblHAL]` to the `$I+` response (and `$I` response when compatibility level is 0) in the upcoming build.

breiler, 2020-12-02:
@terjeio no worries 😄 

I have discussed this with the Grbl_Esp32 team, and they suggested adding a `[MSG:FIRMWARE:grblHAL]` (or something like that) to the `$I`-command which would not break compability. 

@MitchBradley had some ideas that he suggested he would write up in a specification. So you might want to wait on adding the `[FIRMWARE: grblHAL]` message so that we might get a unified way to do this?

terjeio, 2020-12-02:
> adding [MSG:...] ... to the $I-command which would not break compability.

Short of checking all senders this will not cut it. The specification for the `$I` response is that it is delimited by a `ok` at the end. Adding [MSG:...], or anything else for that matter, inside that response is IMO _not_ in compliance with the spec and is logically wrong as well as push messages is something that is to be displayed for the user to see.

From the [Wiki](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface):

> ... These build info messages are followed by an ok to confirm the $I was executed, like this:

```
[VER:1.1d.20161014:]
[OPT:,15,128]
ok
```

A unified way to do this would be nice though - but please not by adding push messages to request responses...

MitchBradley, 2020-12-04:
My first draft of a proposal addressing the identification issue is at https://github.com/bdring/Grbl_Esp32/wiki/Grbl-Version-Reporting---Proposal .  Comments are solicited, either here, or I can create a specific ticket somewhere.

terjeio, 2020-12-04:
@MitchBradley 

> When a new connection is established on an interface that is intended for use with the grbl serial protocol, the grbl controller shall issue a welcome message.

Not possible for a number of the processors/cards grblHAL supports. This means that getting the welcome message is not possible short of issuing a soft reset by ctrl-X.

To complicate matters many grblHAL drivers also supports my MPG extension where a secondary input only port is used when in MPG mode. The primary input port that a sender connects to is then inactive and any queries received will be discarded. A sender can (fairly reliably?) detect this mode by listening for real-time reports for a little while after connecting. 

Perhaps a further complication is that my [MPG implementation](https://github.com/terjeio/GRBL_MPG_DRO_BoosterPack) uses the welcome message to detect an external controller reset as this is AFAIKT the only way to reliably do so.

So IMO the welcome message is useless and cannot be used for identification. 

I have added a new real-time report request character to grblHAL, `0x87`, that could be used to detect an extended version of the protocol. I added this so my sender could get a complete status when required, this since some elements are only added to the report on a change. If a reply is received then an extended version of the `$I` report could be used for getting the neccesary details. I have added `$I+` for that.

breiler, 2020-12-04:
@MitchBradley thank you for your time writing up the spec!

@terjeio do you have a proposal for the specification of `0x87` and `$I+`? It could solve a lot of things by introducing a `$I+` that would allow any [key:value] combination to be extended with in the future. 

About the welcome message, could it be solved by adding a real time command for requesting a new one? This could be used if the sender wasn't greeted with one upon connection? The response could also only be sent on the channel that requested it (primary, secondary, network etc).

FYI I have invited Sonny Jeon to this thread, hopefully he will have some time to share his insights.

terjeio, 2020-12-04:
> do you have a proposal for the specification of 0x87 and $I+?

`0x87` just returns a complete real-time report including any on-change only elements and extensions. IMO this is needed regardless as it is not desirable to soft-reset the controller on connect. 

`$I+` returns additional information in two categories - one aimed at senders for configuration the other aimed at support. An example from the controller I have running now:

```
[VER:1.1f.20201201:]
[OPT:VNMSL,35,1024,3,0]
[NEWOPT:ENUMS,*EEPROM,ES,TC,SS,PID]
[FIRMWARE:grblHAL]
[DRIVER:MSP432]
[DRIVER VERSION:201125]
[BOARD:CNC BoosterPack (Trinamic)]
[PLUGIN:TMC2130 v0.01]
[PLUGIN:MODBUS v0.02]
[PLUGIN:HUANYANG VFD v0.02]
[PLUGIN:ODOMETERS v0.02]
ok
```
`NEWOPT` and possibly `FIRMWARE` is for sender configuration, the rest after that is for support(?).
`NEWOPT` options should be from a standard list? 
Having a support category means that anyting needed for that can be added without breaking the extended protocol.
Note that I do not like the idea to use `[MSG:...]` as part of this response. 

> About the welcome message, could it be solved by adding a real time command for requesting a new one?

I do not think this is needed, and would require another message to indicate a reset? It would break backwards compatibility too? A separate channel for the repsonse is not neccesarily available when UART communication is used, for some of my drivers only the Rx pin of the secondary UART is brougth out, Tx from the primary is shared.

`0x87` paired with `$I+` and `$+` (for requesting all settings) is IMO a clean way to bring the protocol forward and still maintain backwards compatibility. The exception will be those senders that relies upon getting the welcome message on connect, this will no longer be universally true and is not always possible to fix on the controller side due to hardware or firmware limitations.

A `0x87` request _could_ be used by the controller as well, if received the controller then knows that the sender is capable of handling protocol extensions?

MitchBradley, 2020-12-04:
I propose that we discuss one subtopic at at time. In the standards committees that I have been on, the only way we could make progress was to start with a draft, then break it down into small subtopics and discuss each one individually.  I think the draft sections provide a reasonable list of subtopics, namely

* Welcome message format
* Triggering welcome messages
* Identification message format
* List of standard ID messages
* Triggering ID messages
* Guidelines for sender implementers

I suggest that we start with "Identification message format".

MitchBradley, 2020-12-04:
On the topic of "identification message format":

I proposed the [MSG:tag: value] encapsulation because [MSG: ...] is already part of the legacy protocol, so it is likely to work with a lot of existing senders.  Old senders will not understand the information contained in the tagged messages, but they will probably not break at the protocol parsing level.  The likely result is that the sender will just display the message somewhere that the user can see.  Even if the sender cannot directly use the information, the user can see it and report it to support people.

One alternative is to dispense with the [MSG: prefix and just use [tag: value] directly.  That is more likely to confuse legacy senders that may not be prepared for arbitrary messages.

Discussion?

breiler, 2020-12-04:
Upsides by using the [MSG:]-encapsulation:
1. easily extendable
2. should not effect current sender implementations

Downsides by using the [MSG:]-encapsulation:
1. we are building on the legacy protocol which could be confusing over time
2. we would be using it in an unintentional way given the original protocol
3. need a secondary identification system for controllers that doesn't yet use the message tag, i.e welcome message

Upsides by creating an extended/new command (i.e. `$I+`)
1. easily extendable with a new protocol

Downsides:
1. More steps for protocol negotiation ([welcome message + `0x87`] + `$I+`)
2. Need a secondary identification system for controllers that doesn't yet use it, i.e welcome message

MitchBradley, 2020-12-04:
> we are building on the legacy protocol which could be confusing over time

I'm not sure I understand this point. I don't see how the addition of tag structure to MSG: is any more confusing over time than the addition of new messages that are not prefixed with MSG:.  It seems the same to me with respect to future evolution.

> we would be using it in an unintentional way given the original protocol

It boils down to whether or not MSG: values that are not in the set that is listed on the Grbl wiki confuse senders.  There is evidence that it does not, namely the fact that Grbl_Esp32 uses MSG: for a lot of new messages and is known to work with a lot of senders.

> need a secondary identification system for controllers that doesn't yet use the message tag, i.e welcome message

That is true regardless of whether new information is encapsulated in MSG: or not.  Old controllers do not use either format. Senders that wish to handle old controllers have to adapt using whatever information is already available.  

> Upsides by creating an extended/new command (i.e. $I+)

I think that is more properly discussed under the "triggering ID messages" subtopic.  I think that a new trigger like $I+ is a good idea.  Perhaps you are suggesting that new messages can only appear in response to the new trigger, so we don't need to shoehorn into an existing protocol loophole.  That is plausible, but less valuable, I think, than allowing messages to appear at other times, guarded by an already-existing safe encapsulation.

terjeio, 2020-12-04:
> I suggest that we start with "Identification message format".

Ok, IMO that is what I did with my previous message.

>  Old senders will not understand the information contained in the tagged messages, but they will probably not break at the protocol parsing level

Old senders should never see the new format - if not updated they are stuck in the past. What is needed as step one is for senders that are actively developed to be able to tell what they are talking to so they can make use of new features. One way to do that is to add a request character. Since I had to add one, `0x87`, for my sender to get a complete status report from grblHAL this could be used.

> An additional advantage of the [MSG: encapsulation is that they are asynchronous, so senders are supposed to handle them at any time. That further increases the chances that they will not break existing parsers.

Parsers needs to be updated anyway to handle extensions to messages. Guidelines for how to write parsers "passively" should be published - if followed protocol extensions should not cause them to fail.  

> One alternative is to dispense with the [MSG: prefix and just use [tag: value] directly. That is more likely to confuse legacy senders that may not be prepared for arbitrary messages.

[MSG... is a push message and, as I wrote above, I do noot like the idea of including them in response messages. By adding $I+ (and $+ for settings as I understand Grbl_Esp32 already has) legacy senders does not see any protocol extensions at all. The challenge then boils down to how a sender can get to know what kind of controller it is talking to. Requesting $I+ and wait a bit for a response could be one solution, another is the full status report request and there may be others.

> More steps for protocol negotiation ([welcome message + 0x87] + $I+)

IMO forget about the welcome message, it cannot be used without causing trouble. For some of the processors/cards that grblHAL supports it is not possible nor desirable to do a hard or soft reset on connecting. And I do not want to exclude them from the fold. If legacy sender does not handle that then the user has to switch to a compatible sender or a different processor/board.

> Need a secondary identification system for controllers that doesn't yet use it, i.e welcome message

I do not understand this statement. If a controller does not use the new protocol then it should work as before - at least for processors/cards that is hard reset on connect? BTW I have made it configurable/selectable in my sender how a serial connection is initiated: toggle DTR, toggle DTR or do nothing. This means the user can decide whether a hard reset is performed on connect for cards that support that.

terjeio, 2020-12-04:
@MitchBradley Do you understand the passage _These messages are "pushed" from Grbl and may appear at anytime._  in the [interface specification](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#grbl-push-messages) to mean that these are even allowed in request responses? And handled correctly by all senders? IMO then the welcome message, a full realtime report and an alarm message (if an alarm is active) can be added to a $I request without breaking the protocol...

MitchBradley, 2020-12-05:
"All senders" is an impossibly high threshold because there are so many of them, some of which are very badly written and break even with some classic GRBL implementations.  I do know that Grbl_Esp32 uses [MSG: ...] extensively to report lots of configuration and debugging information beyond the gnea list of [MSG: variants, and that does not seem to cause sender problems.  (The senders that have problems with Grbl_Esp32 tend to be ones that expect very specific welcome messages, and ones that expect an automatic welcome message; those are problems to address separately).

I would not advocate adding welcome, realtime status, and alarm to $I, since none of those are included in the "may appear at any time" section.  Especially not welcome, because it indicates that GRBL is in a "reinitialized state".

If the concern is that including [MSG: between $I and ok might be dangerous, that could be solved by issuing the [$MSG reports after the ok.

MitchBradley, 2020-12-05:
The Grbl 1.1 message format is explicitly designed to be context-free - c.f. [Message Summary](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#message-summary)

>In v1.1, Grbl's interface protocol has been tweaked in the attempt to make GUI development cleaner, clearer, and hopefully easier. All messages are designed to be deterministic without needing to know the context of the message.

I think that feature is very important, since it greatly simplifies the response parsing task, decoupling it from the sending task in event-driven senders like CNCjs.  Having new messages that can only be understood in the context bracketed by $I+ .. ok violates that core philosophy.

I do not have a problem with the addition of a new command like $I+ for the purpose of triggering some messages, but I don't think that it should create a parsing context for understanding the messages.  Context-dependent parsing can cause all sorts of timing/sequencing dependencies, especially when you consider the possibility of multiple senders listening on different interfaces.

breiler, 2020-12-05:
Both are making strong cases and we might not be able to agree on anything. The thing is that people from grblHAL reached out to me because we should implement support for it in UGS whereas Grbl_Esp32 just worked seemingly complying with the legacy protocol extending it carefully.

GrblHAL has in my eyes already broken stuff, for instance:
* $5 which now is a bitmask instead of a boolean in (compability mode 0)
* Not able to always send the welcome message upon startup (which by the way was the recommendation by Sonny)
* When using a MPG it's assumed the sender are asking for a status reports to figure out if the sender is allowed to send messages (if I understood its usage correctly from the sparse documentation)

This is all good stuff with good intentions, but if it can't be used by other senders you have a lot of work in front of you implementing both the firmware and sender. 

Even if it is maybe semantically not correct to use the MSG-tag for sending version data it looks like it's already working in the Grbl_Esp32 implementation. Which is why I'm leaning towards that solution because I believe that, in the end this will attract a larger user group - the users don't care about protocol stuff they just want their machines to work.

The MSG-solution is well written up and is something that I can work with. The grblHAL-solution requires a bit more work from our side but can probably eventually be done. I would appreciate if you (@terjeio) could write a guide for the protocol negotiating with GrblHAL.

Thank you both for your time trying to solve this!

[26 more replies]

closed 2022-01-27 by terjeio