terjeio/grblHAL #109: ALARM at bootup on DUE

sokertsv, 2020-11-09:
Hello,
I am in the process of developing a software which allows me to control a pick and place machine very specific to my needs (camera, registration of coordinates ...). 
The problem is that when I start arduino due it goes into alarm from the start. the same by disabling the parameters homing, homing cycle etc ... ($22=0 etc...)
My software works well (IDLE ok and JOG works) only when I launch "GCode Sender.exe" and execute a reset, then close "GCode Sender.exe" and restart my software (without reseting the arduino). 
without this the controller status remains in ALARM systematically.
what does "GCode Sender.exe" do more than my software at startup ? I dont know.

In some cases I see ALARM: 0 but I can't find any documentation for this one .... 

if someone has an idea I would be very grateful to him

phil-barrett, 2020-11-09:
A guess but what are you grbl settings for estop, door, etc.?  grblHAL will go into alarm when it thinks the door is open  or estop has been pressed. You can try inverting door and estop in settings: grbl or you can jumper them on your due'.

sokertsv, 2020-11-09:
thank you phil-barrett !
just uncomment  #define INVERT_CONTROL_PIN_MASK SIGNALS_BITMASK 
in config.h
it works fine !

phil-barrett, 2020-11-09:
You can also do that in Settings:Grbl in ioSender.

phil-barrett, 2020-11-09:
by the way, this issue probably should be closed.

phil-barrett, 2020-11-09:
sorry, hit the button accidentally.

closed 2020-11-09 by sokertsv

terjeio, 2020-11-09:
> In some cases I see ALARM: 0 but I can't find any documentation for this one ....

This is a minor bug. Will be fixed in next build.