terjeio/grblHAL #110: HOW TO COMPILE

irshadmgs011, 2020-11-10:
I reade wiki but i don't understand how to compile code, i want to compile SAM3X8E

terjeio, 2020-11-10:
Did you try to follow the instructions in in [this Wiki page](https://github.com/terjeio/grblHAL/wiki/Compiling-GrblHAL)? If so what exactly didn't you understand?

irshadmgs011, 2020-11-11:
@terjeio Yes i read but i don't understand, this instruction is not simple, just like merlin and esp32 grbl, Can you make a video tutorial

terjeio, 2020-11-11:
I have made a .zip-library for you, import this in your Arduino IDE with _Sketch > Include Library > Add .ZIP library..._
After doing that you should find _grblHAL for Due_ under _File > Examples_ for you to compile and upload.

[grblHAL_Due.zip](https://github.com/terjeio/grblHAL/files/5526641/grblHAL_Due.zip)

irshadmgs011, 2020-11-12:
[image]
@terjeio Thanks

closed 2020-11-12 by irshadmgs011