terjeio/grblHAL #112: Backlash not showing in $$ report

LaurentLM, 2020-11-11:
Hi,

I have compiled GRBL Driver STM32F401 with Backlash compensation enabled in config.h
This was then flashed on a Nucleo F446RE.

For some reason the backlash settings ($160 to $162) do not show in the $$ result. The last setting I see is $132=200.00
I seem to be able to set it correctly ($160=0.05 returns ok).
Has anyone encountered this issue?

terjeio, 2020-11-11:
> Has anyone encountered this issue?

You could well be the first to use this. The missing settings report is due to a regression, for a temporary fix can you change the `AXIS_N_SETTINGS` value in line 37 below from 6 to 7? I'll fix this in the next commit.

https://github.com/terjeio/grblHAL/blob/b7863758a81774dbd8ecd3d31aa2ca2da9676d40/grbl/settings.h#L36-L40

LaurentLM, 2020-11-11:
I tried to fix it myself but I definitely wasnt looking in the right place. 
Your fix worked, thanks a lot!

closed 2020-11-11 by LaurentLM