terjeio/grblHAL #113: Y limit works for homing but not as hard limit

LaurentLM, 2020-11-14:
Hello,

Thanks again for this great piece of code. 
I am having a strange problem with my limit switches. I am using a F446RE with the Arduino UNO mapping (using a screw terminal shield).

The Y switch works for homing and the pin change registered by grbl when teh limit is activated outside of homing. The issue is that it doesnt doesnt trigger the alarm state and allows the machine to crash into the switch. 
The X axis behaves normally and triggers an alarm when the limit switch is activated. I then have to reset the machine (or home) to get rid of the alarm state.
I do not have limits/home sensor wired for the Z axis to investigate further.

Here are the $$ settings I am using:
[grblHAL_settings.txt](https://github.com/terjeio/grblHAL/files/5541074/grblHAL_settings.txt)

Many thanks!

terjeio, 2020-11-14:
Ouch, this is a bad one. I followed the standard grbl shield mapping for the UNO and forgot about a STM32 limitation where only one pin number from any given port can be used as an interrupt source. Y and Z limits are connected to D10 and D12 when variable spindle is enabled, these corresponds to pin PA6 and PB6 - and then only one of them can have an interrupt assigned. Z "wins" because that is set up last.

So not possible to fix without mapping either Y or Z to a pin that can be assigned to an interrupt. A piece of wire from one of them to a pin not used by the Arduino shield, e.g. PA6 to PA11 or PA12 and remapping the Z limit could be a solution?

HuubBuis, 2020-11-14:
I have "the same" problem on RAMPS shield where there are no interrupts for the limit pins. I am going to solve it by polling these limit pins when the processor is idle.

How about polling these pins in the system tic routine and generating the limit pin interrupt (when not homing). A 1 ms slow response is always better than none!

phil-barrett, 2020-11-14:
Do limit pins actually NEED debouncing? Once you have seen a limit switch event and stop the machine, the switch bouncing will have stopped.

HuubBuis, 2020-11-14:
> Do limit pins actually NEED debouncing?
On one of my lathes, the Z-axis limit pins is activated as soon as I power the spindle. I couldn't solve it without a 20 ms software debounce. I now use software debouncing on all limit pins on both lathes.

terjeio, 2020-11-15:
> How about polling these pins in the system tic routine and generating the limit pin interrupt (when not homing).

Next build to be committet to the test branch has polling enabled for the Uno map - it seems to work. Thanks for that suggestion.

> Do limit pins actually NEED debouncing?

IMO debouncing limit pins is generally a bad idea. If needed then there are electrical interference that should be routed out. The legacy grbl method of using NO switches as default and usually(?) only the MCU pullup resistor to supply current to the switch is bad engineering. Since limit switches doubles up as homing switches and are polled when homing interference/bad switches will likely make the home position less repeatable too.

The pullup resistor is usually in the 50K range and may deliver [insuficcient current](https://en.wikipedia.org/wiki/Wetting_current) for proper switch operation. 5V / 50K = 100 µA. And the load seen by any interference is IMO way to small. NC switches should be the norm (grblHAL defaults to that) and the switch current should be many times more than a MCU pullup resistor can provide. Typically I use optcouplers and/or photointerrupters in my designs, with switch current around 10mA.

HuubBuis, 2020-11-15:
> Since limit switches doubles up as homing switches and are polled when homing interference/bad switches will likely make the home position less repeatable too.

That depends on the seek rate and the positioning, type etc of the switches. I spend about a 8 months to get the repeatability better than 0.005 mm  on the lathe X-axis (using 2 proximity switches).

> Typically I use optcouplers

I also use optocouplers but have limited the current to 3mA. In the next design I will try your 10mA.

If I remember correct in the original grbl code, during homing limit switches are polled and debouncing is not used!

LaurentLM, 2020-11-15:
> Ouch, this is a bad one. I followed the standard grbl shield mapping for the UNO and forgot about a STM32 limitation where only one pin number from any given port can be used as an interrupt source. [...]

I'm impressed, that's some very good troubleshooting :-)
Thanks for the clear explanation, I'll remap the Y limit to some unused pins (Z limit or some other morpho pins).

terjeio, 2020-11-15:
> I'm impressed, that's some very good troubleshooting :-)

No so, I suspected this immediately as I have been designing a [Nucleo-64 breakout board](https://github.com/terjeio/CNC_Breakout_Nucleo64) recently. I had to reassign pins a lot of times due to the same constraint to get it right.

Note that the coming test branch commit let X or Y "win" instead of Z, and polling will be enabled for Z as default. 
If not in a hurry it is better to wait for that? I guess I'll do the commit tomorrow.

terjeio, 2020-11-16:
[Test branch](https://github.com/terjeio/grblHAL/tree/test) now updated.

LaurentLM, 2020-11-17:
Just tested it, it's all working fine. Thanks!

closed 2020-11-17 by LaurentLM