terjeio/grblHAL #115: Sender Compatibility:  bCNC at COMPATIBILITY_LEVEL 0

phil-barrett, 2020-11-19:
Not an issue per se but rather good news.  I am running a Teensy 4.1 grblHAL 4 Axis configuration with COMPATIBILITY_LEVEL unset (i.e. 0) . bCNC running on a Raspberry PI 3 is working without any problems. I will test out some longer GCode programs and report if I have any problems.

FWIW, I followed the [RasPi instructions here.](https://github.com/vlachoudis/bCNC/wiki/Installation)

einencool, 2021-03-01:
Hi Phil,
Did you have any problems with bCNC and GRBLHal?
Your 4-Axis Config, is this with a ganged gantry?
If yes, do you have one Limit switch for both motors, or two switches for „squaring“ it?

I used the software in the past, but since I‘d setup a new Laptop for the machine, I didn‘t want to install bCNC and then I see, that it doesn‘t work ;-)

But when you say with the Comp_level = 0 it is working, then I think I‘d give it a little try...

Sadly the Jogging Feature is not very good in bCNC, but the NC-Editor and the visualization of the „milled“ area is also really good.

phil-barrett, 2021-03-01:
I have not seen any problems but do not use it that much. I currently do not have a ganged axis drive machine.  I do have an Avid Pro4848 on order which has a dual motor Y Axis.  Will test out bCNC with it though I plan to use ioSender.  The Avid will have limit/home switches for each motor on Y and plan to use auto squaring. Avid is fairly backlogged, it may be another month or 2.

einencool, 2021-03-01:
Ah ok, that sounds interesting, maybe you can than do another test.
I think you understand much more from this than me :-)

So if you remember in this, it would be nice, if you could make a little test for squaring the gantry.

BCNC could also just be used to edit the nc-file in a simple way :-)

avizienis, 2021-03-03:
I'm using bCNC with no problems at all. The only inconvenience is that the configuration tool isn't compatible with grblHAL. But I used iosender to config the controller the first time, then individual adjustments are easy enough through the terminal.

einencool, 2021-03-03:
Great, thanks for your reply.

Both Senders have really great features, and in the moment I think the IO-Sender is the one to go.
For me these points are interesting :
Pro IO-Sender 
- Great Probing functionality 
- Great Jogging feature e.g. With Xbox Controller 
- Small program 
- Good configurable 
- See limit switch states etc

Pro bCNC
- different colors in 3d View for milled areas
- built-in NC editor is absolutely great
- Programm is highly configurable 
- 3d View is better to follow than in IO-Sender (for my opinion)
- Many options for simple programs like surfacing drilling etc.

- Negativ in my opinion is the way to install the bCNC software, because it's not so easy for "normal windows users"

If the first two points from the bCNC software would be implemented into IO-Sender, that would be great

avizienis, 2021-03-03:
Btw bCNC also shows activated limit switches/probe in the status bar top left (e.g. IDLE[x] for triggered x limit). For me it's an easier choice as I'm running a Linux machine. bCNC runs natively and for iosender I have to fire up a virtual machine. Doable but inconvenient.

einencool, 2021-03-03:
I installed it today on my laptop and did some small tests side by side with both programs.

The great thing is, that both programs are running without any issues, so I didn‘t have to change the Firmware compatibility level or something like this.

When I changed between the Programs (close the first and then start the second), it was nice, that both keeps the status like „homed“ or all the coordinates. 

And like I mentioned before, the Jogging and Probing features from IO-Sender are amazing, When I then switched to bCNC and had to move the machine step by step for maybe 100mm it was a little bit scary, when the machine runs always with the full Speed from the machine. Since the acceleration is on my new machine much higher, than on my old one, it‘s not the way I want to jog...

But the NC-Editor from bCNC is unbeatable and absolutely amazing. I took an old program and deactivated one section within some seconds and in the window you could directly see which area is choosen. 

@terjeio is there not a easy way that you could talk to @vlachoudis or @Harvie for the Editor part :-) ?

Harvie, 2021-03-03:
@einencool If you can implement proper joging in bCNC the PR will be massively welcome :-)
BTW if i remember correctly there were some attempts to provide HTML5 joystick for highly interactive jogging using the web (or smartphone) pendant which is part of bCNC. But i don't remember if it ended up being merged or not...

einencool, 2021-03-03:
@Harvie , if I could help, I would like to do it, but I don‘t have any idea from coding :-(

Changing some values are not the problem, but I have absolutely NO Idea what you amazing guys are doing there 👍 
That‘s always fascinating for me, to see, what is possible when you know what you‘re doing

The Probing Section from IO-Sender is also Genius, and since I have a 3D-Finder, I don‘t want to miss these features. And that‘s also one little point against bCNC.
With my old homemade CNC I used bCNC for around two years, and I was always happy, also when I had sometimes Problems with a hanging bCNC, but I think that you helped me a lot with these problems 🥇

terjeio, 2021-03-03:
@einencool My immediate plan for ioSender is to get to the first production release - until then I will only add fixes within the current feature set. Some of the changes you want are already in the [roadmap](https://github.com/terjeio/Grbl-GCode-Sender/wiki/Roadmap), others I will have to check out to see how they fit in.

grblHAL has taken a lot of my time lately, I hope this will soon slow down so that I can spend some time on the sender without beeing distracted too much.

closed 2022-01-27 by terjeio