terjeio/grblHAL #119: Job just stopped

phil-barrett, 2020-11-20:
I was running one of my stress tests (Concentric Circles as Segments - 9 or more hour run time) on the new Pro board and it ran for 58 minutes.  Then, it stopped.  State was Idle.  Console shows this:

> [MSG:Pgm End]
> |MPos:107.324,231.504,-13.500,0.000|Bf:0,1023|FS:1500,1000|Pn:XYZA>
> 100,ok
> [MSG:Restoring spindle]
> [MSG:]
> [MSG:Restoring spindle]
> [MSG:]
The first line (Pgm End) was from a previous run. I had not interacted with the system (other than positioning the 3D View point of view) after starting the job so the 100,ok was not in response to anything I did. I have LEDs on the stepper signals and they show the enable lines high (disabled). Pressing feed hold followed by cycle start did nothing. I noticed Run Time display in the lower left corner is still going.  I will leave it alone for a few hours if you want me to look at anything.

4 Axis, ethernet, qei enabled build of the master branch (ethernet is not used)

phil-barrett, 2020-11-21:
More info.  I've rerun this twice and it exits around the same spot - just short of 60 minutes.  Pretty consistent.  The USB serial port seems to still be working.  I can open a terminal on it but get no response from grblHAL.

I don't think this is noise related.  I do have a single stepper driver/motor going on this but it seems too consistent to be noise. I will take the motor out, just use LEDs and rerun the test. There would be little EMI in that case.

I will also try using the Arduino USB serial wrapper rather than the PJRC one.

terjeio, 2020-11-21:
Strange.
Sleep mode is enabled?
Master or test branch?
Status report is still delivered?
ctrl-X (Reset) restores operation?

> |MPos:107.324,231.504,-13.500,0.000|Bf:0,1023|FS:1500,1000|Pn:XYZA>
100,ok

Could be due to corrupted data. Noise?

phil-barrett, 2020-11-21:
Master. Didn't try ctl-X or EStop.  Will next time.  Hooking up with terminal got nothing (cr, $$, ...).

Maybe another clue.  Power cycled the teensy. Reran ioSender but it couldn't connect (unable to open connection error message). 

Sleep Mode?

phil-barrett, 2020-11-21:
OK, something is very strange. I reran it without the stepper driver connected at all and it simply stopped executing commands at about 18 minutes. Run time was continuing to accumulate and the ioSender UI behaved like it was running a job however, stepper output LEDs showed no activity. Feed hold and cycle start worked but did not cause it to resume executing GCode.  Stop ended the job and I could restart the job.

I noticed that I didn't have aggressive buffering turned on in settings so I did that and restarted the job.  It ran for over an hour so I stopped the job, hooked the stepper driver back up and restarted the job.  It has run for 92 minutes now.  I will continue to test overnight.  If it doesn't fail, I will turn aggressive buffering off and try that. 

This could well be EMI - my test set up has all sorts of "bad practices". Though, the only likely source of EMI is a 12V stepper motor. The BOB is a 4 layer board with full ground plane. More testing needed.

phil-barrett, 2020-11-21:
Overnight run failed.  I think it has to be EMI so I will close this issue.  I have a lot of things to look at.  Always hate this kind of problem.

closed 2020-11-21 by phil-barrett

phil-barrett, 2020-11-21:
To close the loop, it was a bad USB hub causing the problem.  took the hub out of the set up and I have completed 4 hours of run time.