terjeio/grblHAL #123: PS2 keyboard plugin

Harvie, 2020-11-24:
Might be interresting to make clone of keypad plugin to use ps2 keyboard instead of i2c keypad

https://github.com/terjeio/grblHAL/tree/master/plugins/keypad

[image]

terjeio, 2020-11-24:
You want to give it a try?

Harvie, 2020-11-25:
it actualy seems quite easy to implement, so maybe later :-)
first i have to fix compatibility issues between bCNC and grblHAL. people reported some issues when running more than 3 axes, but that should be easy fix. But today i noticed the overrides are not working and that i cannot unpause paused job, i'll have to check what changed since original grbl protocol...

terjeio, 2021-04-30:
Create at PR over at the [new repo](https://github.com/grblHAL) if you ever get around to implement this. Or perhaps a new plugin submodule is needed?

closed 2021-04-30 by terjeio