terjeio/grblHAL #126: Compilation error Stm32F1xx 4th axis

S2000Stefan, 2020-11-28:
Unfortunately there are problems in the new test version when activating the 4th axis in config.h, three axes are o.k. STM32F1xx
Plugin: Odometers, Eeprom, USB
with UART Compilation o.k.
It is about time that my new controller comes with more flash memory (STM32 F303CCT6)  ;)
After activating the 4th axis Compilation error is displayed:

> c:\st\stm32cubeide_1.3.0\stm32cubeide\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.7-2018-q2-update.win32_1.0.0.201904181610\tools\arm-none-eabi\bin\ld.exe: GRBL Driver STM32F103C8.elf section `.data' will not fit in region `FLASH'
c:\st\stm32cubeide_1.3.0\stm32cubeide\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.7-2018-q2-update.win32_1.0.0.201904181610\tools\arm-none-eabi\bin\ld.exe: region `FLASH' overflowed by 388 bytes
collect2.exe: error: ld returned 1 exit status
make: *** [makefile:52: GRBL Driver STM32F103C8.elf] Error 1
"make -j12 all" terminated with exit code 2. Build might be incomplete.

09:45:35 Build Failed. 1 errors, 0 warnings. (took 3s.119ms)

terjeio, 2020-12-01:
> Unfortunately there are problems in the new test version

Initial test versions has settings information for `$help` etc. for _all_ settings in the core. I am moving plugin/driver specific information locally, freeing up enough to allow compilation. But only for a while? With limited flash/RAM this processor is not future proof when it comes to extensions. I'll try to keep it active as long as possible with the basic stuff.

I'll commit an update soon.

S2000Stefan, 2020-12-01:
> Initial test versions has settings information for $help etc. for all settings in the core. I am moving plugin/driver specific information locally, freeing up enough to allow compilation. But only [...]

Yes, I noticed that the memory of this controller is already at its limit.
The new STM32 F303CCT6 is finally on its way to me by post, now it can only take a few days until it will hopefully arrive at my place. Then my memory problem with the old chip should be solved for the time being. If I get this right, my english is not that good, there are still many innovations planned, which I am eagerly waiting for. 
In the long run, I will probably not be able to avoid betting on a really future-proof controller.
BTW are your address data on your homepage still up-to-date? So that I can send you a new controller (STM32 F303CCT6) when it finally arrives.

closed 2020-12-01 by S2000Stefan

terjeio, 2020-12-01:
Use this:

Terje Io
Ievegen 120
N-5917 Rossland
Norway

S2000Stefan, 2020-12-10:
O.K. the package with the controller STM32 F303CCT6 is on its way to you, I also put in a small christmas present for your tireless work for us and that I have annoyed you so often.

For your information, the Boot 0 jumper is the jumper on the side of the USB port.

I also tried to install the STMF4xx driver without success, which was to be expected given my capabilities,  
I'd better leave such things to you.  ;-)

terjeio, 2020-12-11:
> O.K. the package with the controller STM32 F303CCT6 is on its way to you

Thanks, perhaps we can retire the F103 driver at some point in the future when this is up and running.

>  I also put in a small christmas present for your tireless work for us and that I have annoyed you so often.

Thanks again - and thanks to you too, without annoying users nothing is going to move forward ;-)

terjeio, 2021-01-06:
Thank you for the christmas present, very beautiful! Yggdrasil?

I'll post information about the F303 driver when ready for test at issue #105.

S2000Stefan, 2021-01-07:
> Thank you for the christmas present, very beautiful! Yggdrasil?

I am very pleased that the package has finally arrived. 
The small gift is the least I can give you for your fantastic work.
Yes, it's called the tree of life in our house, yggdrasil, I didn't know the name, I learned something again, my niece drew it for me and then I made it into a relief.
 The wood is appropriately yew.

> I'll post information about the F303 driver when ready for test at issue #105.

O.k. no problem, there's no hurry, i have a very well functioning controller.   ;-)