terjeio/grblHAL #127: Two Inputs for Probing possible? Teensy Board

einencool, 2020-12-06:
Hi @terjeio and @phil-barrett ,

I‘ve got one question about the grblHAL, the Sender and the Teensy 4.1 Board.

Would it be possible to use two different Inputs for Probing, one for the Touch Probe and one for the 3D-Finder?
I think about changing my Touch probe to a Probe with magnetic connection.
The connection will only be used, when I put in the 3D-Finder. 
The Touch Probe for the Z Height should be always connected.

Both are Normally Closed switches, and that is why I would like to take a new input for the 3D-Finder, because then I see instantly that the 3D-Finder is not connected, because the Signal is Open and not closed.
Maybe it could be possible to use the Limit-Input from the B-Axis or something else...

Thank you in advance.
Greets Chris

terjeio, 2020-12-06:
Do you want the switchover to be automatic or manual? If manual then M64/M65 can be used from a macro to switch between them using a small relay, but requires that @phil-barrett (or I?) publish a piece of code that is currently unpublished. A manual switch can also be used.
Automatic will be harder(?) as triggering the probe should not "disconnect" it, but perhaps a custom plugin could be made to handle the second input?

einencool, 2020-12-06:
Must the inputs be separated?
It's just a thought, if  the following things were possible...

- both inputs will be shown in the "status LED bar" like probe 1 and probe 2
- both inputs react as one. It doesn't matter, which one is triggered, the desired probing will stop, if one of them is triggered 

Then in my thoughts there doesn't have to be a switch or a macro or anything like this.

What is your thought about it?

phil-barrett, 2020-12-06:
What you are asking for is 2 separate inputs.  Two NC inputs could be connected in series for a single input to the controller but there could not be independent status reported.   As Terje suggested, a relay could be used to switch them based on an M64/65.  Though that would require issuing an M64 to grblHAL via MDI. A simple switch could also work and, I think, be easier to use.

I have played with a "probe mixer" that allows input of either NC or NO probes and outputs NC or NO to the Motion Controller. It currently works on 5V only.  I haven't decided if I want to take it to the level of working with 12 and 24V I/O. Easy enough to do, just not sure there is enough interest.

einencool, 2020-12-06:
Hi @phil-barrett 
Yes you are right, I would like to configure 2 separate inputs.

Now I have the two probes connected in a row on the probing input.
But when I have the new 3D-Finder with the magnetic Connection, it will be stored in the case. So the connection will be open.
With a switch it would be possible, but there is the risk of using the wrong position of the switch.
It was just a thought, because the big machines have also 2 separate inputs for probing.

But then I have to think about the way I'll use it in future.

terjeio, 2020-12-06:
> both inputs will be shown in the "status LED bar" like probe 1 and probe 2

Ouch - this would require changes to the driver(s), core, protocol and the sender(s). That is not something done easily/quickly. The "probe mixer" could be a good enough solution?

einencool, 2020-12-06:
Hmm,
Ok, I’ve hoped that there could be a simple solution where both probes can be used at the same time or only the touch Probe.
But then I’ll do it like before and make an adapter for the magnetic connector that the circuit will be closed without the probe attached to the machine.

Thank you, I’ll close this issue so that you can focus on more important things ;-)
Greets Chris

closed 2020-12-06 by einencool

terjeio, 2020-12-06:
> Ok, I’ve hoped that there could be a simple solution where both probes can be used at the same time or only the touch Probe.

Not everyting is simple, I have a longer term plan to look into probe handling as I want proper probe protection. Could add this to the todo list when this is done.

> Thank you, I’ll close this issue so that you can focus on more important things ;-)

There are IMO some more important things going on just now. I am pretty close to release a new test version that will get rid of the problem of keeping the sender and controller in sync when new settings or setting flags are added. Agreeing on a common format for post legacy grbl is also important - hopefully this will lead to more senders beeing able to use the full potential of grblHAL and other ports as well. And I really want to get ioSender out of beta, I'll try to add other stuff that you are waiting for the first production release. Getting back to my workshop is important too, I have spent an insanely large amount of hours on grblHAL and ioSender the past few months so my machines has been mostly idle...

One good ting is that the user base is growing, IMO it is important to keep that momentum going too.