terjeio/grblHAL #128: corexy.c:77:29: error: Uninitialized variable: a_steps [uninitvar] (CPPCheck)

Harvie, 2020-12-08:
I've been messing with static analyzers. Unfortunately i was not able to run scan-build (clang analyzer) on the codebase yet, however even the cppcheck found some issues. You might want to check these:

```
$ cppcheck . ../../plugins/ > /dev/null
../../plugins/networking/WsStream.c:866:63: warning: The address of local variable 'mask' might be accessed at non-zero index. [objectIndex]
                        if(!WsStreamRxInsert(*payload++ ^ mask[i % 4]))
                                                              ^
../../plugins/networking/WsStream.c:846:48: note: Address of variable taken here.
                    uint8_t *mask = (uint8_t *)&session->header.mask;
                                               ^
../../plugins/networking/WsStream.c:866:63: note: The address of local variable 'mask' might be accessed at non-zero index.
                        if(!WsStreamRxInsert(*payload++ ^ mask[i % 4]))
                                                              ^
grbl/corexy.c:77:29: error: Uninitialized variable: a_steps [uninitvar]
    target_steps[A_MOTOR] = a_steps + b_steps;
                            ^
grbl/corexy.c:77:39: error: Uninitialized variable: b_steps [uninitvar]
    target_steps[A_MOTOR] = a_steps + b_steps;
                                      ^
grbl/corexy.c:78:29: error: Uninitialized variable: a_steps [uninitvar]
    target_steps[B_MOTOR] = a_steps - b_steps;
                            ^
grbl/corexy.c:78:39: error: Uninitialized variable: b_steps [uninitvar]
    target_steps[B_MOTOR] = a_steps - b_steps;
                                      ^
web/backend.c:103:54: error: Subtracting pointers that point to different objects [comparePointers]
    const size_t favicon_ico_size = (favicon_ico_end - favicon_ico_start);
                                                     ^
web/backend.c:102:32: note: Variable declared here.
    extern const unsigned char favicon_ico_end[]   asm("_binary_favicon_ico_end");
                               ^
web/backend.c:103:38: note: Array decayed to pointer here.
    const size_t favicon_ico_size = (favicon_ico_end - favicon_ico_start);
                                     ^
web/backend.c:101:32: note: Variable declared here.
    extern const unsigned char favicon_ico_start[] asm("_binary_favicon_ico_start");
                               ^
web/backend.c:103:56: note: Array decayed to pointer here.
    const size_t favicon_ico_size = (favicon_ico_end - favicon_ico_start);
                                                       ^
web/backend.c:103:54: note: Subtracting pointers that point to different objects
    const size_t favicon_ico_size = (favicon_ico_end - favicon_ico_start);
                                                     ^
web/backend.c:124:86: error: Subtracting pointers that point to different objects [comparePointers]
    return httpd_resp_send(req, (const char *)ap_login_html_start, ap_login_html_end - ap_login_html_start);
                                                                                     ^
web/backend.c:123:32: note: Variable declared here.
    extern const unsigned char ap_login_html_end[]   asm("_binary_ap_login_html_end");
                               ^
web/backend.c:124:68: note: Array decayed to pointer here.
    return httpd_resp_send(req, (const char *)ap_login_html_start, ap_login_html_end - ap_login_html_start);
                                                                   ^
web/backend.c:122:32: note: Variable declared here.
    extern const unsigned char ap_login_html_start[] asm("_binary_ap_login_html_start");
                               ^
web/backend.c:124:47: note: Array decayed to pointer here.
    return httpd_resp_send(req, (const char *)ap_login_html_start, ap_login_html_end - ap_login_html_start);
                                              ^
web/backend.c:124:86: note: Subtracting pointers that point to different objects
    return httpd_resp_send(req, (const char *)ap_login_html_start, ap_login_html_end - ap_login_html_start);
                                                                                     ^
webui/server.c:931:44: error: Buffer is accessed out of bounds: msg [bufferAccessOutOfBounds]
                                    strcpy(msg, "Error: Cannot apply changes");
                                           ^
webui/server.c:941:44: error: Buffer is accessed out of bounds: msg [bufferAccessOutOfBounds]
                                    strcpy(msg, "Error: Cannot apply changes");
                                           ^
webui/server.c:947:40: error: Buffer is accessed out of bounds: msg [bufferAccessOutOfBounds]
                                strcpy(msg, "Wrong authentication!");
                                       ^
webui/server.c:952:32: error: Buffer is accessed out of bounds: msg [bufferAccessOutOfBounds]
                        strcpy(msg, "Error: Incorrect password");
                               ^
webui/server.c:969:44: error: Buffer is accessed out of bounds: msg [bufferAccessOutOfBounds]
                                    strcpy(msg, "Error: Incorrect password");
                                           ^
webui/server.c:977:44: error: Buffer is accessed out of bounds: msg [bufferAccessOutOfBounds]
                                    strcpy(msg, "Error: Incorrect password");
                                           ^
webui/server.c:1069:86: error: Subtracting pointers that point to different objects [comparePointers]
    return httpd_resp_send(req, (const char *)index_html_gz_start, index_html_gz_end - index_html_gz_start);
                                                                                     ^
webui/server.c:1066:32: note: Variable declared here.
    extern const unsigned char index_html_gz_end[]   asm("_binary_index_html_gz_end");
                               ^
webui/server.c:1069:68: note: Array decayed to pointer here.
    return httpd_resp_send(req, (const char *)index_html_gz_start, index_html_gz_end - index_html_gz_start);
                                                                   ^
webui/server.c:1065:32: note: Variable declared here.
    extern const unsigned char index_html_gz_start[] asm("_binary_index_html_gz_start");
                               ^
webui/server.c:1069:47: note: Array decayed to pointer here.
    return httpd_resp_send(req, (const char *)index_html_gz_start, index_html_gz_end - index_html_gz_start);
                                              ^
webui/server.c:1069:86: note: Subtracting pointers that point to different objects
    return httpd_resp_send(req, (const char *)index_html_gz_start, index_html_gz_end - index_html_gz_start);
                                                                                     ^
webui/server.c:1022:42: error: Uninitialized variable: user [uninitvar]
                strcpy(session->user_id, user);
                                         ^
webui/server.c:387:9: error: Uninitialized variable: path [uninitvar]
    if(*path == '\0')
        ^
webui/server.c:440:9: error: Uninitialized variable: path [uninitvar]
    if(*path == '\0') // in case something failed...
        ^
webui/server.c:609:10: error: Uninitialized variable: path [uninitvar]
        *path = '\0';
         ^
webui/server.c:720:6: error: Uninitialized variable: path [uninitvar]
    *path = '\0';
     ^
nofile:0:0: information: Too many #ifdef configurations - cppcheck only checks 12 configurations. Use --force to check all configurations. For more details, use --enable=information. [toomanyconfigs]

```

terjeio, 2020-12-08:
you found issues in cppcheck too - I cant't see anything wrong with this code:

```
static void corexy_target_to_steps (int32_t *target_steps, float *target)
{
    uint_fast8_t idx = N_AXIS;
    int32_t a_steps, b_steps;

    do {
        switch(--idx) {
            case X_AXIS:
                a_steps = lroundf(target[idx] * settings.axis[idx].steps_per_mm);
                break;

            case Y_AXIS:
                b_steps = lroundf(target[idx] * settings.axis[idx].steps_per_mm);
                break;

            default:
                target_steps[idx] = lroundf(target[idx] * settings.axis[idx].steps_per_mm);
                break;
        }
    } while(idx);

    target_steps[A_MOTOR] = a_steps + b_steps;
    target_steps[B_MOTOR] = a_steps - b_steps;
}
```
Too complicated for cppheck to figure out (it does not know/understand that minimum number of axes is 3)? I'll take a look at the rest later.

Harvie, 2020-12-08:
Some of the reported errors might be false positives. But it's always worth checking, because sometimes it can help to find actual bugs...

Harvie, 2020-12-08:
Perhaps i can try to build drivers/Simulator in scan-build. That way there would be no conflict with esp-idf. But i guess the actual motion control code doest not compile in simulator build.

terjeio, 2020-12-08:
> Some of the reported errors might be false positives, that's kinda spirit of static analysis. But it's always worth checking, because sometimes it can help to find actual bugs...

True that. Enabling MISRA rules would help too? Could be a scary thing to do though.

> Perhaps i can try to build drivers/Simulator in scan-build. That way there would be no conflict with esp-idf. But i guess the actual motion control code doest not compile in simulator build.

Not sure what you mean by "the actual motion control code" - if you mean code provided in driver.c then there are two variants in the simulator, one that actually has (simulated) motion control code and one (for gcode check) that does not. Then there are the other driver.c implementations and configurations - you fast end up with a large amount of different sources if all variants are to be checked. At least I have gotten rid of many of the #define options that was in the original by moving them to settings instead, so checking this part has been made simpler...

Harvie, 2020-12-08:
Ok, so i've managed to run the simulator in clang analyzer scan-build. Just the grbl core, without any plugins. Just single warning was issued:

```
$ scan-build -v  make >/dev/null
main.c:93:9: warning: Although the value stored to 'retval' is used in the enclosing expression, the value is never actually read from 'retval' [deadcode.DeadStores]
    if((retval = select((socket_fd > sim.socket_fd ? socket_fd : sim.socket_fd) + 1, &cfds, NULL, NULL, &tv)) > 0) {
        ^        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 warning generated.
```

There were also few warnings generated directly by GCC, but i guess you've already seen these:

```
grbl/report.c: In function ‘report_grbl_settings’:
grbl/report.c:572:23: warning: unsigned conversion from ‘int’ to ‘uint_fast8_t’ {aka ‘unsigned char’} changes value from ‘256’ to ‘0’ [-Woverflow]
  572 |     if(all) for(idx = Setting_AxisSettingsMax + 1; idx <= Setting_SettingsMax; idx++) {
      |                       ^~~~~~~~~~~~~~~~~~~~~~~
grbl/nvs_buffer.c: In function ‘nvs_alloc’:
grbl/nvs_buffer.c:262:35: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
  262 |         mem_address = (uint8_t *)((uint32_t)(mem_address - 1) | 0x03) + 1; // Align to word boundary
      |                                   ^
grbl/nvs_buffer.c:262:23: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
  262 |         mem_address = (uint8_t *)((uint32_t)(mem_address - 1) | 0x03) + 1; // Align to word boundary
      |                       ^
simulator.c: In function ‘sim_socket_out’:
simulator.c:148:17: warning: variable ‘continuation’ set but not used [-Wunused-but-set-variable]
  148 |     static bool continuation = 0;
      |                 ^~~~~~~~~~~~
driver.c:264:13: warning: ‘spindleSetStateVariable’ defined but not used [-Wunused-function]
  264 | static void spindleSetStateVariable (spindle_state_t state, float rpm)
      |             ^~~~~~~~~~~~~~~~~~~~~~~
driver.c:127:13: warning: ‘stepperPulseStartDelayed’ defined but not used [-Wunused-function]
  127 | static void stepperPulseStartDelayed (stepper_t *stepper)
      |             ^~~~~~~~~~~~~~~~~~~~~~~~
serial.c:63:48: warning: ‘rxbuffer’ is static but used in inline function ‘serialRxCount’ which is not static
   63 |     uint_fast16_t head = rxbuffer.head, tail = rxbuffer.tail;
      |                                                ^~~~~~~~
serial.c:63:26: warning: ‘rxbuffer’ is static but used in inline function ‘serialRxCount’ which is not static
   63 |     uint_fast16_t head = rxbuffer.head, tail = rxbuffer.tail;
      |                          ^~~~~~~~
```

terjeio, 2020-12-08:
> There were also few warnings generated directly by GCC, but i guess you've already seen these

I've seen some compilers issue more warnings than other, the one used by ESP-IDF is the one that currently complains where alle the others remains silent. I have never bothered to dig into what the default compiler settings are for different setups I have, maybe I should...

I do not understand the warning in grbl/report.c - perhaps it is related to me using uint_fast8_t instead of uint_fast16_t as I should. I'll change that as it is a logical error - soon as I am about to commit an update to the test branch.

The warning grbl/nvs_buffer.c is due to my attempt to get the pointer arithmetic correct? I have not seen this from any of the compilers I use, and the code works. Perhaps a c expert suggest a better approach?

Warnings in the simulator code can easily be fixed - for later...

closed 2021-08-22 by terjeio