terjeio/grblHAL #136: Unable to build for LPC1768

Creepop, 2020-12-11:
Hello everyone.

The original board that came with my CNC uses the original grbl and as it turns out, it is a pretty terrible piece of hardware with major issues relating to EMC, causing both USB communication aswell as the MCU itself to reset randomly.
I have an old MKS SBASE that uses the LPC1768 microcontroller laying around that I wanted to use to replace the original controller board and I'm very much struggling to get it working. At first I tried smoothieware but since they have no implementation of pause whatsoever, that route quickly became a dead end.
So getting to the grblHAL, I've been trying to the better part of the day to figure out how to get it built and installed and I have made some progress, but sadly that has come to an end. After editing the diskio.h header file a bit (downloaded the fatfs version stated in the readme) because the disk_write and disk_read definitions didn't match the definitions in the diskio.c file in the firmware itself, I got it to compile properly. After that I spent another few hours until I realised that I can't use the built-in bootloader to flash the firmware from sd card, I need to flash it through UART (this should probably be stated somewhere).
Anyway, it still didn't work, it seemed like the code just crashed somewhere. Set up some LED's at different points in the code and narrowed down the point at which the controller crashes to the function read_global_settings() in the settings.c file in settings_init() function. The code is well behaved up to that point, but no more instructions after this point are executed.
Now I can hardly read C++ code (having never used C++ myself), but searching the code, it seems that hal.nvs.memcpy_from_nvs is only defined anywhere if EEPROM is used (which is not the case), in case of using flash memory for permanent storage, hal.nvs.memcpy_from_flash is used. Ofcourse, just changing the hal.nvs.memcpy_from_nvs line to hal.nvs.memcpy_from_flash didn't work, so here I am, waiting for some input on how to move forward.
Am I doing something terribly wrong seeing as the developer of the grblHAL had his LPC1768/9 board running, presumably with the same software that is published, or has some stuff in the drivers section of the firmware gone wrong at some point during development?

terjeio, 2020-12-11:
Can you give the [test branch](https://github.com/terjeio/grblHAL/tree/test) a try if you not already did so? This has extenal dependencies removed and I just added support for Bigtreetech SKR v1.3 earlier today. I tested this with uploading via a J-Link probe and a SD-card (via bootloader) and both worked for me. 

You can also try to disable loading settings from flash here:

https://github.com/terjeio/grblHAL/blob/498320fd5dc1fe318941bf3458a1ebfde6a99682/drivers/LPC1769/src/driver.h#L88-L92

Set  FLASH_ENABLE to 0 in line 89 for that.

I have added a build option for bootloader compatible firmware in the test branch, some info [here](https://github.com/terjeio/grblHAL/tree/test/drivers/LPC1769).

dresco, 2020-12-11:
> Anyway, it still didn't work, it seemed like the code just crashed somewhere. Set up some LED's at different points in the code and narrowed down the point at which the controller crashes to the [...]

May be a red herring, but I had the same crash when I first tried to compile the STM32 build (was not using the documented build process). My issue was that I hadn't used the expected linker script to set up the flash EEPROM emulation..

Creepop, 2020-12-12:
Alright, I have successfully flashed the board, everything went pretty well. I used the test branch and flashed through the UART port (for some reason I haven't been able to get the bootloader working again even after flashing it, but that is a minor concern right now).
Seems to be working as intended (the compiler also wasn't complaining about the disk_write and disk_read functions being incompatible with their prototypes). I did use the Debug build configuration yesterday (didn't notice that there are two build configurations available), any chance that this might have caused the described behavior?

terjeio, 2020-12-12:
> I did use the Debug build configuration yesterday (didn't notice that there are two build configurations available), any chance that this might have caused the described behavior?

Possibly - hard to tell. ARM code IIRC is only partially relocatable - could be a reason. I last programmed in ARM assembly nearly 20 years ago so this may no longer be true. Building for a bootloader is certainly different as the linker has to be told about the base address to use as this is not the same as the default address. The Debug build is not suitable for bootloader flashing due to this.

Have you made a pin map file for your board or could you use an existing one? If you made a new is sharing it possible?

Creepop, 2020-12-14:
MKS SBASE is a clone of smoothieboard, so the pin mapping is the same. Some of the special pins might have to be changed as they are quite nonsensical for the board at hand but that is a minor concern.
Anyway, I am back at where the actual CNC is (had just taken the board with me for the holiday), and upon trying it out, I seem to be running into some more issues.
The inputs appear to be working fine, endstops, probe etc. work. The outputs are a bigger issue though. I'm not very familliar with the NXP LPC platform (or ARM chips in general for that matter) so you can imagine I have quite a bit of trouble wraping my head around setting gpio pins by writing directly to memory banks linked to gpio registers, so I haven't really been able to diagnose the issue yet, but the step pin works (which I would assume is set automatically by timers), the direction pin is always stuck in the same state as well as enable pin being always on. So my motors are always on and always moving in the same direction, regardless of what they are being told. 
Another problem I encountered wat that inverting for the PROBE pin also doesn't appear to be working, probe_invert_mask always appears to be 0 regardless of any settings in memory. Didn't debug further, just hard coded the inversion for now. 
I don't mind sharing any code I make if it comes to it, but for the time being, getting everything working is what I will be working towards. If any useful and publishable code comes from it, I will share it.

terjeio, 2020-12-14:
I have just added a [MKS SBASE V1.3 map file](https://github.com/terjeio/grblHAL/blob/test/drivers/LPC1769/src/mks_sbase_map.h) to the test branch map file you may use as a starting point for changes if this is the board you have. There is tentative code for setting motor current too via the on-board I2C pots.

Which pin are you using for the probe input?

zmrdko, 2020-12-14:
I have come across strange behavior with probe. There is no way to revert pin behavior. $6=1 and $6=0 makes no difference.

terjeio, 2020-12-14:
Fix for probe invert bug now in [test branch](https://github.com/terjeio/grblHAL/tree/test).

@Creepop Motor enable and direction pins are working for me - just tested with the new MKS SBASE map.

Creepop, 2020-12-14:
zmrdko: yes, that is the same behavior that I observed. But that is easy enough to fix (atleast with hardcoding). For the time being, I'm using P1.22 for the probe pin, but if anyone has any reason for choosing a different pin, I can sway, not really bothered with what goes where since a lot of pins are free.
As for the motion control itself, there is definitely something strange going on. It is somewhat difficult to get a grip on what exactly is happening without a debugger (probably won't be buying one just for one board), but gpio control seems to be pretty broken. Changing the GPIO mode changes the behaviour dramatically and, for example, step pins only work in GPIO_SHIFT0 mode, as for direction pins, in BITBAND mode only the Z axis worked, in GPIO_MAP mode, the X and Z axis worked, can't use bitshifting since the layout doesn't allow for that (the pins, while on the same port, aren't subsequent). Also have noticed some weird behavior on the enable lines aswell, but haven't payed too much attention to that since I could just hardcode or even hardwire that for the time being if necessary.
I even flashed the smoothieware firmware again just to make sure that the hardware is still fine, and, sure enough, everything worked flawlessly.
Assuming I can work out what is happening with the motion control, I will test the I2C POT aswell...
Just to confirm that I'm not screwing something up, could you compile a binary file for the SBASE with USB Serial disabled (preferably for bootloader flashing, but a standard hex would be fine aswell)? I can work around any pin layout for the reset/cycle start/feed hold pins, don't really care as long as they are actually accessible on the board. That way atleast I can confirm whether there is an issue with the code itself or the problem is somewhere between the chair and the screen.

terjeio, 2020-12-14:
>  Changing the GPIO mode changes the behaviour dramatically

Oops, the smoothieboard map is broken. Can you try with the new MKS SBASE map file instead? I'll take a closer look at the smoothieboard map later. Note that I do not own any of these boards so I can only do simple testing with a Re-ARM board instead...

BTW is there some agreement on which pins should be used for what when a 3D printer board is configured for grbl use?

terjeio, 2020-12-15:
More fixes commited to test branch, hopefully ok now.
Note that control signals for the MKS SBASE board has been moved to thermistor inputs as they need interrupt capable pins.

Creepop, 2020-12-15:
Ok, so just did a clean build of the latest test branch, changed absolutely nothing aside from disabling usb serial and sd support.
Enable pins require inversion (which is fine), after that, the Z axis works as intended, the step pins don't seem to be doing anything for the x and y axis. Since I can't move x and y axis, haven't checked what the direction pins do for these axis.
Basically the same or very simillar behaviour as I was observing yesterday. Didn't fiddle around with any GPIO modes yet, but something is definately broken. Can you send the HEX file you were using to test on your end? Since I have confirmed the hardware to be working fine, doesn't seem to make a lot of sense for it to be misbehaving on my end and working beautifully on the Re-Arm board. 
And yes, I was using the MKS Sbase pin map yesterday.
On a more positive note, yes, the probe inversion now works correctly.

terjeio, 2020-12-15:
> Can you send the HEX file you were using to test on your end?

With USB enabled:

[GRBL Driver LPC176x.zip](https://github.com/terjeio/grblHAL/files/5694850/GRBL.Driver.LPC176x.zip)

With UART enabled (not tested):

[GRBL Driver LPC176x UART.zip](https://github.com/terjeio/grblHAL/files/5694890/GRBL.Driver.LPC176x.UART.zip)

Output from $I:

[VER:1.1f.20201212:]
[OPT:VNMSL,35,1024,3,0]
[NEWOPT:ENUMS,*FLASH,TC]
[FIRMWARE:grblHAL]
[DRIVER:LCP1769]
[DRIVER VERSION:201214]
[BOARD:MKS SBASE V1.3]
ok

Note that I test with LEDs only and a Re-ARM board - I do not have a machine or motors connected.

Creepop, 2020-12-15:
Good news: all axis work fine with the HEX file you provided.
Question is, what is so different about me compliling the source code.
At this point I'm just grasping straws, but can you upload the MCUXpresso workspace project directly? Then I can see if me compiling the same project brings the same results, and if that works aswell, try to look for differences.

terjeio, 2020-12-15:
> At this point I'm just grasping straws, but can you upload the MCUXpresso workspace project directly?

Here you go:

[GRBL Driver LPC176x.zip](https://github.com/terjeio/grblHAL/files/5695165/GRBL.Driver.LPC176x.zip)

> try to look for differences.

Have you checked the $I output to see if the correct board is in fact compiled in?

Creepop, 2020-12-15:
Alright, in case anyone else runs into this issue: use Debug build configuration instead of Release. That appears to be the main culprit of motion control issues.
Can you explain the differences between the build configurations? Or is the Release build config just an artifact of creating the MCUXpresso project and simply hasn't been configured at all?
And yes, the $I output states the correct board name in both scenarios.

Creepop, 2020-12-15:
Two more things I would like to note:
1) Would probably be a good idea to add another #define and #ifdef statement for the MAX limit pins. I have my limit switches wired in series (the original GRBL uses the same pin for both min and max) in a normally closed configuration (to detect brakes in cable), therefore I need to invert the limit pins. Since nothing is plugged into the MAX limit pins, there is nothing pulling them down and they act as always triggered, so the limit switches simply don't work at all in this case. Commenting out the pin definitions for the MAX limit pins fixes the issue.
2) In the MKS SBASE map PROBE and RESET share the same pin.
Anyway, so far so good, hopefully will have a working CNC this evening. If I find any more bugs or issues, I will post them here.

terjeio, 2020-12-15:
> Would probably be a good idea to add another #define and #ifdef statement for the MAX limit pins.

Added to my_machine.h, LIMIT_MAX_ENABLE - default off.

> In the MKS SBASE map PROBE and RESET share the same pin.

I need input on pin mapping - which one to use for probing? Or for other inputs? Probe input does not need to be an interrupt capable pin so one from port 1 will be ok.

> Can you explain the differences between the build configurations?

It seems to me that enabling optimization optimizes away code. When changed to _None (-O0)_ it works. Compiler or linker "bug"? I have not seen this with other compilers I use...

Creepop, 2020-12-16:
I don't think there is any consensus regarding pin mapping. In case this project ever leads to the addition of an LCD (for example the 3d printer standard "RepRapDiscount Full Graphic Smart Controller") or maybe even a custom designed controller board, I would probably leave EXP1 and EXP2 connectors untouched unless necessary. I personally find using the pins in the top right corner (P1.22, P1.23, P2.12, P2.11 and P4.28) the most reasonable way to connect extra accessories. When using anything that doesn't have a specialized connector for it, one will always struggle finding grounding points anyway, no matter which pins you use. 
Regarding the build configurations: thanks for clearing this up and thanks for the help in general. Has been quite the struggle getting the CNC working properly, but seems to be more or less well behaved right now. Haven't milled anything yet, but during testing everything seems fine. The motor current control using the i2c digital pots also works (haven't measured the exact current to make sure that the scaling is correct, but it most certainly does change the motor current).

terjeio, 2020-12-18:
P1 is not very usable for inputs since its pins are not interrupt capable. That all LPC176x based 3D printer boards I've seen has limit pins assigned to P1 is an engineering mistake? I had to add polling for the limit pins so that hard limits can be enabled...

I am moving the probe input to P1.23 for the MKS SBASE as this is polled by the core. Should I enable the I2C pots as default as well?

Creepop, 2020-12-18:
Since the I2C pots do work, I think enabling them by default is a good idea, since there is no other way to change the motor current on those boards. Very small steppers overheat with the default current setting and large steppers have nowhere near enough current, meaning that most people wouldn't be able to use the board without tuning the current anyway. Plus, if any bugs do arise, catching them will take less time if more people are using the code.
But yes, 3D printer hardware is rarely optimized for speedy pin switching or reading, from what I've seen, it usually just boils down to: whatever is easier to route.

infamous-panda, 2021-02-09:
I have a couple spare Sbases I'd like to give this a go with a laser cutter setup. Couple things to consider for my build

I will have 2 independent motors for Y axis. Do you know if gantry squaring is supported?

Is the pwm pin for firing the laser still the the heatbed connector?

terjeio, 2021-02-09:
> I will have 2 independent motors for Y axis. Do you know if gantry squaring is supported?

Currently not, some code has to be added to the driver for that.

> Is the pwm pin for firing the laser still the the heatbed connector?

It seems so by default. From the map file:

```
#ifdef SPINDLE_PWM_PIN_2_4
#define SPINDLE_PWM_CHANNEL         PWM1_CH5    // MOSFET3 (P2.4)
#else
#define SPINDLE_PWM_CHANNEL         PWM1_CH6    // BED MOSFET (P2.5)
#endif
```

`SPINDLE_PWM_PIN_2_4` is not defined anywhere (it has to be defined in the project proerties to switch PWM to P2.4).

terjeio, 2021-08-21:
> I will have 2 independent motors for Y axis. Do you know if gantry squaring is supported?

[Squaring support](https://github.com/grblHAL/LPC176x/blob/a15a67d7f2d09cb25f2e89effd269666ec5b6f23/src/my_machine.h#L42-L54) has recently been added (in the new repo).

closed 2021-08-21 by terjeio