terjeio/grblHAL #142: Laser raster speed

theomajigga, 2020-12-16:
(Xpost from Phil's teensy project)

Great work on this. I know very little when it comes to this type of ports, but my realization of how cheap M4 boards have become, are leading me down the path to upgrade me co2 laser to a proper (freshly written) 32-bit port of GRBL.

First: Sorry for an issue report, I know this isn't the proper route.

Second: I am currently running GRBL-LPC for my laser, I get stuttering on high dpi, fast movements (254dpi/300mm/s). I am hoping to get some increased speeds out of my machine, as it is capable of 500mm/s (with out imploding :) )

Per Oz of the project Lightburn:

An image at 254 dpi means 0.1mm dots, or 10 dots per mm. If you run 100 mm/sec, that’s 1000 dots per second, and each dot takes a gcode instruction to produce.

Stock Smoothieware hits 800 instructions/second, or 80mm/sec with 254dpi.
8-bit GRBL is about 400, or 40mm/sec with 254 dpi

If you’re running GRBL-LPC or the new cluster Smoothieware (available on C3D’s forum) you can hit about 250mm/sec at 254 dpi.

After doing extensive research, it led me to your project, and some of the work you did with the gnea fork.

I guess, my question is, would upgrading to the teensy 4.x and the board you have produced on tindie, get me the speeds I am looking for?

Thank you!

terjeio, 2020-12-16:
> I guess, my question is, would upgrading to the teensy 4.x and the board you have produced on tindie, get me the speeds I am looking for?

Possibly, but not without tuning. What is needed is at least a substantially larger planner buffer as grbl always plan to a full stop. Sender/controller interaction may also play into this - as senders waiting for a ok before sending the next line is not likely to keep the input buffer from emptying and thus the controller waiting.

theomajigga, 2020-12-16:
I believe that lightburn does buffering, and not sync, so that shouldn't be an issue?

terjeio, 2020-12-16:
Check this [blog post](https://embeddedtronicsblog.wordpress.com/2020/07/20/grblhal/) - and others by the same author.

Tuning is trivial, no coding required - only one or a few #defines to change.

Do you have a test file that is representative of what you want to achieve? If so can you post it here?

infamous-panda, 2020-12-31:
Same boat as the OP. I have been able to achieve 3-5x the speed of lpc with grbl-HAL on teensy 4.1 by increasing the buffer to 1024. There is some other stuff going on that phil and I are still trying to sort out but I can confirm it works with a physical test

theomajigga, 2021-01-04:
> There is some other stuff going on

What is this other stuff, just out of curiosity.

infamous-panda, 2021-01-04:
> > There is some other stuff going on
> 
> What is this other stuff, just out of curiosity.

Phil build with the increased buffer does not seem to run any faster than LPC. I built grbl-HAL from the files in the release not master and it runs much faster. I do not know what the difference that matter are.

terjeio, 2021-01-04:
@infamous-panda Can you give the test branch a try? And if possible upload your test g-code for me to try with?

infamous-panda, 2021-01-04:
> @infamous-panda Can you give the test branch a try? And if possible upload your test g-code for me to try with?

Can you point me specifically to which files are the test branch? I am not very familiar with compiling firmware so it's a bit of struggle to get it all prepared just right. I compiled my version using the files in Build 20200503 release with Arduino IDE, and when trying to compile the more current "master" files so that I could enable ethernet support I got stuck and asked Phil to do it for me. The firmware he sent ran quite a bit slower as noted even though he changed the planner to 1024. There must be other differences with some other settings

Test file below. 

https://drive.google.com/file/d/1D5yneFmeh0anS1S_EWtQJQ13Kyi44FTT/view?usp=sharing

terjeio, 2021-01-05:
@infamous-panda What max feed rates did you achieve with the master and Phil's version?

When I am testing with the test branch feed rate tops out around 30000mm/min (500mm/sec) with S commands removed, around 25000mm/min with S commands present.

S commands are not needed in laser mode (unless gray scale engraving) as G1/G0 switches the laser on/off automatically when M4 is used.

infamous-panda, 2021-01-05:
I get around 50,000 mm/m. Phil's around 17,000 mm/m

Sample code is worse case. We do sometimes vary the laser power between
dots.

Are you using a teensy 4.1?

terjeio, 2021-01-05:
> Are you using a teensy 4.1?

Yes.

>We do sometimes vary the laser power between dots.

Is it possible to only add S commands when power is changing?

Have you seen the edit I made to the previous comment?

terjeio, 2021-01-05:
A bit more testing. Other senders I have tested performs badly. The best I have seen so far achieves just over 17000mm/min (LaserGrbl). So feeding the controller with data rapidly enough is part of the problem.

I increased the planner buffer to 1400 and tried with ioSender again (with the _Console_ tab open) and got this:

[image]

Time to completion was 0:58.

If I run the program with the _Program_ or _3D View tab_ open the sender goes crazy as it cannot keep the UI up to date. It eventually hangs.

So at these insane feedrates the sender is the main limiting factor...

infamous-panda, 2021-01-05:
terjeio,

Would you be able to send me a copy of your firmware files with Ethernet enabled? I would love to get some testing done on it since the one Phil provided capped out at about 17000. 

I have two PCs I am testing with. The one I would like to attach to the laser is an older surface with an Atom processor. At 50,000 mm/m it is at 100% cpu usage per the task manager and it seems to stutter a couple times through this code. My workstation handles it easily. 

A stable 50,0000 using ethernet would be ideal for me since this is about the limit of my laser being able to turn off and on with usable power. However I am sure there are other laser's out there which can take advantage of this. It would be nice to have a gcode compatible system that can provide as much speed as DSP machines.

terjeio, 2021-01-05:
@infamous-panda I have uploaded a hex file and the edge version of ioSender [here](http://io-engineering.com/downloads).

The sender has new settings that may be useful. If you enable _Filter "ok" response_ then you can enable _Verbose_ mode in the console without the sender kneeling over. I do this to check free space in the planner buffer - the first value in the `Bf:` element in the real-time report.

[image]

infamous-panda, 2021-01-05:
> @infamous-panda I have uploaded a hex file and the edge version of ioSender [here](http://io-engineering.com/downloads).

Thank you! I will give it a try this morning. 

Quick question. For the Settings:GRBL, I am unsure how the Save Reload and Backup buttons work. I assume this is to save parameters likes steps/deg on the controller which would be great, however there is no prompt or anything when those buttons are pressed.

terjeio, 2021-01-05:
> I am unsure how the Save Reload and Backup buttons work

* Save - for writing settings to the controller.
* Reload - for reloading settings from the controller.
*  Backup - saves settings to the file `settings.txt` in the sender folder. This can be used for a restore later. I should add a Restore button for that...

infamous-panda, 2021-01-05:
> * I should add a Restore button for that...

Ahh yes that would be great. I understood "Reload" was for that function but it never work as expected. Thanks

infamous-panda, 2021-01-05:
> @infamous-panda I have uploaded a hex file and the edge version of ioSender [here](http://io-engineering.com/downloads).

I am unable to connect Gcode Sender to the board using this firmware through USB. Is it ethernet only? If so what IP settings would I use?

terjeio, 2021-01-05:
> I am unable to connect Gcode Sender to the board using this firmware through USB.

I see I have compiled with FRAM enabled. If not soldered on your board then this could cause it to hang. Too many details to keep track of sometimes...

A new hex file without FRAM/EEPROM support is just uploaded.

BTW I have changed the test branch so there is no longer a need to copy files around before building. Just import the project from the driver/IMXRT1062 folder.

>  Is it ethernet only? If so what IP settings would I use?

No. If the board is up and running it is likely it gets it IP-address via DHCP. This can then be found using `$I` from a terminal or maybe also when checking your router for connected devices. The default hostname is "GRBL".

infamous-panda, 2021-01-05:
> I increased the planner buffer to 1400

Is this the reasonable maximum?

terjeio, 2021-01-05:
> Is this the reasonable maximum?

Hard to say - I  have tried increasing it more only for compilation to fail due to lack of memory. Reducing input buffer sizes may free up memory for adding more if needed. Each entry needs about 40 bytes.

And I am not sure if adding too much may slow down the troughput instead of increasing it.

Cncninja, 2021-02-10:
Any updates I still can't even get it to compile, I'm stuck with the precompiled generic option.  when I do my own compile it errors asking for ioport.h ??
[image]

phil-barrett, 2021-02-10:
I just place 3, 4 and 5 axis [precompiled binaries](https://github.com/phil-barrett/grblHAL-teensy-4.x/blob/master/grblHAL_pb1024.zip) in the github repository for the board.

terjeio, 2021-02-10:
> it errors asking for ioport.h ??

Strange, it should not without a configuration change and custom code added based on the [ioports template.](https://github.com/terjeio/grblHAL/tree/test/templates)
What is the build version from _grbl/grbl.h_? It is found in the `#define GRBL_VERSION_BUILD` line.

Cncninja, 2021-02-12:
It was the one in the Grblhal master folder, Phil uploaded a precompiled version and that worked like a charm

infamous-panda, 2021-04-08:
Quick Question. Is there a limit to the decimal precision on the XYXA positions that GRBL / GRBL-HAL will recognize. My custom CAM software generates really small angle helixes (continuous laser engraving on rotary) that we output 4 decimal places typically. I tested a run where I rounded this to 2 decimal places to see if we can improve speed by reducing the amount of code streamed, but it ended up making the machine stutter. I think this is because when the positions in the helix were rounded the motion was no longer smooth. Makes we wonder if we could add 5 or 6 decimal places if it would be better?

terjeio, 2021-04-08:
Internal representation is [single precision](https://en.wikipedia.org/wiki/Single-precision_floating-point_format) floating point and input values are converted to that without any explicit rounding. The default reporting format is 3 decimal places (1 micrometer resolution) for metric and 4 (1 thou resolution) for inches mode. For most machines (all running grbl?) this is likely better than the mechanical precision available and is a waste of bandwidth to use more?

closed 2022-01-27 by terjeio