terjeio/grblHAL #146: DUAL_AXIS_HOMING for anti-racking implemented?

sigxcpu76, 2020-12-19:
I see bits of this feature (from grbl 1.1h) around in the code but some other stuff is missing so it is unclear if this has been ported to grblHAL. https://github.com/gnea/grbl/commit/b75e5571eeaeb22a88304716fb1e7411f9c28be0

The important missing setting is which axis is the one that has dual motors.

The only 32-bit firmware that supports this somewhat is `Smoothieware`, but pausing/stopping is totally unsafe, so it is not suitable for a (big) CNC, `grbl` being the only option.

terjeio, 2020-12-19:
It has been ported but not many drivers/boards have support yet. Configuration is at the board level in the driver, the [T41U5XBB board](https://github.com/terjeio/grblHAL/blob/master/drivers/IMXRT1062/grblHAL_Teensy4/src/T41U5XBB_map.h) for iMXRT1062 (Teensy 4.1) is one of the few that has. I have not yet made any attempt at bringing configuration out in the my_machine.h configuration file in a standarized way - I want to do that at some point.

See issue #45 for a related discussion. Notably the [test branch](https://github.com/terjeio/grblHAL/tree/test) has been [updated](https://github.com/terjeio/grblHAL/issues/45#issuecomment-727875879) with additional functionality related to auto squaring.

sigxcpu76, 2020-12-19:
Thank you. I am running a MKS SBASE but I also have some Arduino Due around as I don't need to use the motor drivers (they are external, closed loop).
On Due I know that there is no NV storage compared to other boards. I've never used grbl on it, so I have one more question: are the settings saved somewhere or I need to carefully pre-configure the firmware in machine defaults?

terjeio, 2020-12-19:
>  are the settings saved somewhere

In flash by default, you can also add an external I2C EEPROM or FRAM (recommended).

Note that only the [cmcgrath_rev3_map.h](https://github.com/terjeio/grblHAL/blob/test/drivers/SAM3X8E/grblHAL_Due/src/cmcgrath_rev3_map.h) for the Due has auto-quaring options - and I that I have not tested this myself.

sigxcpu76, 2020-12-19:
OK, thank you very much. I'll go that route.

closed 2021-02-20 by terjeio