terjeio/grblHAL #147: Usage with external microcontrollers for closed loop control.

Roiki11, 2020-12-19:
Would it be possible or feasible to create a plugin that can access the motion planner output(setpoint, velocity etc) and passes them to an external controller(via serial Etc.) for use in position feedback loop with digital drives and linear encoders. And possibly passes the true, measured position back to grbhal?

terjeio, 2020-12-19:
I am not sure about hooking into the planner as there is continuous recalculation going on. What may be possible is to hook into the step output function `hal.stepper.pulse_start` and send data only on new segments (one is generated every 5 ms). Passing back the the true position for feedback - not trivial and requires rewriting parts of the core?

FYI @shooter64738 has an [ongoing project](https://github.com/shooter64738/Talos-CNC) for a controller that attempts to separate the parser/planner from step generation - now using a dual core MCU if I am not mistaken. It is likely that he understands what is required/possible far better than I do.

Roiki11, 2020-12-19:
Does that function contain the number of steps the stepper will move and the velocity? Or how are these handled.
I'm after the distance and velocity information that is generated by the planner. Which i assume is converted to the steps needed by the steppers. If that function contains that information then that's what I'm after. Which would then be passed on to outside drives.

Passing back the true position would be useful mainly for display purposes on the UI and monitoring following error for alarms. If there's a way the planner keeps track of the current position of the machine, that's where you would like to inject the measured position. though this isn't strictly necessary.

terjeio, 2020-12-20:
> Does that function contain the number of steps the stepper will move and the velocity

Yes, a pointer to this struct is provided as the sole argument: 

https://github.com/terjeio/grblHAL/blob/2569bb233f63f109d2217c0a56b624182af6d9ad/grbl/stepper.h#L78-L105

The current rate (velocity) can be acessed via the pointer to the exec_segment struct. This also has an id field which can be used to track start of a new segment. I do this in [spindle sync code](https://github.com/terjeio/grblHAL/blob/2569bb233f63f109d2217c0a56b624182af6d9ad/drivers/MSP432/driver.c#L309):

A plugin can redirect `hal.stepper.pulse_start` to itself, one example of how this is done can be found in the [odometer plugin](https://github.com/terjeio/grblHAL/blob/master/plugins/odometer/odometer.c).

Monitoring can be done by "subscribing" to `grbl.on_execute_realtime` event?

See [grbl/hal.h](https://github.com/terjeio/grblHAL/blob/master/grbl/hal.h) for more HAL entry points.

Roiki11, 2020-12-21:
I think that's what I'm looking for. I've been looking at the code you pointed as well as the other plugins and I think this should get me started.

Is there a way to pass information back to be sent to the UI? Or would this require it's own plugin as well?

terjeio, 2020-12-21:
> Is there a way to pass information back to be sent to the UI?

There are several ways to do that, you can add elements to the real time report by hooking into `grbl.on_realtime_report()` as I have done in the SD card plugin:

https://github.com/terjeio/grblHAL/blob/2569bb233f63f109d2217c0a56b624182af6d9ad/plugins/sdcard/sdcard.c#L544-L545

Other options are via a [M-code](https://github.com/terjeio/grblHAL/blob/master/templates/mcodes.c), a [$-command](https://github.com/terjeio/grblHAL/tree/master/plugins/odometer) or even via a single character request character of your choice by hooking into `hal.stream.enqueue_realtime_command()`.

You can also push messages when you chose to from the foreground process by adding the message function to a queue:

https://github.com/terjeio/grblHAL/blob/2569bb233f63f109d2217c0a56b624182af6d9ad/grbl/protocol.c#L857-L871

>  Or would this require it's own plugin as well?

Everything can be in the same plugin if that is what you want.

closed 2021-03-15 by Roiki11