terjeio/grblHAL #148: UART times out on Teensy4.x (tested with 4.0)

Hamoudy99, 2020-12-22:
While experimenting with different USB serial settings. I noticed that the teensy becomes unresponsive after a few seconds. I need to be able to use the UART pins (not native USB) for communicating with the teensy. any ideas as to why it doesn't work?

terjeio, 2020-12-22:
> While experimenting with different USB serial settings. I noticed that the teensy becomes unresponsive after a few seconds. 

The USB cable or hub is causing problems? I just tested my T4.0 with both USB stacks and they are working. This is with the [test branch](https://github.com/terjeio/grblHAL/tree/test).

>  I need to be able to use the UART pins (not native USB) for communicating with the teensy. any ideas as to why it doesn't work?

Did you comment out `#define USB_SERIAL_CDC` in my_machine.h for that? I so could it be a wiring/settings problem as this too is working for me. You may have to disable hardware handshake for the port in your computer or ground the CTS pin on the UART breakout board to get transmission going.

Hamoudy99, 2020-12-22:
thanks for the quick reply. 

I solved the problem. I was playing around with software serial and forgot to remove the code when I started to use UART. so there was a conflict between the two libraries. I'm not usually this slow :D a little tired perhaps

closed 2020-12-22 by Hamoudy99