terjeio/grblHAL #160: Error while compiling: lwipopts.h not found

mattbue, 2020-12-29:
Hi,
I'm trying to compile grblHAL for Teensy 4 with networking plugin.
The grbl and plugin files are copied into the corresponding paths within the Teensy 4 driver.
In my_machine.h ethernet is enabled.

When compiling with Arduino IDE (1.8.13) the following error is shown:

`C:\Users\___________\Documents\Arduino\libraries\grblHAL_Teensy4\src\enet.c:30:22: fatal error: lwipopts.h: No such file or directory
compilation terminated.`

Is the file lwipopts.h really missing within the driver directory or am i doing something wrong?
Thank you very much!

terjeio, 2020-12-29:
Do you have the lwIP library linked to from the [driver page](https://github.com/terjeio/grblHAL/tree/master/drivers/IMXRT1062) installed?

mattbue, 2020-12-30:
Oh... I forgot to add the library. Shame on me :-D

Now it's compiling!
But i can't connect to the board.
Here's the part of the config in my_machine.h:

`#define USB_SERIAL_CDC       2 // 1 for Arduino class library and 2 for PJRC C library. Comment out to use UART communication.
//#define USB_SERIAL_WAIT    1 // Wait for USB connection before starting grblHAL.
//#define SPINDLE_HUANYANG   1 // Set to 1 or 2 for Huanyang VFD spindle. Requires spindle plugin.
//#define QEI_ENABLE         1 // Enable quadrature encoder interfaces. Max value is 1. Requires encoder plugin.
#define ETHERNET_ENABLE      1 // Ethernet streaming. Requires networking plugin.
//#define SDCARD_ENABLE      1 // Run gcode programs from SD card, requires sdcard plugin.
//#define KEYPAD_ENABLE      1 // I2C keypad for jogging etc., requires keypad plugin.
//#define PLASMA_ENABLE      1 // Plasma/THC plugin. To be completed.
//#define PPI_ENABLE         1 // Laser PPI plugin. To be completed.
#define ODOMETER_ENABLE      1 // Odometer plugin. To be completed.
//#define EEPROM_ENABLE      1 // I2C EEPROM support. Set to 1 for 24LC16(2K), 2 for larger sizes. Requires eeprom plugin.
//#define EEPROM_IS_FRAM     1 // Uncomment when EEPROM is enabled and chip is FRAM, this to remove write delay.

//#define ESTOP_ENABLE       0 // When enabled only real-time report requests will be executed when the reset pin is asserted.
                               // Note: if left commented out the default setting is determined from COMPATIBILITY_LEVEL.

#if ETHERNET_ENABLE > 0
#define TELNET_ENABLE           1 // Telnet daemon - requires Ethernet streaming enabled.
#define WEBSOCKET_ENABLE        1 // Websocket daemon - requires Ethernet streaming enabled.
#define NETWORK_HOSTNAME        "GRBL"
#define NETWORK_IPMODE          0 // 0 = static, 1 = DHCP, 2 = AutoIP
#define NETWORK_IP              "192.168.99.100"
#define NETWORK_GATEWAY         "192.168.99.1"
#define NETWORK_MASK            "255.255.255.0"
#define NETWORK_TELNET_PORT     23
#define NETWORK_WEBSOCKET_PORT  80
#define NETWORK_HTTP_PORT       80
#endif`

Is the board supporting ping requests? I was sending but got no response.
I also tried connecting with the Windows embedded telnet client using the IP address configured in my_machine.h without success.

terjeio, 2020-12-30:
What is the output from `$I`? and `$$`?

Sometimes restoring defaults is needed when changing the configuration, try with `$RST=*` .

Ping is supported.

mattbue, 2020-12-30:
Output of `$I`:
```
[VER:1.1f(IMXRT1062).20201103:]
[OPT:VNMSL,35,1024,3,0]
[NEWOPT:*FLASH,ES,ETH,TC]
[DRIVER VERSION:201014]
[DRIVER OPTIONS:USB.2]
[BOARD:T41U5XBB]
[IP:0.0.0.0]
```

Output of `$$`:
```
$0=10.0
$1=25
$2=0
$3=0
$4=0
$5=0
$6=0
$10=511
$11=0.010
$12=0.002
$13=0
$14=0
$15=0
$16=0
$17=0
$18=0
$19=0
$20=0
$21=0
$22=0
$23=0
$24=25.000
$25=500.000
$26=250
$27=1.000
$28=0.100
$29=0.0
$30=1000.
$31=0.
$32=0
$33=5000.000
$34=0.000
$35=0.000
$36=100.000
$37=0
$39=1
$40=0
$41=0
$42=2
$43=1
$44=4
$45=3
$46=0
$56=5.000
$57=100.000
$58=-5.000
$59=500.000
$60=0
$61=0
$62=0
$63=2
$64=0
$65=0
$70=3
$100=250.000
$101=250.000
$102=250.000
$110=500.000
$111=500.000
$112=500.000
$120=10.000
$121=10.000
$122=10.000
$130=200.000
$131=200.000
$132=200.000
$300=GRBL
$301=0
$302=192.168.99.100
$303=192.168.99.2
$304=255.255.255.0
$305=23
$307=80
$341=0
$342=30.0
$343=25.0
$344=200.0
```

After adding a DHCP server into the network the output of `$I` is:
```
[VER:1.1f(IMXRT1062).20201103:]
[OPT:VNMSL,35,1024,3,0]
[NEWOPT:*FLASH,ES,ETH,TC]
[DRIVER VERSION:201014]
[DRIVER OPTIONS:USB.2]
[BOARD:T41U5XBB]
[IP:192.168.99.200]
```
The IP range of the DHCP server is from 192.168.99.200 to 192.168.99.254.
Ping and telnet works.

But why does the board get the IP from DHCP?
I have set the ip mode to static (0).

mattbue, 2020-12-30:
Update:
suddenly it worked. The only thing i have done is powering off and on again the board.

phil-barrett, 2020-12-30:
Not a mystery! Network setting changes require a power cycle.

mattbue, 2020-12-30:
Good to know.
Thanks to both of you!
So I close the issue.

closed 2020-12-30 by mattbue