terjeio/grblHAL #165: ESP32: ../driver.c:51:34: fatal error: networking/TCPSTream.h: No such file or directory

Harvie, 2020-12-31:
i cannot build ESP32 port with webui support (or even plain wifi).

terjeio, 2020-12-31:
> Probably something is missing in CMakeLists?

No, I did not update all files I should when changing settings handling recently. Fix committed to test branch.

Harvie, 2020-12-31:
Just updated test branch from your repo and `idf.py build` still fails with this:

```gcc
FAILED: CMakeFiles/grbl.elf.dir/driver.c.obj 
...
../driver.c:51:34: fatal error: networking/TCPSTream.h: No such file or directory
```

Not sure if that helps, but this is my config:

```diff
diff --git a/drivers/ESP32/CMakeLists.txt b/drivers/ESP32/CMakeLists.txt
index 3a050c4..e527eb5 100644
--- a/drivers/ESP32/CMakeLists.txt
+++ b/drivers/ESP32/CMakeLists.txt
@@ -9,19 +9,19 @@ cmake_minimum_required(VERSION 3.5)
 # If none is enabled pin mappings from generic_map.h will be used.
 
 OPTION(BOARD_BDRING_V3P5 "Compile for v3.5 3-axis board" OFF)
-OPTION(BOARD_BDRING_V4 "Compile for bdring v4 3-axis board" ON)
+OPTION(BOARD_BDRING_V4 "Compile for bdring v4 3-axis board" OFF)
 OPTION(BOARD_BDRING_I2S6A "Compile for bdring 6-axis I2S board - untested!" OFF)
 OPTION(BOARD_CNC_BOOSTERPACK "Compile for CNC BoosterPack" OFF)
 OPTION(BOARD_ESPDUINO32 "Compile for ESPDUINO-32 Wemos D1 R32" ON)
 
 # The following plugins/options are supported:
 
-OPTION(Networking "Wifi + protocols" OFF)
-OPTION(Bluetooth "Bluetooth" OFF)
+OPTION(Networking "Wifi + protocols" ON)
+OPTION(Bluetooth "Bluetooth" ON)
 OPTION(Keypad "I2C Keypad" OFF)
 OPTION(SDcard "SD Card Streaming" OFF)
 OPTION(Trinamic "Trinamic driver support over I2C" OFF)
-OPTION(WebUI "WebUI services" OFF)
+OPTION(WebUI "WebUI services" ON)
 OPTION(WebAuth "WebUI authentication" OFF)
 OPTION(MPGMode "MPG mode" OFF)
 OPTION(I2SStepping "Use I2S Stepping" OFF)
```

terjeio, 2020-12-31:
> ../driver.c:51:34: fatal error: networking/TCPSTream.h: No such file or directory

Another spelling mistake masked by Windows... Can you change to `TCPStream.h` until I commit a fix?

Harvie, 2020-12-31:
Good observation :-) Took me a while to realize what are you talking about :-D

Harvie, 2020-12-31:
WsSTream.h as well

Harvie, 2020-12-31:
Now it builds, but seems to be too big:

```
/home/harvie/.espressif/tools/xtensa-esp32-elf/1.22.0-97-gc752ad5-5.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: grbl.elf section `.dram0.bss' will not fit in region `dram0_0_seg'
/home/harvie/.espressif/tools/xtensa-esp32-elf/1.22.0-97-gc752ad5-5.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: DRAM segment data does not fit.
/home/harvie/.espressif/tools/xtensa-esp32-elf/1.22.0-97-gc752ad5-5.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: region `dram0_0_seg' overflowed by 3144 bytes
collect2: error: ld returned 1 exit status
```

terjeio, 2020-12-31:
Disable Bluetooth?

terjeio, 2020-12-31:
Hmm, perhaps reducing stream buffer sizes will help, they are quite generous. And with two serial streams, two network streams and a bluetooth stream this adds up fast. And some (all?) of the input stream buffers are duplicated for the tool change protocol.

Default values are set in grbl/stream.h - you could try to halve them if Bluetooth is essential.

IIRC I have also read somewhere that Wifi and Bluetooth does not work well together?

Harvie, 2020-12-31:
Disabling bluetooth helped with linking... Device takes a little longer to boot, but i see grblHAL console on serial. But i do not see wifi network. Not sure how to setup the wifi as my_machine.h containing SSID/password is not used in esp-idf build...

terjeio, 2020-12-31:
$74=ssid
$75=password

Harvie, 2020-12-31:
> $74=ssid
> $75=password

This is not show in `$HELP settings`

Harvie, 2020-12-31:
I suppose `$73=1` is AP mode while =0 is STA ?

terjeio, 2020-12-31:
> This is not shown in $HELP settings output

I know - I am moving some bits away from the core and $HELP changes was not ready when I did the last commit. Coming back soon.

> I suppose $73=1 is AP mode while =0 is STA ?

It depends on AP mode beeing compiled in:

https://github.com/terjeio/grblHAL/blob/3f1584285940be17887d5bd2af639c2a3a9b6d76/drivers/ESP32/wifi.c#L546-L556

More options should be made available from CMakeLists.txt - WIFI_SOFTAP is one of them...

0 is off, 1 is station and 2 is access point (if compiled in).

terjeio, 2020-12-31:
Just committed another update to the test branch with changes for $HELP.

I wonder what to do with networking defaults, move them to a file shared by both `CMakeLists` config and `my_machine.h` config? I do not know if it is possible to add them to CMakeLists as they are not simple on/off switches.

Harvie, 2020-12-31:
I also noticed another weird thing. Wifi only becames online when i connect to the serial port with bCNC. If i just plug the device into the phone charger, it does not connect to wifi... That somewhat defies the point of having wifi in the first place. Not sure if bluetooth has the same issue.

Harvie, 2020-12-31:
> I wonder what to do with networking defaults

I think it would make sense to have some default AP settings and user will then connect to that AP and use it or provide custom (AP or STA) settings over that default AP.

Harvie, 2020-12-31:
Once grblhal is connected to the wifi, i can connect to it using bCNC over wifi:

[image]

terjeio, 2020-12-31:
> I also noticed another weird thing. Wifi only becames online when i connect to the serial port with bCNC.

Weird for sure as this is not how it behaves for me, just tested ok with WebUI and both telnet and websocket comms with my sender, this with a power cycle between tests. Station mode.

> I think it would make sense to have some default AP settings and user will then connect to that AP (or serial) and use it or provide custom (AP or STA) settings over that default AP.

Default settings are provided for AP mode, but may require a settings reset when enabled:

 https://github.com/terjeio/grblHAL/blob/b7c51ea4725736da4e7e568846a8f949e156a01f/drivers/ESP32/driver.h#L198-L208

> Once grblhal is connected to the wifi, i can connect to it using bCNC over wifi:

Great!

einencool, 2021-01-15:
Also same Failure in the last test builds in the Teensy 4.1

In file included from C:\Users\xxx\Documents\Arduino\libraries\grblHAL_Teensy4\src\networking\TCPStream.c:40:0:
C:\Users\xxx\Documents\Arduino\libraries\grblHAL_Teensy4\src\networking\networking.h:10:22: fatal error: lwipopts.h: No such file or directory
compilation terminated.
Fehler beim Kompilieren für das Board Teensy 4.1.

terjeio, 2021-01-15:
@einencool this is a different issue only affecting the Teensy4 driver when the ethernet library is not installed, I have just committed a fix.

closed 2022-01-27 by terjeio