terjeio/grblHAL #166: SAM3X8E - Pwm maximum resolution

Laserprints, 2020-12-31:
Hi all,

Maybe a noob question, but I'm looking for the correct Pwm resolution of the TIOA8 Pin 11 Pwm output pin of a ATMEL SAM3X8E Arm

12bit - 16bit 

I'm looking to run at a frequency of around 8khz for the operation of a laser diode driver for raster image engraving. 

I want to achieve the maximum resolution of pwm and I'm looking for the correct Grblhal 1.1f $30 setting, RPM Max Setting.

My thoughts is to have the minimum voltage increments via pwm from the dues pin 11 pwm pin. 

Any thoughts or help with this would be greatly appreciated. 

Regards Ryan

phil-barrett, 2020-12-31:
Are you reporting a specific problem or just asking a question?  I think you are asking a question. 

I am not familiar with the 3X8E version but using the [cmcgrath mapping](https://github.com/terjeio/grblHAL/blob/master/drivers/SAM3X8E/Arduino%20Due%20shield%20pin%20mappings/cmcgrath_rev3_map.md) spindle PWM does indeed come out on pin 11.  

For any Grbl, you use $33 for F<sub>PWM</sub> so set that to 8000.  $30 is used to determine PWM duty cycle with the Snnn in GCode - Duty cycle = S-value/$30.  I would leave $30 at the default of 1000.  Then, for example, M4S10 would give you a 1% (10/1000) duty cycle, S100, a 10% DC and so on.

Laserprints, 2020-12-31:
So sorry if, this was the incorrect place for a question,

But many thanks for your reply.

Thats perfect thank you for your help

Laserprints, 2020-12-31:
Happy new year to you and your family 😁🎉🎉

phil-barrett, 2021-01-01:
No problem. The discussions section is good for asking questions. There are a few discussions there about laser printing including talk about high speed results.

And a Happy New Year to you, too!  Let's hope 2021 is a much better year.  2020 set a low bar so it's pretty likely.

I'll go ahead and close this item.

closed 2021-01-01 by phil-barrett