terjeio/grblHAL #173: Manual homing

avizienis, 2021-01-04:
Hi, how does one home manually? This option is mentioned in the wiki and in the config.h (DEFAULT_HOMING_ALLOW_MANUAL) but not documented. Is it enough to set machine coordinates and travel limits? Do I need to uncomment this option for that to work?

terjeio, 2021-01-04:
> Do I need to uncomment this option for that to work?

No - you can enable it by setting `$22=33` (unless COMPATIBILTY_LEVEL is set > 1).

I see now that I have not added the new `$22` bitfields to the wiki, Ill fix that. If you are using the latest  test branch you can get help for settings when connected via a terminal such as PuTTY, eg. `$$22` produces this output:

```
 $22: Homing cycle as bitfield where setting bit 0 enables the rest:
    0 - Enable (1)
    1 - Enable single axis commands (2)
    2 - Homing on startup required (4)
    3 - Set machine origin to 0 (8)
    4 - Two switches shares one input pin (16)
    5 - Allow manual (32)
    6 - Override locks (64)
    7 - Keep homed status on reset (128)
```
You can also use `$help homing` for all  details about homing settings or just `$help` to list help arguments.

When homing move to the home position and send `$H`, this will set this as home for all axes that isn't enabled for automatic homing by setting `$44`, `$45` or `$45` non-zero. You may also manually home a single axis if _Enable single axis commands_ is set.

FYI the latest [ioSender](https://github.com/terjeio/Grbl-GCode-Sender) has support for these settings in the UI, other senders likely not.

avizienis, 2021-01-06:
Thanks, will try that out!

closed 2021-01-06 by avizienis