terjeio/grblHAL #177: Spindle is switched off when speed is 0

mattbue, 2021-01-05:
Hi, i'm running the code on Teensy 4.1 and Phil Barretts board.
The spindle enable signal is controlled by the M codes which works fine.
Now i recognized that the enable signal is also switched off in case the spindle speed is 0.
So sending M3 S0 does not switch on the enable signal.

I would prefer that the enable signal is independent from the actual speed.

Can this be configured?

terjeio, 2021-01-05:
> Can this be configured?

I see I have added a run time setting variable for this but not assigned it to a $-setting, forgotten about?

> So sending M3 S0 does not switch on the enable signal.

Oh - this is a mess, I have checked the original code and behaviour depends on using the spindle direction pin as spindle enable. To match grbl behaviour it seems that I need to add another setting as most (all?) grblHAL drivers do not repurpose the direction pin as enable.

So yes this should be configurable - I'll put it on my todo list.

mattbue, 2021-01-05:
Ok. Thanks!
Shall we keep this issue open until the integration is done? Or shall we close it?

phil-barrett, 2021-01-05:
Would that go under $16 or as a new $ Setting number

mattbue, 2021-01-05:
Does your laser have a 24V PWM input? Mine has a 5V input. So the PWM
signal can be connected directly.

Next days I want to start working with the I2C port expanders. Let's see if
I can manage to develop it as plugin.

Do you know for what the "Backup" button is on the grbl settings page in
GCode Sender Beta8?

phil-barrett, 2021-01-06:
The laser module I got (solidstate 5.5W) has a  controller that takes 12V PWM input.

terjeio, 2021-01-06:
Keep the issue open. 

> Would that go under $16 or as a new $ Setting number

It is likely to be a new one as $16 is for signal inversions.

> Next days I want to start working with the I2C port expanders. Let's see if I can manage to develop it as plugin.

If as a plugin then it will have to rely on a common driver implemented I2C interface. And then only usable for extra inputs and outputs? If for core signals then it IMO best kept as a part of the driver as `driver.c` has to be changed too.

[Here](https://github.com/terjeio/grblHAL/blob/master/drivers/SAMD21/grblHAL_MKRZERO/src/ioexpand.c) is code for a I2C expander implementation I made for the SAMD21 driver (MKRZERO).

> Do you know for what the "Backup" button is on the grbl settings page in GCode Sender Beta8?

It saves the current settings to `settings.txt` in the sender folder. I have to add a _Restore_ button. Currently I restore by pasting the content into PuTTY.

phil-barrett, 2021-01-06:
> It saves the current settings to settings.txt in the sender folder. I have to add a Restore button. Currently I restore by pasting the content into PuTTY.
Very useful feature.  Could I suggest you name the buttons Load Settings and Save Settings? I can see this as very useful for providing different settings for users that are just starting out.

terjeio, 2021-01-10:
> Would that go under $16 or as a new $ Setting number

New setting $7 to be added.

> So sending M3 S0 does not switch on the enable signal.

Looking at the [original code](https://github.com/gnea/grbl/blob/master/grbl/spindle_control.c) this is AFAIKT the correct behaviour. A new setting option is needed to allow that? 

The original is `SPINDLE_ENABLE_OFF_WITH_ZERO_SPEED` which is used to define the action taken on a `S0` command while the spindle is enabled. 
A new `SPINDLE_ENABLE_ON_FOR_ALL_SPEEDS` setting is needed to define action taken on `M3S0` or `M4S0` (or `M3` or `M4` when the saved `S`-value is 0) in order to not break backwards compatibility? This will of course override `SPINDLE_ENABLE_OFF_WITH_ZERO_SPEED`.

closed 2022-01-27 by terjeio