terjeio/grblHAL #188: Alarm while testing with Teensy

frdfsnlght, 2021-01-07:
I'm not sure if this is the right place for this since I don't know what the problem is exactly, but I have to start somewhere.

I'm following the instruction from [this blog page](https://www.grbl.org/single-post/naked-teensy-party). For reference, I'm using a brand new, bare Teensy 4.1.
* I've downloaded the latest version (as of yesterday) of grblHAL from the master branch.
* I haven't played with any of the configuration files, just compiled and uploaded the code using the Arduino IDE (fresh install, v1.8.13).
* I can connect via serial monitor and issue commands.
* I've downloaded GCode Sender Beta 8 and fired it up.
* I've changed the $14 settings exactly as described in the blog ($14=78).
* I click the Reset button in GCode Sender, but the alarm state never clears.

It's not clear what's causing the alarm. I opened the sender console and turned on verbose mode and I see lots of these:

`<Alarm|MPos:0.000,0.000,0.000|Bf:35,1023|FS:0,0|Pn:XYZ>`

I don't know if that helps. I don't know how to interpret that line or if it even contains the information about what's causing the alarm. Playing around, I understand that "Pn:" are the active pins. I've inverted the limit switches and can make the "Pn:XYZ" part go away, but the alarm is still present.

Any idea what's wrong?

terjeio, 2021-01-07:
Enable _Alarm substatus_ for $10 (add 1024 to the current value) to get the alarm code after `<Alarm` in the realtime report. From that the reason could be determined.

frdfsnlght, 2021-01-07:
Ok, did that. Now I see this:

`<Alarm:10|MPos:0.000,0.000,0.000|Bf:35,1023|FS:0,0|Pn:XYZ>`

terjeio, 2021-01-07:
That is EStop asserted (reset pin). Perhaps the pin is floating since it does not show up in the Pn: element? Can you ground it or connect it to 3.3V (depending on the $14 setting)? Which pin it is depends on which board you have compiled for.

FYI the [edge version](http://io-engineering.com/downloads) of the sender has improved startup reporting for such circumstances.

frdfsnlght, 2021-01-07:
I've compiled for Phil's Teensy 4.1 BoB. The Estop input (as well as all of the other "standard" inputs) are opto-isolated. There's no "floating", really. I connected to both ground and 3.3V and there's no change in the alarm status.

terjeio, 2021-01-07:
Hmm, very strange. Broken Teensy? Or faulty breakout board?
One option to try is to uncomment //#define ESTOP_ENABLE in my_machine.h
Another option is to comment out both `signals.e_stop` and `signals.reset` assignments in `systemGetState()` in driver.c.

phil-barrett, 2021-01-07:
Have you issued a $X?  

I am a little confused, is this just a bare teensy, on my BOB or something else?  What is the voltage at pin 14 on the teensy? 

What are your my_machine.h settings?

frdfsnlght, 2021-01-07:
I've got it working, but I don't understand why.

First off, like I said, it's a brand new, bare Teensy 4.1. No Bob. I haven't even soldered pins on it yet. I didn't make any changes to any of the configs, including the my_machine.h file. I'm just using the files as checked of git.

To get it to work, I quit GCode Sender and opened the Arduino Serial monitor. I did a "$$" and just to be sure, a "$14-78", and another "$$". Then I did a "$X". I then quit the monitor and started GCode Sender which no longer shows an alarm.

What I don't understand is, according to the blog post, the "$X" does a reset. Isn't that what the "Reset" button in the GCode Sender UI does? I can't tell because the GCode Sender console doesn't seem to show what the sender is sending, only what it's receiving?

Maybe related... none of the buttons I click in GCode Sender give _any_ feedback they've been clicked. I would normally expect them to "depress" or change somehow when they're clicked. Maybe the "Reset" button isn't actually getting clicked so a reset isn't being sent? Even the "Save" button in the Setting page doesn't provide click feedback. The buttons have hover feedback, which is slightly odd (and in my opinion not necessary). Click feedback is very important in a UI.

Another problem I now see... once I have it working as above, if I change $14 and uncheck Estop and save, the main page now shows the Estop input as triggered, which is correct. I can still jog the machine using the Jog popout. Should I be able to do that? And if I click the "Reset" button, nothing changes. I would think grbl should go into an alarm state at that point, no? Maybe I'm not understanding the function of "Reset"?

phil-barrett, 2021-01-07:
I will fix the blog,  $X clears the alarm.  True reset is not possible with the Teensy. Perhaps it is a subtle distinction.

frdfsnlght, 2021-01-07:
I understand, but I take issue with "true reset is not possible with the Teensy". I've written a lot of code for the Teensy. I'm pretty sure I've written reset code. From what I can remember, it's a bit hacky, but you can reset a Teensy, hard (I think did [this](https://forum.pjrc.com/threads/44857-How-to-Reset-Restart-Teensy-3-5-using-sotware)). But I understand that "$X" isn't necessarily supposed to do a hard reset, just clear the alarm.

But what is the "Reset" button in the GCode Sender supposed to do? I would think it's supposed to do a "$X" too? Because it doesn't seem to do that.

And I should mention, while playing around, I unplugged the USB cable from Teensy to do a reboot, while GCode Sender was connected. GCode Sender didn't seem to like that and crashed after going unresponsive. That's not a good reaction. Checking to see if the serial port (or Telnet connection) goes away and auto reconnect when it comes back is a much better user experience. I suppose feedback like that belongs in the GCode Sender issues area though.

phil-barrett, 2021-01-07:
OK, perhaps "not possible" was too strong but grblHAL doesn't do it. On other processors like the STM32 family, it is pretty simple.

Yes, USB dis/re connect behavior is a bit rough. I think it is a grbl/gnea legacy thing.

terjeio, 2021-01-07:
> I understand, but I take issue with "true reset is not possible with the Teensy". I've written a lot of code for the Teensy. I'm pretty sure I've written reset code.

I have written reset code for the Teensy too, ESC CTRL-T does that with grblHAL if the driver tells the core about that capability. The Teensy driver does. Not a good thing to do if connected via a native USB or a network connection though. Ok for a UART connection. Making the sender able to gracefully restart after a connection loss is IMO no easy thing to achive. It is in my todo list to check out how I can listen to a disconnected event so that a message can be given before the sender exits.

> But what is the "Reset" button in the GCode Sender supposed to do?

It sends ctrl-X to the controller. Action taken by the controller depends on the state it is in.

> Yes, USB dis/re connect behavior is a bit rough. I think it is a grbl/gnea legacy thing.

Kind of, since legacy grbl controllers does not use native USB comms connection is not dropped on a hard reset.

terjeio, 2021-01-07:
> Maybe related... none of the buttons I click in GCode Sender give any feedback they've been clicked. I would normally expect them to "depress" or change somehow when they're clicked.

All buttons changes the background colour when they are clicked - and they also changes the background colour to a lighter variant of the clicked colour when hovering over them. Could it be that you have a Windows theme installed that does not do that? All buttons except _Home_ and _Reset_ uses the default WPF style, I assume this is compliant with Microsoft guidelines as I only use their libraries (except for the 3D rendering).
BTW I am on Win7.

frdfsnlght, 2021-01-07:
I'm on Windows 10, default theme. I'm also not color blind. I've looking _very_ closely and I can see a _barely_ different button color on click, nothing a typical user would be able to discern from the hover color. I took 2 screen captures, one when hovering, one when clicking. I then looked at the RGB codes for the button backgrounds. The hovered button is R=190, G=230, B=253. The clicked button is R=196, G=229, B=246. As you can see, that's basically the same color.

Hover: [image]
Click: [image]

When you say the Reset button sends a ctrl-X, do you mean ASCII code 0x18 (dec 24)? That doesn't sound very grbl-ly. I don't profess to know anything about grbl, but everything I've seen seems to be purely text based and doesn't include non-visible control codes. What's the controller's expected behavior when receiving a ctrl-X vs a $X?

frdfsnlght, 2021-01-08:
And on the topic of disconnection... I know it's a pain to handle as I've had to do it myself. My method was to detect when the connection goes away (network or comm port, I've done both), disable the entire UI with an overlay that says something like "Disconnected". In the background, the code is trying to reconnect at some small interval, like every second. When a connection happens, remove the overlay, renable the UI and run the same code that runs when the app first starts and connects. I don't try to preserve any state on the front end, basically make it act like you just restarted the app. I find that user experience falls somewhere in the "not surprising" area for most users and isn't quite as jarring as the whole app just quitting and the user having to restart it manually. This behavior is entirely client side and requires nothing on the firmware side (at least it didn't for me). I suppose there could be changes on the firmware side to handle disconnects but I imagine for grbl that would be dependent on it's current state.

Just my $0.02.

terjeio, 2021-01-08:
> What's the controller's expected behavior when receiving a ctrl-X vs a $X?

frdfsnlght, 2021-01-08:
Ah, I understand now. Thanks for the link. I'm still getting (or trying to) a handle on all-things-grbl.

Looking at other native Windows 10 apps, it appears the hover color is a lot lighter and click color darker. In GCode Sender, the hover color (and click color) seem to be close to what I see in other apps as the click color.

I'm not asking you to override the WPF defaults you're using. I was just surprised by the result. It almost sounds like MS needs to fixe something in their WPF defaults for Windows 10.

I hope to eventually be able to contribute more directly to this fine project. Once I get a better handle on how everything's supposed to work and how all the code is organized, I'll be able to dig in and perhaps submit useful pull requests. Someday.

Thanks for the help.

closed 2021-01-08 by frdfsnlght