terjeio/grblHAL #189: STM32F411 nucleo issues/questions.

phil-barrett, 2021-01-08:
I have this mostly working but a couple of small problems. I added Axis B. Put B_STEP_PIN on PC7, B_DIRECTION_PIN on PA10 and B_STEPPERS_DISABLE_PIN on PB7. Works ok. Limit pins are on Port C - 10,11,12,13 and 14. I can not get Limit B (PC14) to work. I have moved Limit B to other pins that do work as a limit pin and it still does not work. I did make some changes to driver.c to support Axis B.  Obviously, I missed something. I attached my map file. Any help would be greatly appreciated.

Also, I can't get PC15 to work at all.  Nothing seems to work on that pin.  Being a complete STM32 newbee, I am probably missing something really obvious.

Changes to driver.c:
after line 203 added:
```
#ifndef B_LIMIT_PORT
  #define B_LIMIT_PORT LIMIT_PORT
#endif
```

After line 244 added:
```
   #if N_AXIS > 4
   BITBAND_PERI(B_STEPPERS_DISABLE_PORT->ODR, B_STEPPERS_DISABLE_PIN) = enable.b;
   #endif
```

After line 497 added:
```
  #ifdef B_LIMIT_PIN
    signals.b = BITBAND_PERI(B_LIMIT_PORT->IDR, B_LIMIT_PIN);
  #endif
```
After line 505 added:
```
  #ifdef A_LIMIT_PIN
    signals.b = (bits & B_LIMIT_BIT) != 0;
  #endif
```
My map file F411BB5X_map.h
```
/*
  F411BB5X_map.h - driver code for STM32F4xx ARM processors

  Part of GrblHAL

  Copyright (c) 2020 Terje Io
  additional Copyright (c) 2021 PL Barrett

  Grbl is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.

  Grbl is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License
  along with Grbl.  If not, see <http://www.gnu.org/licenses/>.
*/

#define HAS_BOARD_INIT
#define SPINDLE_SYNC_ENABLE

void board_init (void);

// Define step pulse output pins.
#define STEP_PORT       GPIOC
#define X_STEP_PIN      0
#define Y_STEP_PIN      5
#define Z_STEP_PIN      9
#define X_STEP_BIT      (1<<X_STEP_PIN)
#define Y_STEP_BIT      (1<<Y_STEP_PIN)
#define Z_STEP_BIT      (1<<Z_STEP_PIN)

#if N_AXIS > 3
#define A_STEP_PIN      6
//#define A_STEP_PIN      15
#define A_STEP_BIT      (1<<A_STEP_PIN)
#endif

#if N_AXIS > 4
//#define B_STEP_PIN 		15
#define B_STEP_PIN 		7
#define B_STEP_BIT      (1<<B_STEP_PIN)
#endif

#if N_AXIS < 4
#define STEP_MASK       (X_STEP_BIT|Y_STEP_BIT|Z_STEP_BIT) // All step bits
#endif

#if N_AXIS == 4
#define STEP_MASK       (X_STEP_BIT|Y_STEP_BIT|Z_STEP_BIT|A_STEP_BIT) // All step bits
#endif

#if N_AXIS > 4
#define STEP_MASK       (X_STEP_BIT|Y_STEP_BIT|Z_STEP_BIT|A_STEP_BIT|B_STEP_BIT) // All step bits
#endif

#if N_AXIS > 5
#error F411BB5X_map >5 Axes not supported
#endif

#define STEP_OUTMODE GPIO_MAP

// Define step direction output pins.
#define DIRECTION_PORT      GPIOA

#define X_DIRECTION_PIN     0
#define Y_DIRECTION_PIN     4
#define Z_DIRECTION_PIN     15
#define X_DIRECTION_BIT     (1<<X_DIRECTION_PIN)
#define Y_DIRECTION_BIT     (1<<Y_DIRECTION_PIN)
#define Z_DIRECTION_BIT     (1<<Z_DIRECTION_PIN)

#if N_AXIS > 3
//#define A_DIRECTION_PIN     10
#define A_DIRECTION_PIN     9
#define A_DIRECTION_BIT     (1<<A_DIRECTION_PIN)
#endif

#if N_AXIS > 4
#define B_DIRECTION_PIN     10
#define B_DIRECTION_BIT     (1<<B_DIRECTION_PIN)
#endif

#if N_AXIS < 4
#define DIRECTION_MASK      (X_DIRECTION_BIT|Y_DIRECTION_BIT|Z_DIRECTION_BIT) // All direction bits
#endif

#if N_AXIS == 4
#define DIRECTION_MASK      (X_DIRECTION_BIT|Y_DIRECTION_BIT|Z_DIRECTION_BIT|A_DIRECTION_BIT) // All direction bits
#endif

#if N_AXIS > 4
#define DIRECTION_MASK      (X_DIRECTION_BIT|Y_DIRECTION_BIT|Z_DIRECTION_BIT|A_DIRECTION_BIT|B_DIRECTION_BIT) // All direction bits
#endif

#define DIRECTION_OUTMODE   GPIO_MAP

// Define stepper driver enable/disable output pin.a2

#define X_STEPPERS_DISABLE_PORT   GPIOA
#define X_STEPPERS_DISABLE_PIN    1
#define Y_STEPPERS_DISABLE_PORT   GPIOB
#define Y_STEPPERS_DISABLE_PIN    12
#define Z_STEPPERS_DISABLE_PORT   GPIOB
#define Z_STEPPERS_DISABLE_PIN    1
#define X_STEPPERS_DISABLE_BIT    (1<<X_STEPPERS_DISABLE_PIN)
#define Y_STEPPERS_DISABLE_BIT    (1<<Y_STEPPERS_DISABLE_PIN)
#define Z_STEPPERS_DISABLE_BIT    (1<<Z_STEPPERS_DISABLE_PIN)
#if N_AXIS > 3
#define A_STEPPERS_DISABLE_PORT   GPIOB
#define A_STEPPERS_DISABLE_PIN    6
#define A_STEPPERS_DISABLE_BIT    (1<<A_STEPPERS_DISABLE_PIN)
#endif

#if N_AXIS > 4
#define B_STEPPERS_DISABLE_PORT   GPIOB
#define B_STEPPERS_DISABLE_PIN    7
#define B_STEPPERS_DISABLE_BIT    (1<<B_STEPPERS_DISABLE_PIN)
#endif

// Define homing/hard limit switch input pins.
#define LIMIT_PORT       GPIOC
#define X_LIMIT_PIN      10
#define Y_LIMIT_PIN      11
#define Z_LIMIT_PIN      12
#define X_LIMIT_BIT      (1<<X_LIMIT_PIN)
#define Y_LIMIT_BIT      (1<<Y_LIMIT_PIN)
#define Z_LIMIT_BIT      (1<<Z_LIMIT_PIN)
#if N_AXIS > 3
#define A_LIMIT_PIN      13
#define A_LIMIT_BIT      (1<<A_LIMIT_PIN)
#endif

#if N_AXIS > 4
#define B_LIMIT_PIN      14
#define B_LIMIT_BIT      (1<<B_LIMIT_PIN)
#endif

#if N_AXIS < 4
#define LIMIT_MASK       (X_LIMIT_BIT|Y_LIMIT_BIT|Z_LIMIT_BIT) // All limit bits
#endif

#if N_AXIS == 4
#define LIMIT_MASK       (X_LIMIT_BIT|Y_LIMIT_BIT|Z_LIMIT_BIT|A_LIMIT_BIT) // All limit bits
#endif

#if N_AXIS > 4
#define LIMIT_MASK       (X_LIMIT_BIT|Y_LIMIT_BIT|Z_LIMIT_BIT|A_LIMIT_BIT|B_LIMIT_BIT) // All limit bits
#endif

#define LIMIT_INMODE GPIO_BITBAND

  // Define spindle enable and spindle direction output pins.
#define SPINDLE_ENABLE_PORT         GPIOB
#define SPINDLE_ENABLE_PIN          3
#define SPINDLE_ENABLE_BIT          (1<<SPINDLE_ENABLE_PIN)
#define SPINDLE_DIRECTION_PORT      GPIOB
#define SPINDLE_DIRECTION_PIN       5
#define SPINDLE_DIRECTION_BIT       (1<<SPINDLE_DIRECTION_PIN)

// Define spindle PWM output pin.
#define SPINDLE_PWM_PORT            GPIOA
#define SPINDLE_PWM_PIN             8
#define SPINDLE_PWM_BIT             (1<<SPINDLE_PWM_PIN)

// Define flood and mist coolant enable output pins.
#define COOLANT_FLOOD_PORT          GPIOB
#define COOLANT_FLOOD_PIN           10
#define COOLANT_FLOOD_BIT           (1<<COOLANT_FLOOD_PIN)
#define COOLANT_MIST_PORT           GPIOB
#define COOLANT_MIST_PIN            4
#define COOLANT_MIST_BIT            (1<<COOLANT_MIST_PIN)

// Define user-control controls (cycle start, reset, feed hold) input pins.
#define CONTROL_PORT                GPIOC
#define CONTROL_RESET_PIN           8
#define CONTROL_FEED_HOLD_PIN       3
#define CONTROL_CYCLE_START_PIN     4
#define CONTROL_SAFETY_DOOR_PIN     1
#define CONTROL_INMODE GPIO_MAP
#define CONTROL_RESET_BIT           (1<<CONTROL_RESET_PIN)
#define CONTROL_FEED_HOLD_BIT       (1<<CONTROL_FEED_HOLD_PIN)
#define CONTROL_CYCLE_START_BIT     (1<<CONTROL_CYCLE_START_PIN)
#define CONTROL_SAFETY_DOOR_BIT     (1<<CONTROL_SAFETY_DOOR_PIN)
#define CONTROL_MASK                (CONTROL_RESET_BIT|CONTROL_FEED_HOLD_BIT|CONTROL_CYCLE_START_BIT|CONTROL_SAFETY_DOOR_BIT)

// Define probe switch input pin.
#define PROBE_PORT                  GPIOB
#define PROBE_PIN                   15
#define PROBE_BIT                   (1<<PROBE_PIN)

// Spindle encoder pins.

#define SPINDLE_INDEX_PORT  GPIOA
#define SPINDLE_INDEX_PIN   7
#define SPINDLE_INDEX_BIT   (1<<SPINDLE_INDEX_PIN)

#define SPINDLE_PULSE_PORT  GPIOD
#define SPINDLE_PULSE_PIN   2
#define SPINDLE_PULSE_BIT   (1<<SPINDLE_PULSE_PIN)

// Auxiliary I/O
#define AUXINPUT0_PORT  GPIOA
#define AUXINPUT0_PIN   5
#define AUXINPUT0_BIT   (1<<AUXINPUT0_PIN)
#define AUXINPUT1_PORT  GPIOA
#define AUXINPUT1_PIN   6
#define AUXINPUT1_BIT   (1<<AUXINPUT1_PIN)

#define AUXOUTPUT0_PORT GPIOB
#define AUXOUTPUT0_PIN  14
#define AUXOUTPUT0_BIT  (1<<AUXOUTPUT0_PIN)
#define AUXOUTPUT1_PORT GPIOB
#define AUXOUTPUT1_PIN  2
#define AUXOUTPUT1_BIT  (1<<AUXOUTPUT1_PIN)

#if KEYPAD_ENABLE
#define KEYPAD_PORT         GPIOB
#define KEYPAD_STROBE_PIN   0
#define KEYPAD_STROBE_BIT   (1<<KEYPAD_STROBE_PIN)
#endif

#if SDCARD_ENABLE
#define SD_CS_PORT  GPIOC
#define SD_CS_PIN   8
#define SD_CS_BIT   (1<<SD_CS_PIN)
#endif

// EOF

```

darkfibre-nl, 2021-01-08:
Have a look at the ST document UM1724 (UM1724, User manual, STM32 Nucleo-64 boards).

On page 20, solder bridges are being used to enable / disable the use of the 32kHz LSE crystal. 

SB48,49  (X2 crystal)(3) 
OFF - X2, C31, C32, R34 and R36 deliver a 32 kHz clock. PC14, PC15 are not connected to CN7. 
ON - PC14, PC15 are only connected to CN7. Remove only R34, R36 

Might this be the cause of PC14/15 not functioning like you expect? I verified this somewhat using the stm32CubeMX tool, the default setting for the nucleo F411 board shows that the 2 pins are in use by the LSE oscillator.

phil-barrett, 2021-01-08:
Bingo. Thanks!  Should have looked at the schematic in the first place.

Pin assignment now verified except for digital inputs. have to figure out to test those.

phil-barrett, 2021-01-08:
As to the Lim B, it now works so I must have gotten a wrong pin somewhere.  The Morpho pinout is an exercise in frustration - just finding the right pin to hook up to is inducing massive eyestrain!

terjeio, 2021-01-08:
> Pin assignment now verified except for digital inputs. have to figure out to test those.

Simple testing can be done with M66. It will terminate early if the condition is met. You can find code for my  Nucleo board [here](https://github.com/terjeio/grblHAL/blob/test/drivers/STM32F4xx/Src/st_morpho.c).

Another way is to hijack the probe input and read this from the pin you want to test. Testing encoder input will be harder...

phil-barrett, 2021-01-08:
Thanks! lot of crossing in email going on!

closed 2021-01-08 by phil-barrett