terjeio/grblHAL #190: spi.c in STM32F4xx implementation faulty

huegene, 2021-01-08:
Hi, 
while porting the spi (spi.h and spi.c) functionality from STM32F4xx to STMF1xx I noticed that the spi_init function can only be run once ( static init variable preventing a second time) 
once spi_disable function is run, the spi_enable function can no longer reenable the SPI interface
instead of the static variable "init", the "hspi1.State" should be used to indicate that the interface has been enabled. This way the spi_disable would have acces to the same hspi.State variable to reset it and make reinitialization possible 

kind regards 
huegene

terjeio, 2021-01-08:
IMO spi_disable() should be removed since I have now started adding Trinamic SPI driver support. FatFS which was set up to call this previously can no longer be allowed to disable SPI. Perhaps you are porting for an earlier version of the test branch?

I am curious about why porting is needed, is it for using another SPI port? A new feature?

huegene, 2021-01-08:
Currently I am editing the drivers/STM32F1xx branch (cloned today). In the attempt to include Trinamic  drivers, compilation failed, because spi.c and spi.h where missing in Src and Inc folder. After I copied them from the STM32F4xx branch, I started changing the necessary pieces to adjust it for the F103 µC . Which is when I noticed the "static init=0;" line. 
Some minor porting (GPIO Init) was needed because of differences between the stm32F4 and STM32F1

terjeio, 2021-01-08:
> Currently I am editing the drivers/STM32F1xx branch (cloned today)

Ok, so you are backporting from F4xx to F1xx. FYI I have just made a driver for the F303 Blackpill which is a lot more capable than the aging (and cloned/faked) F103. See #105.

Dod you clone from master or [test branch](https://github.com/terjeio/grblHAL/tree/test)? There are a lot of changes in the test branch due for merge with the master.

huegene, 2021-01-08:
> Ok, so you are backporting from F4xx to F1xx.

Sorry that I have not stated that more clearly. 

> FYI I have just made a driver for the F303 Blackpill
Ahh, great I guess I'll look into the test branch then. Sadly both (F103 and F303) boards are currently out of stock at Robotdyn (sadly they seem out of stock for quite a while now). I happen to have a few F103 blackpills. Fortunatly both boards seem Pin-compatible as I have just made a board for the F103  blackpill. 

I guess this issue can be  considered closed then. 
Thank you for your time and effort!
kind regards

closed 2021-01-08 by huegene

terjeio, 2021-01-09:
test branch is now updated, added [F303 driver](https://github.com/terjeio/grblHAL/tree/test/drivers/STM32F3xx). If you have I2C EEPROM/FRAM on your board (recommended) then the F303 is not pin compatible with the F103 pills.