terjeio/grblHAL #200: STM32F411 PWM output mapping

avizienis, 2021-01-16:
Hi, I'm experimenting with a black pill mapping but spindle PWM out doesn't seem to work on either PB1 or PB10, which are supposed to be PWM capable. Does the driver have to be adapted to assign a different timer?

terjeio, 2021-01-16:
> Does the driver have to be adapted to assign a different timer?

Yes, a different timer and/or a different channel on the same timer. Quite a bit of work to add all PWM capable pins as an option "out-of-the-box". _Table 9 Alternate function mapping_ in th STM32F411 data sheet (DocID026289 Rev 7) has a list of possible combinations. IMO the only viable option is to add additional pins as needed.

avizienis, 2021-01-16:
Thank you, I certainly won't mess with the internals. PA8 it is then.
By the way, I noticed that one of the stock mappings (uno_map.h) has PWM on PA7. According to the table it's TIM1_CH1N instead of TIM1_CH1. Not sure if that'll cause any issues.

terjeio, 2021-08-21:
The latest build has more options for assigning PWM to pins, and it is easier to extend by adding a few lines in [driver.h](https://github.com/grblHAL/STM32F4xx/blob/3c515180c35d5e43b44c8b30b12d2226058a8a7c/Inc/driver.h#L148-L177).

closed 2021-08-21 by terjeio