terjeio/grblHAL #201: Question for Parking Option

einencool, 2021-01-16:
Hello Terje,

Today I tried the Parking Feature with the Safety Door Switch.

When I press the button (open the Door), the Spindle turns off and then it begins to drive up.
Would it be possible, that the spindle keeps spinning until the "pullout Distance" is reached?

It seems, that when I "close the door" then the Z-Axis drives to the pullout Distance point, and then turns on the spindle and waits for a few seconds. So here it is for me handled correctly.
Maybe it's configurable, but I didn't see it 👍 
Thank you and have a nice weekend 💯

terjeio, 2021-01-23:
Sorry for answering a bit late, there has been a lot going on lately.

I guess you have `$61` _Ignore door when idle_ enabled. This is a new feature I added for my laser so I could jog while the door is open, this since I then have a red aiming laser activated for positioning. When `$61` is enabled the spindle (laser) will be switched off immediatedly for safety reasons. I will change the code to only do this only when the controller is in laser mode.

Have a nice weekend!

einencool, 2021-01-23:
Good morning Terje,

Yes you are right, I have enabled it, and since I didn't test it before I didn't know that this option was new with a different behavior :-)

Thank you and have a nice weekend too

closed 2021-02-20 by terjeio