terjeio/grblHAL #202: Simple i2c Jogging plugin

ElfyShort, 2021-01-17:
Hello,

I have been using IMXRT1062 grblHAL older version since March 2020 on my CNC with no problems at all.
Absolutely loving it!

Now I faced with a need to make a simple jogging solution for Z axis.
In my case 4 buttons, each handling predefined GCODE line would have been sufficient.

At first I thought to build an ATTiny based button handler and send a GCODE lines directly to a second Serial port (if it was available, like in Marlin for example - you can enable 2 serials at once).

Since there is no 2nd Serial available another option is to send simple 1-2 bit commands using i2c (like 'U' or 'D') and use a simple plugin that will bind a hard-coded GCODE line to those commands and send them to Serial directly.

I know there is already a more sophisticated keypad plugin for that, but I do not quite understand how to use that one (like if I send 'U' how HAL will know what distance and speed should be).

If I were to write such a plugin it would take month to understand how to use already implemented HAL functions (like I2C_GetKeycode() or hal.stream.enqueue_realtime_command() ).
Therefore I am asking for some help/advice.

Thanks!

terjeio, 2021-01-17:
> I know there is already a more sophisticated keypad plugin for that, but I do not quite understand how to use that one (like if I send 'U' how HAL will know what distance and speed should be).

There are a bunch of settings for that that shows up when the keypad plugin is activated:

[image]

Distance is controlled by the "strobe" signal, as long as this is held low jogging will continue - when set high motion will stop (except in step mode when motion stops when the step is completed).

> If I were to write such a plugin it would take month to understand how to use already implemented HAL functions (like I2C_GetKeycode() or hal.stream.enqueue_realtime_command() ).

There is no need to write a new plugin, you can use the existing with no modifications.
The protocol is quite simple, when a key is pressed the ATTiny code should pull the "strobe" pin low. The plugin will then request a byte via I2C, the ATTiny should then respond with the keycode. The plugin will then decode that and issue the corresponding command:

`U` = up
`D` = down
`h` - toggle jog mode: fast -> slow -> step -> fast...

A button for generating `h` is recommended.

The default I2C address is `0x49`.

ElfyShort, 2021-01-17:
Thanks for clarification, Terjeio. I think this can work. Also save me 1 button XD.
Have a few more questions though.

Does the screenshot represents '$' commands that will become available if I activate the keypad plugin?
Like $50=150 will set step jog speed to 150?

The strobe signal means it should continue sending 'U' every n milliseconds as long as I hold down the button, if I am at slow or fast mode? Or should it send one 'U' as i push and one more as I release? 

How my ATTiny would know what mode is set after I cycle using 'h' command? Will HAL send back some kind of echo?
This way I would be able to set an LED to tell me the current mode.
Another thing is how ATTiny will know what state we are now if I just turned on the CNC and have not cycled 'h' yet?

In principle I can sync them once and save the mode state in ATTiny EEPROM every time it changes, but it does not seem reliable)

terjeio, 2021-01-17:
> Does the screenshot represents '$' commands that will become available if I activate the keypad plugin?
Like $50=150 will set step jog speed to 150?

Yes. And you can activate the plugin without  a keypad connected to see for yourself. The latest versions of the controller (currently in test branch only) and ioSender has a new way of handling settings so the UI is a bit different from before. 

> The strobe signal means it should continue sending 'U' every n milliseconds as long as I hold down the button, if I am at slow or fast mode? 

No, the plugin will _request_ the keycode once on a falling "strobe" signal by an I2C read command. When the "strobe" signal goes high it sends a jog cancel command to the controller. So you should not push the keycode.

> How my ATTiny would know what mode is set after I cycle using 'h' command? Will HAL send back some kind of echo?

Good question. No information is sent back by default, a bit of code has to be added for that. I have done this in a customized version for my CO2 laser - by sending a I2C command to the keypad controller. I did this by subscribing to a keypad plugin event, `keypad.on_jogmode_changed`. I could add a similar one as an inbuilt option to the protocol to make it simpler.
Any registered event handler will be called on startup too so there is no need to keep track of state in the ATTiny.

ElfyShort, 2021-01-17:
> No, the plugin will request the keycode once on a falling "strobe" signal by an I2C read command. When the "strobe" signal goes high it sends a jog cancel command to the controller. So you should not push the keycode.

Is "strobe" signal is something that will go up after Wire.beginTransmission() and go low with Wire.endTransmission() in terms of using standard Arduino Wire library?

> I did this by subscribing to a keypad plugin event, keypad.on_jogmode_changed. I could add a similar one as an inbuilt option to the protocol to make it simpler.

So essentially if you implement this there will be a space in the plugin where I could change/set what my echo message will be on particular-mode-is-set event? I think as long as this feedback msg is standardized and documented somewhere it should work for any case :)

terjeio, 2021-01-17:
> Is "strobe" signal is something that will go up after Wire.beginTransmission() and go low with Wire.endTransmission() in terms of using standard Arduino Wire library?

It should go low on button down and high on button up. Simple as that. Preferably the output should be open drain so that other I2C devices can share the same input pin. For the MSP430 I use I fake open drain by switching the pin to input mode when the strobe input is to be high (it is pulled up by the controller). I do not have much of knowledge about Arduino libraries, I assume it is possible to react to a I2C read request - possibly via an interrupt handler?

_Button down > strobe low > plugin issues an I2C read request > respond with keycode -> button up -> strobe high._

Not that the plugin may skip issuing the request if the strobe signal went high early or if the controller is in a state where no meaningful action can be taken.

> So essentially if you implement this there will be a space in the plugin where I could change/set what my echo message will be on particular-mode-is-set event?

You can do this already by customizing the driver.

> I think as long as this feedback msg is standardized and documented somewhere it should work for any case, whatever it is :)

Yes, I was thinking about adding a standardized I2C message as a compile time option for the plugin. It could be a simple as writing a byte with predefined bits. In my CO2 laser I do that - IIRC two bits for jog mode and a number of other bits for status LEDs.

ElfyShort, 2021-01-17:
> I assume it is possible to react to a I2C read request - possibly via an interrupt handler?
Button down > strobe low > plugin issues an I2C read request > respond with keycode -> button up -> strobe high.

Now it is clear! As it turned out there is Wire.onRequest(handler), that can do the trick :)

> I was thinking about adding a standardized I2C message as a compile time option for the plugin. It could be a simple as writing a byte with predefined bits. In my CO2 laser I do that - IIRC two bits for jog mode and a number of other bits for status LEDs.

It would be awesome if you do. Maybe we can leave this case open until you post a feedback byte map so anyone will know how to decode/use it?

Thanks again for helping!

closed 2022-01-27 by terjeio