terjeio/grblHAL #204: Consistent step loss

LaurentLM, 2021-01-20:
Hello all,

I am having a rather strange issue with my Nucleo F446 where my X axis is loosing steps.
It happens very consistently by running the following:

> G1 X-140 F250
> G0 X0

The position at the end of travel doesnt match the start (it is off by about 0.7mm, this is not a backlash [issue).](url)

This is a bit odd as the motors are definitely not stalling. I tried to change the settings to something less agreessive (acceleration, max axis speed, feeds) and it still happens.
In my investigation I tried flashing Grbl Advanced (with the same settings) and running the same gcode. The problem went away \o/

See below a photo of running 2 times the program and marking the start/end position:
[image]

I am wondering if that could be due to some wrong clock frequency/prescaler settings. I am running the current code with the symbols NUCLEO_F446 & STM32F446xx (I tried both release and debug build configs). 

If anyone have any clue why this is happening, I would be very grateful. This has been driving me a bit mad. 
Happy to provide more details on the machine if needed.

[grblHAL_settings2021-01-20.txt](https://github.com/terjeio/grblHAL/files/5845798/grblHAL_settings2021-01-20.txt)

terjeio, 2021-01-21:
This is odd, I have just tested with my machine simulator and the pulse count is correct - also when running a 9000+ lines test program. This with the latest [test branch](https://github.com/terjeio/grblHAL/tree/test) version - 20210112.

What kind of drivers do you have?

LaurentLM, 2021-01-21:
I am using DM542T drivers. As part as of this troubleshooting I added a 74HCT541 to make sure they don't miss steps because of TTL levels (that hasn't changed anything, but I'll keep it).
The fact it works ok with Grbl Advanced makes me think this is to do with the firmware. I was able to reproduce this by switching from grblHAL and grbl advanced several times.

I will try to characterise a bit more specifically when and where this happens with a dial indicator and maybe hook a logic analyser to the step & direction pins.

avizienis, 2021-01-21:
Maybe try inverting step impulse?

phomann, 2021-01-21:
Check that the stepper settings are the same in grblHAL and grbl advanced. 
It could be that step pulse width, acceleration, etc is different causing steps to be missed.

terjeio, 2021-01-21:
> The fact it works ok with Grbl Advanced makes me think this is to do with the firmware.

I agree, since you are using DM542T drivers it could be due to some step pulses beeing to narrow. It cannot be due to step pulses beeing to close to a dir change as there are too many steps lost for that.

Could you try to increase $0 - step pulse time a bit? And also add a pulse delay by setting $29 > 0?

I'll check the step pulse width.

LaurentLM, 2021-01-21:
I calculated 16µs per step at the max rate. This is within the spec of the driver (2.5µs minimum step width).

I tried inverting the step pulse, and changing the $0 to 15 and/or $29 to 5. I had to lower the max speed to 1000mm/min in some cases but still had the issue. 

From measurements, it seems the G1 command is not going far enough (instead of the G0 overshooting).

terjeio, 2021-01-21:
> From measurements, it seems the G1 command is not going far enough (instead of the G0 overshooting).

So G0 and G1 behaves differently?
Y-axis show the same error?

I have checked the step pulse width with a scope and it it is spot on and has very little jitter. And counting pulses (with a PSoC 5 and a hardware up/down counter) shows the correct result every time. Currently I do not have any way of measuring the pulse width of ever step pulse, 140mm of movement is 350000 pulses - that is a lot of data to check. Maybe I can add some code to the PSoC to automate a check...

Can you try by lowering the step pulse time as well - $0=5?

LaurentLM, 2021-01-21:
I deleted all the files I had and started from scratch again with the current test branch.
I can't reproduce the problem anymore, everything seems to works perfectly. 😃 

Really no clue what could have caused that in the first place. I didn't edit any of the code.

closed 2021-01-21 by LaurentLM