terjeio/grblHAL #205: Support for PCF8574 port expander driver development

mattbue, 2021-01-21:
Hi
as i wrote in another issue I want to use PCF8574 port expander on Phils breakout board.
So i startet getting familiar with the code and trying to communicate with the PCF8574 using the I2C driver (i2c.c & i2c.h) in the Teensy4 folder.

For basically tests I have modified main.c to the following:
```
#include "grbl/grbllib.h"
#include "i2c.h"

#define IOEX_ADDRESS (0x40)
#define READ_INPUT   0
#define RW_OUTPUT    1

void setup ()
{
    grbl_enter();
    
    i2c_init();
    
    uint8_t cmd[2];
    cmd[0] = RW_OUTPUT;
    cmd[1] = 0xAA;
    
    I2C_Send(IOEX_ADDRESS, cmd, 2, true);
}

void loop ()
{
}
```

In i2c_init() I added 
`#define I2C_PORT    4`
to select the port.

I was expecting to have the pins of the PCF8574 switch to the following mask: 10101010. To see this i have LEDs connected to the pins.
Unfortunatelly non of the LEDs is switched on. Maybe I'm on the wrong way...

I have the following questions:
Does someone every worked with the PCF8574? Is the way I try to communicate correct?
Am I using the i2c functions the correct way?

Thanks!

Please don't worry, this dirty way is just a brief test while getting familiar with everything.

mattbue, 2021-01-21:
By using the example "Scanner" of the Wire library i found out that my PCF8574 has the address 0x20 instead of 0x40.

So i used the example "master_writer" of the Wire library to send data to the PCF8574.
Using the following code the LEDs are switched on and off correctly:
```
#include <Wire.h>

void setup()
{
  Wire2.begin(); // join i2c bus (address optional for master)
}

uint8_t led = 1;

void loop()
{
  Wire2.beginTransmission(0x20);  // transmit to device #9
  Wire2.write(~led);              // sends five bytes
  Wire2.endTransmission();        // stop transmitting

  led = led << 1;

  if(led == 0)
    led = 1;
    
  delay(500);
}
```

All these test are done on the Teensy 4.1 mounted on Phils breakout board. So there was no modification in the hardware setup.
Am I using the I2C functions in the grblTeensy project in a wrong way?

terjeio, 2021-01-21:
> Am I using the I2C functions in the grblTeensy project in a wrong way?

I don't think so - but there are "native" grblHAL functions that can be used for data transfer:

https://github.com/terjeio/grblHAL/blob/579abbe27fe0aad5f3be705db6c5a924a8e2ffe8/drivers/IMXRT1062/grblHAL_Teensy4/src/i2c.h#L51-L53

Using this might be better as having different low-level code access the same hardware _could_ be problematic.

Depending on what you want to use the extra pins for you may want take a look at [the code](https://github.com/terjeio/grblHAL/blob/test/drivers/IMXRT1062/grblHAL_Teensy4/src/T41U5XBB.c) used for adding I/O with g-code support (M62-M68).

You can also write a plugin that redirect standard signals such as coolant on/off to the I2C interface.

mattbue, 2021-01-21:
When using the Wire library this was another project. I didn't mix it up within the grblTeensy project.
So when trying to to send data from within the grblTeensy project I already used the I2C_Send function but without success.
Can you confirm that the communication should work when using the function like this:
```
uint8_t cmd[1];
cmd[0] = 0xAA; // sending the bitmask 10101010 to the outputs
I2C_Send(0x20, cmd, 1, true);
```

terjeio, 2021-01-21:
This code should work, there are no registers in this chip that requires an address byte as part of the message. IIRC I have not tested single byte writes with my code so cannot be 100% sure. However, it _is_ used for the EEPROM/FRAM interface and works for that.

You should put the code in _driver.c_ instead of _main.c_ - or better just add _my_plugin.c_ to the project with your code:

```
#include "i2c.h"

void my_plugin_init (void)
{
  uint8_t cmd[1];
  cmd[0] = 0xAA; // sending the bitmask 10101010 to the outputs
  I2C_Send(0x20, cmd, 1, true);
}
```
`my_plugin_init()` will be called by _driver.c_ after basic configuration is completed, there is no need to modify source code for that.

Enable the keypad by removing the comment from `#define KEYPAD_ENABLE` in my_machine.h if  `#define EEPROM_ENABLE` is not already commented in, this will cause the _driver.c_ to initialize the I2C driver for you.

Check this [template code](https://github.com/terjeio/grblHAL/blob/test/templates/my_plugin.c) for an example of _my_plugin.c_ code.

mattbue, 2021-01-21:
Putting the 3 lines into my_plugin_init() works
#include "i2c.h"
```
__attribute__((weak)) void my_plugin_init (void)
{
    uint8_t cmd[1];
    cmd[0] = 0x00;
    I2C_Send(0x20, cmd, 1, true);
}
```

Let's see if I can get it to send e.g. the states of the limit switches.

phil-barrett, 2021-01-21:
I haven't used an expander but here is the code I use to read from an I2C device (ATTiny in I2C slave mode) plugged into the I2C header on the breakout board.  Uses the pjrc wire library and just prints out the data received in an intelligible fashion. 

```
#include <Wire.h>
#include "pendant_proto.h"
char *Axismsg[] = {"X Axis", "Y Axis", "Z Axis", "A Axis"};
int expon10[] = { 1, 10, 100, 1000};

int led = LED_BUILTIN;
int sigpin = 33;  //slave signal

union {
    char wbuffer[6];
    pendant_t pendant;
} p;

void setup()
{
  pinMode(led, OUTPUT);
  pinMode(sigpin, INPUT_PULLUP);
  
  Wire2.begin();  // join i2c bus (address optional for master)
  Serial.begin(9600);       // start serial for output
}

void loop()
{
 
  int i;
  long cnt = 0;
  char *bp;

  while(1){
 //   Serial.printf("sigpin: %d\n", digitalRead(sigpin));
    if(digitalRead(sigpin) == LOW) {  // data available
      cnt++;
      Wire2.requestFrom(I2CDEVNUMBER, 1);   // request from pendant processor (slave)
      bp = p.wbuffer;
      uint8_t c = Wire2.read();   // read a byte

      switch(c) {
        case STARTBUTTON:
          Serial.printf("Start Button\n");
          break;
        case HOLDBUTTON:
          Serial.printf("Hold Button\n");
          break;
        default:
          Serial.printf("%s X%d %c%d Steps\n", Axismsg[c & 0x3], expon10[(c>>3)&0x3], (c>>2)&0x1?'+':'-', (c>>5));
      }
  
//      Serial.printf("%s %d %d %s %s\n", Axismsg[p.pendant.p_axis] , p.pendant.p_mag , p.pendant.p_steps , ((p.pendant.p_cmd&SWBIT) ? "Switch Pressed":" "), 
//                    ((p.pendant.p_cmd&ESTOPBIT) ? "EStop":" " ));
//      bp = p.wbuffer;
//      for(i=0;i<6;i++)Serial.printf("%x", *bp++ );
//      Serial.println();
    }
    delay(100);

  }
}
```

mattbue, 2021-01-21:
@terjeio do you have an hint for me how to get/read the current limit switch states out of my plugin?

terjeio, 2021-01-21:
> Putting the 3 lines into my_plugin_init() works

Did you add _my_plugin.c_ (in the same folder as _driver.c_ is found)? It seems to me that you have modified the original code in the grbl folder. Don't do that. Your _my_plugin.c_ will override the contents in _grbl/my_plugin.c_ automagically.

>  do you have an hint for me how to get/read the current limit switch states out of my plugin?

You have to "claim" HAL entry points for this. Limit switches are a bit tricky to start with since you have to wire the `INT` pin from the PCF8574 to an interrupt capable pin and handle that. To get a feeling with the HAL it will be easier to start with outputs? 

Anyway, limit pins handling is implemented by a driver (or a plugin) by setting function pointers to the code to be executed.

To read set or enable limit pins these has to be set to point to your code (example is from _driver.c_):

https://github.com/terjeio/grblHAL/blob/2d871b2d0a71193df8015f98da1bc565d12341d0/drivers/IMXRT1062/grblHAL_Teensy4/src/driver.c#L2078-L2079

In _driver.c_ these are implemented like this:

https://github.com/terjeio/grblHAL/blob/2d871b2d0a71193df8015f98da1bc565d12341d0/drivers/IMXRT1062/grblHAL_Teensy4/src/driver.c#L977-L1027

In your interrupt handler you have to tell the core about what happened, again from _driver.c:_

https://github.com/terjeio/grblHAL/blob/2d871b2d0a71193df8015f98da1bc565d12341d0/drivers/IMXRT1062/grblHAL_Teensy4/src/driver.c#L2440

If you are going to support pullup enable/disable you have to hook into the settings changed event and pass the event on:

https://github.com/terjeio/grblHAL/blob/2d871b2d0a71193df8015f98da1bc565d12341d0/drivers/IMXRT1062/grblHAL_Teensy4/src/driver.c#L2066

You have to save the original value of this pointer to a local copy before changing it, and then in your code call the original code via the saved copy. Here is how it is done for another entry point in the template code linked above:

https://github.com/terjeio/grblHAL/blob/579abbe27fe0aad5f3be705db6c5a924a8e2ffe8/templates/my_plugin.c#L177-L179

and in the local copy the original is called before any local actions:

https://github.com/terjeio/grblHAL/blob/579abbe27fe0aad5f3be705db6c5a924a8e2ffe8/templates/my_plugin.c#L158-L162

_driver.c_ has to be modified as well so that the current limit pins can be freed for other uses?

FYI the [SAMD21 driver](https://github.com/terjeio/grblHAL/blob/master/drivers/SAMD21/grblHAL_MKRZERO/src/driver.c) has code for how to use an [expander for outputs](https://github.com/terjeio/grblHAL/blob/master/drivers/SAMD21/grblHAL_MKRZERO/src/ioexpand.c), defined by the [CNC BoosterPack map](https://github.com/terjeio/grblHAL/blob/master/drivers/SAMD21/grblHAL_MKRZERO/src/cnc_boosterpack_map.h). This changes code in _driver.c_ directly - another option for achieving this.

closed 2022-01-27 by terjeio