terjeio/grblHAL #209: STM32F103CBT6 doesnt send anything via UART

e-zeki, 2021-01-23:
Hi, 
I compiled the code and load it intu a bluepill (I've replaced the mcu) but I'm unable to get any messages from uart. 
Code compiles and runs correctly. serial init is setted. but no response what so ever.
do I need to twig some other files other than mentioned in read me file? 

Best regards
Zeki

terjeio, 2021-01-23:
Hav you commented out `#define USB_SERIAL_CDC`  in my_machine.h?

e-zeki, 2021-01-23:
Yes I have

terjeio, 2021-01-23:
> I went step by step in debug it seem it goes all the way through searial send,

Even in the interrupt handler? Set a breakpoint here to check:

`        USART1->DR = txbuf.data[tail++];        // Send next character and increment pointer`

Receive is working?

e-zeki, 2021-01-23:
receive is not workin I just checked. Also 
`USART1->DR = txbuf.data[tail++]; // Send next character and increment pointer` is working tail is incrementing and I also see the welcoming message in data array. 
I opened both GRBL and receiver MCU (custom stm32f401 board.)
GRBL sends 1 char tail increments by one, but receiver (f401) receives 3 consecutive 0h,0h,0h for every char GRBL sends.
its constant GRBL sends 1 char, receiver receives 3 0h always.
Also another question do I need to remove `cnc_bossterpack_map.h` and  `cnc3040_map.h` files. would these cause any confusion?

terjeio, 2021-01-23:
>receive is not workin I just checked. 

By setting a breakpoint in the interrupt handler?

> GRBL sends 1 char tail increments by one, but receiver (f401) receives 3 consecutive 0h,0h,0h for every char GRBL sends.
its constant GRBL sends 1 char, receiver receives 3 0h always.

If the correct inserted into `USART1->DR` then either you have a hardware problem or a baud rate problem. Since something is actually output on the wire then IMO not likely a hardware problem.

> Also another question do I need to remove cnc_bossterpack_map.h and cnc3040_map.h files. would these cause any confusion?

No - but check that both boards are commented out in my_machine.h to be sure.

Di you have a scope or logic analyzer available?

e-zeki, 2021-01-23:
>By setting a breakpoint in the interrupt handler?

Yes.

>If the correct inserted into USART1->DR then either you have a hardware problem or a baud rate problem

I should check DR for hex value 
> Di you have a scope or logic analyzer available?

I'll have tomorrow I'll check and let you know

terjeio, 2021-01-23:
You have to store the character to send in a char variable and check that. This since you cannot read the character back from DR.

```
char c = txbuf.data[tail++] ;
USART1->DR = c;
```

e-zeki, 2021-01-23:
>char c = txbuf.data[tail++] ;
USART1->DR = c;

I've did this and char is ok I can see the value but still I receive 3 0h for  every character sent.
I've set baud rates 9600 both sides too

HuubBuis, 2021-01-24:
> Have you commented out `#define USB_SERIAL_CDC` in my_machine.h?

I have had a similar problem. Sometimes it worked by commenting out and sometimes not. I was unable to reproduce the working and not working situation.  I expect it is more STM32Cube IDE problem.
I solved it by commenting out the line and adding a new #define to (if I remember it correct!!!!)

```
#if !(defined(NUCLEO_F411) || defined(NUCLEO_F446)) // The Nucleo-F411RE board has an off-chip UART to USB interface.
//#define USB_SERIAL_CDC       1 // Serial communication via native USB.
#endif
#define USB_SERIAL_CDC      0 // Serial communication via uart
```

e-zeki, 2021-01-24:
@HuubBuis do you mean something like commenting might not work? But define parameter forces to compiled correctly? 
That's a very odd thing if actually works. But I'll try that too.

terjeio, 2021-01-24:
Since data is put into the transmit register and something is ouput on the wire then the `#define` is working as it should.
Sometimes when changing `#define`s it may be a good idea to _Clean Project_ and perform an _Index > Rebuild_ to be sure everything get compiled correctly and presenting itself as it should in the IDE (greyed out or not).

BTW I have connected to a F103 via a FTDI breakout and serial comms is working for me. This with the latest test branch.

Are you sure the F401 is set up correctly, the new CPU is ok and not something else than a real F103?

HuubBuis, 2021-01-24:
> 
> 
> @HuubBuis do you mean something like commenting might not work? But define parameter forces to compiled correctly?
> That's a very odd thing if actually works. But I'll try that too.

I don't know what caused the problem because adding the line fixed it and after removing, it still worked and suddenly it was back again. I also tried rebooting the PC, restarting the IDE, cleaning , rebuild index etc.

e-zeki, 2021-01-24:
>Are you sure the F401 is set up correctly, the new CPU is ok and not something else than a real F103?

Yes I'm sure 401 is OK because like I said, I have 2 F103 board, one is yours the otherone is robomecs build. When I replace these two, F401 gets everything from robomecs build, but your build is giving out all zeros. I couldn't get a hold on a scope today I'll try with both scope and a ftdi module to be sure.
Also clean build might help I'll check that too

terjeio, 2021-01-24:
Ok, this is weird. Is USB working?

e-zeki, 2021-01-27:
Hello again,
I think I found the problem.  My PCLK2 migth not working some how.
`USART1->BRR = UART_BRR_SAMPLING16(HAL_RCC_GetPCLK2Freq(), 115200);`
My Baudrate is set here (I didnt find anywhere else I assume it sets here?)
But  I looked it with a scope and my output rate is 12.5kHz
when I change it to `USART1->BRR = UART_BRR_SAMPLING16(HAL_RCC_GetPCLK2Freq(), 9600);`
it drops to 1.25kHz 
bytes are sent, there is no mistake on sending. diffent bytes goes. 
That explanes receive not working too my F401 is set to 115200 F103 just coulnt catch a byte that fast with 12.5kHz i think.

This `HAL_RCC_GetPCLK2Freq()`  returns 4.500.000 
SystemCoreClock is set to 72.000.000
and I noticed something else on the start it waits around 100-250 ms at `SystemCock_Config()`
Would be my MCU is not behaving correctly, maybe oscilator failure? 
I ordered new MCU and oscilators I'll try to change them one by one and let you know the results
Zeki.

terjeio, 2021-01-27:
> on the start it waits around 100-250 ms at SystemCock_Config()

I do not think this is unusual as typically there is some delay added to wait for the oscillator to stabilize.

> Would be my MCU is not behaving correctly, maybe oscilator failure?

Or could it be that the FTDI chip I am testing with is more tolerant to baud rate errors?

Again, is USB working? If not then surely the clock system is off. Or the crystal and/or load caps are bad?

e-zeki, 2021-01-27:
>Or could it be that the FTDI chip I am testing with is more tolerant to baud rate errors?

I highly doubt that I checked the baudrate with oscilloscope its exactly 12.5 kHz when its set 115200 . when I set it to 9600 its 1.25 khz. it's clearly a oscilation error on my part.  

It's 99% my MCU's osc pin damaged. I read 300-500 ohm between osc in pin which should be insulated. 
I checked other blue pills on my inventory everone of them insulated. Also didnt see a wave on scope too. I changed the oscillator with one of the boards didnt have any effect. 
I didnt try USB I dont know how actually I never tried these builds with USB before sorry I'll check that too.

terjeio, 2021-01-27:
> I highly doubt that I checked the baudrate with oscilloscope its exactly 12.5 kHz when its set 115200 . when I set it to 9600 its 1.25 khz. it's clearly a oscilation error on my part.

You wrote that the board works with Robomechs port regardless? Confusing...

e-zeki, 2021-01-29:
I figured the problem and solved it. 
Yes I wrote board is working with robomechs build I have 2 blue pill boards and a custom build pcb to sit this board on inorder to work with motors and uart. I changed the MCU one of the blue pills from C8T6 to CBT6 make a CBT Bluepill.
I tried to workit with CBT module and didnt work as you know, I changed back to C8T6 MCU same board. and it started working with robomechs build again!. 
CBT6's OSC_IN Pin is damaged. which caused the problem. because since oscillator is ok, your build goes HSE_ON but it actually cant produce 8 MHz signal. 
Aynway I'm receiving and sending uart signals correctly rightnow. 
Just another issue I'm getting error 18 which is not listed. Am I getting al the weirdest faults here??

Thank you for your help and care on this issue you've helped me tons.

e-zeki, 2021-01-29:
Hello again so sorry to bother you this much. but I couldn't quite understand the reset pin configuration.
on blue pill its conencted to 3.3V with 10k resistor. and a reset button which takes down to gnd. 
on this state I get error 18, I removed the 10k resistor from 3.3V and put it to GND. But it went crazy.  All I get from grbl is some  random bytes. 
What did I do wrong this time?

terjeio, 2021-01-29:
> What did I do wrong this time?

I do not know. There are two reset pins, could it be that you have removed the resistor from the processor "hard" reset? That needs to be pulled high for the processor to run correctly.
grblHAL (and Grbl) has its own reset pin for a softer reset, this should be wired to ground or $14 set to [invert its input](https://github.com/terjeio/grblHAL/wiki/First-Run-Grbl-Settings).

e-zeki, 2021-01-29:
I'm an idiot I went and change the hardware reset so sory. you can close this here 
Thanks a lot

closed 2021-01-29 by e-zeki