terjeio/grblHAL #210: Spindle Invert?

phil-barrett, 2021-01-24:
Downloaded current master branch today and rebuilt for iMXRT1062 using T41U5XBB mapping. Invert Spindle Signals ($16) - Invert PWM, bit 2, does not cause PWM inversion.  It also is not persistent - set it, power cycle board and the setting is cleared.  (Other settings persist a power cycle so it is not an incorrect EEPROM setting.)  This also exists in grblHAL versions as far back as May of 2020, by the way.

Invert CCW, bit 1, does work and persists.
Invert Spindle On, bit 0, also works and persists but confuses ioSender at start up after a power cycle.  ioSender sees the spindle  as on initially (but it is not).  Clicking on Off gets ioSender back in sync.

A brief perusal of driver.c did show anything obvious.

There is also some strange behavior when you switch enable invert while running the spindle. grblHAL (or ioSender?) thinks running is stopped and vice versa.  Probably unrelated and not a good idea to do on a real machine...

An ioSender suggestion (I will post there as well):  In Settings:Grbl, Invert Spindle Settings, change Spindle CCW to Spindle Direction and change Spindle On to Spindle Enable.  (though Spindle On is debatable, Enable seems to be the more common usage.)

terjeio, 2021-01-24:
> A brief perusal of driver.c did show anything obvious.

Check the driver [capabilities matrix](https://github.com/terjeio/grblHAL/blob/master/drivers/ReadMe.md), inverted spindle PWM is not implemented for iMXRT1062. This is why the setting does not persist.

I guess I should raise an error for settings not available, and for the new settings enumeration (`$ES`) remove the option. Or implement all options for all drivers.

>  In Settings:Grbl, Invert Spindle Settings, change Spindle CCW to Spindle Direction and change Spindle On to Spindle Enable. 

Settings specifications used to create th UI is fetched from the controller now - except the description that is still fetched from a file. It is on my todo list to add descriptions to the controller output, perhaps via a separate enumeration command. Flash space is an issue for some drivers though, and is the reason it is not already in place.

The text should be checked, both the UI bits and the description - preferably by a native english speaker with good knowledge of the technical lingo involved?

phil-barrett, 2021-01-24:
> Check the driver [capabilities matrix](https://github.com/terjeio/grblHAL/blob/master/drivers/ReadMe.md), inverted spindle PWM is not implemented for iMXRT1062. This is why the setting does not persist.

My bad for not checking that.  Why is it not supported?  Seems like a very simple thing to do.  Though, not that useful except when a HW designer has a brain fade and gets an optoisolator input backwards...  I have heard of laser power controllers that use an inverted PWM input but it seems to be one oddball piece of hardware.
> 
> I guess I should raise an error for settings not available, and for the new settings enumeration (`$ES`) remove the option. Or implement all options for all drivers.

Maybe not show options that aren't supported?

> Settings specifications used to create th UI is fetched from the controller now - except the description that is still fetched from a file. It is on my todo list to add descriptions to the [...]

In my experience, Scandinavians speak better English than most native English speakers... I will take a look, if you wish.

terjeio, 2021-01-24:
> Why is it not supported? Seems like a very simple thing to do.

For some processors it is simple, for others not so. E.g. when switching the spindle off the PWM output should stay high when inverted. Perhaps it is simple for the iMXRT1062 - I will have to consult the datasheet to see if is just a matter of flipping a bit in a register.

> Maybe not show options that aren't supported?

That is what I want to achieve, preferably without sender side code handling that. And I want the core to be able to enumerate settings, even for plugins and drivers, in a generic way. 

> I will take a look, if you wish.

That would be nice. For UI data except descriptions you should use output from `$E*` (in the file below)? The latest descriptions, but not the latest other details, can be found in _hal_setting_codes_en_US.txt_ in the sender folder.

[enums.txt](https://github.com/terjeio/grblHAL/files/5861669/enums.txt)

terjeio, 2021-08-22:
Spindle PWM inversion will be available from driver version 210822 (to be committed later).

closed 2021-08-22 by terjeio